arendst/Tasmota #8784: No PN532 Event

ChrisVonFrosta, 2020-06-25:
### PROBLEM DESCRIPTION
Triing to do a rule in Tasmota on ESP2866. With standards like
`Rule1 ON Power1#State DO Power2 1 ENDON`
it works.
But i got no Event by reading an NFC Tag.

i read that from https://tasmota.github.io/docs/PN532/ that i have to activate the events by editing the source files and compile it by my self. Is ther no other way? Maybe with console?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP2866
- [ x] Tasmota binary firmware version number used: 8.3.0 (sensors)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
```

- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
stat/tasmota_5FB8B8/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":470,"Rules":"ON PN532_UID#STATE DO Power2 TOGGLE ENDON"}

stat/tasmota_5FB8B8/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":470,"Rules":"ON EVENT#PN532_UID DO Power2 TOGGLE ENDON"}
```
- [x ] Provide the output of this command: `Status 0`:
```
 15:40:03 CMD: Status 0
15:40:03 SRC: WebConsole from 192.168.168.158
15:40:03 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS = {"Status":{"Module":18,"FriendlyName":["SMART-Lichtsensor","Tasmota2"],"Topic":"tasmota_5FB8B8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:58","StartupUTC":"2020-06-25T14:14:05","Sleep":200,"CfgHolder":4617,"BootCount":277,"BCResetTime":"2020-04-27T23:34:13","SaveCount":516,"SaveAddress":"F9000"}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-03-20T14:48:25","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"442/699"}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Phradio","WashTeam"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00040000","00000000"]}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS5 = {"StatusNET":{"Hostname":"SMART-Lichtsensor","IPAddress":"192.1XXXXX","Gateway":"192.16XXXX","Subnetmask":"255.255.255.0","DNSServer":"192.1XXX","Mac":"EC:FA:XXXXB8:B8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.168.167","MqttPort":1883,"MqttClientMask":"SMART-Lichtsensor","MqttClient":"SMART-Lichtsensor","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS7 = {"StatusTIM":{"UTC":"2020-06-25T14:40:03","Local":"2020-06-25T15:40:03","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:48","Sunset":"20:58"}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS10 = {"StatusSNS":{"Time":"2020-06-25T15:40:03","ANALOG":{"Illuminance":120}}}
15:40:03 MQT: stat/tasmota_5FB8B8/STATUS11 = {"StatusSTS":{"Time":"2020-06-25T15:40:03","Uptime":"0T00:25:58","UptimeSec":1558,"Heap":22,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":21,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":2,"SSId":"XXXX","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:10"}}}
15:40:04 WIF: Checking connection...

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```

```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

dabos1, 2020-06-25:
you could use tasmocompiler

Jason2866, 2020-06-25:
Or use Gitpod to compile your version. You just need a browser. Nothing needs to be installed.
Hiw it is done is explained in the docu.

closed 2020-06-25 by ascillato2