arendst/Tasmota #8785: Tuya based curtains don't move the correct % because tasmota send wrong value to MCU

MadDoct, 2020-06-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have tuya based curtain motors that have dpid2 set as dimmer
They used to work perfectly until I updated tasmota to the latest release. Now, when I set 50% opening it send only 15% to the MCU... If I set 15, it sends 1...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): tuya based curtain motor
- [x ] Tasmota binary firmware version number used: 8.3.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: OTA update
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:35:42 CMD: Backlog Template; Module; GPIO 255
23:35:42 SRC: WebConsole from 192.168.2.104
23:35:42 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
23:35:42 SRC: Backlog
23:35:42 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
23:35:42 MQT: stat/CortinaSalaJantar/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:35:43 SRC: Backlog
23:35:43 CMD: Group 0, Index 1, Command "MODULE", Data ""
23:35:43 MQT: stat/CortinaSalaJantar/RESULT = {"Module":{"54":"Tuya MCU"}}
23:35:43 SRC: Backlog
23:35:43 CMD: Group 0, Index 1, Command "GPIO", Data "255"
23:35:43 MQT: stat/CortinaSalaJantar/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:37:00 MQT: stat/xxxxxxxxxx/STATUS = {"Status":{"Module":54,"DeviceName":"xxxxxxx","FriendlyName":["xxxxxx"],"Topic":"CortinaSalaJantar","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:37:00 MQT: stat/CortinaSalaJantar/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"5N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"19T07:01:34","StartupUTC":"2020-06-06T15:35:26","Sleep":50,"CfgHolder":4617,"BootCount":898,"BCResetTime":"2020-05-21T15:10:36","SaveCount":3896,"SaveAddress":"F7000"}}
23:37:00 MQT: stat/xxxxxx/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"394/699"}}
23:37:00 MQT: stat/xxxxxx/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxx","xxxxxxx"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08029","2805C8000100060000005A00000000000000","00010000","00000000"]}}
23:37:00 MQT: stat/xxxxxxx/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:37:00 MQT: stat/xxxxxxxxxxx/STATUS5 = {"StatusNET":{"Hostname":"xxxxxxxxxxx","IPAddress":"192.168.xxxx","Gateway":"192.168.xxx","Subnetmask":"255.255.255.0","DNSServer":"192.168.x.x","Mac":"2C:F4:32:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":0.0}}
23:37:00 MQT: stat/xxxxxxxx/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_22EEA4","MqttUser":"xxxx","MqttCount":92,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:37:00 MQT: stat/xxxxxxx/STATUS7 = {"StatusTIM":{"UTC":"2020-06-25T22:37:00","Local":"2020-06-25T23:37:00","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"04:48","Sunset":"20:57"}}
23:37:00 MQT: stat/xxxxxxxx/STATUS10 = {"StatusSNS":{"Time":"2020-06-25T23:37:00"}}
23:37:00 MQT: stat/xxxxxxxxx/STATUS11 = {"StatusSTS":{"Time":"2020-06-25T23:37:00","Uptime":"19T07:01:34","UptimeSec":1666894,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":92,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxx","BSSId":"00:11:32:xxxxxxx","Channel":6,"RSSI":64,"Signal":-68,"LinkCount":3,"Downtime":"0T08:05:27"}}}

```
- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
23:32:59 MQT: Received Topic "cmnd/CortinaSalaJantar/dimmer", Data Size 2, Data "50"
23:32:59 SRC: MQTT
23:32:59 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
23:32:59 MQT: stat/CortinaSalaJantar/RESULT = {"POWER":"ON","Dimmer":50}
23:32:59 TYA: Send dim value=15 (id=2)
23:32:59 TYA: Send "55aa00060008020200040000000f24"
23:32:59 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA03070008020200040000000F28","Cmnd":7,"CmndData":"020200040000000F","DpType2Id2":15,"2":{"DpId":2,"DpIdType":2,"DpIdData":"0000000F"}}}
23:32:59 TYA: fnId=21 is set for dpId=2
23:32:59 TYA: RX Dim State=15
23:32:59 MQT: tele/CortinaSalaJantar/RESULT = {"TuyaReceived":{"Data":"55AA0307000D0104000102020200040000000127","Cmnd":7,"CmndData":"01040001020202000400000001","DpType4Id1":2,"1":{"DpId":1,"DpIdType":4,"DpIdData":"02"},"DpType2Id2":1,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000001"}}}
[... 59 lines cut ...]
23:33:21 TYA: fnId=21 is set for dpId=2
23:33:21 TYA: RX Dim State=1
23:33:21 SRC: Switch
23:33:21 CMD: Group 0, Index 3, Command "DIMMER", Data "1"
23:33:21 TYA: Send dim skipped value=0
23:33:21 MQT: stat/CortinaSalaJantar/RESULT = {"POWER":"ON","Dimmer":1}
23:33:21 CFG: Saved to flash at F8, Count 3895, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
set dimmer value to 50% or any other value

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
get a corresponding opening

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-26:
Have you correctly set `DimmerRange` ?

MadDoct, 2020-06-26:
Hi. Yes, I have. As said before, it was working perfectly before... It only started happening after updating and I did check that the dimmerrange was still correct.
By the way, I have 4 such curtain motors and they all behave the same...

blakadder, 2020-06-27:
Have you disabled LedTable since dimmer is for dimming not curtaining? Here's a lot more on the topic https://blakadder.com/tuya-covers/

MadDoct, 2020-06-30:
You were right @blakadder . Ledtable solved it... It must have been introduced in the meantime... Thanks!

closed 2020-06-30 by MadDoct