arendst/Tasmota #8789: Next try of zero crossing PWM control after recent changes

stefanbode, 2020-06-26:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
AC Dimmer need the PWM synced with the AC sinus. This was I possible due to the PWM implementation until know. latest changes generate a very fine and reliable signal. The zero crossing therefore only need do Trimm after some time. Not on every sinus.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Check if and how we can integrate the zero crossing as an input signal.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-27:
AC Dimming in Tasmota is going to be a challenge because it is not a Real Time Software for making a precise ZCD.

Anyway, It can be checked also how it is implemented in ESPHome (https://github.com/esphome/esphome/pull/880)

stefanbode, 2020-06-27:
Here are my idea how we can make it possible:
- User PWM1 and PWMfrequency 100 or 120 depending on country
- Use COUNTER1 as input for the zero crossing
- Every 1000 steps (10 seconds) do some slight magic to keep the PWM start in Sync. There is now a PHASE that we might can use (@s-hadinger can you help how we can influence the phase synchronisation?)

Real Time Software is not required because be can use the PWM frequency. This is now after several fixes the last week very stable. We need "just" start the PWM with the zero cross and then check every 10 seconds if it is in phase and not shift it a few ns.

stefanbode, 2020-06-29:
I changed the counter driver with just some lines. After 1 second i realign the phase with the zero trigger. Works fine on the oscilloscope. Anyhow my first test with a 60w bulb did not work. Now it look like the triac is broken. The data sheet say 2A ok and max 5A. The bulb in operation only creates about 0.3A. Not heat detected. Any idea? Will share code changes for retest.

Jason2866, 2020-06-29:
@stefanbode  Probably harmonic oscillations in the high-frequency range were created.
That kills.
https://books.google.de/books?id=4OTRBgAAQBAJ&pg=PA150&lpg=PA150&dq=Triac+RC+glied&source=bl&ots=OD3hV2Rk7e&sig=ACfU3U13Ynlf-BIL_GRs2ZpVw558e-he0A&hl=en&sa=X&ved=2ahUKEwj2l4KIsKfqAhVC_qQKHYm7BcsQ6AEwCHoECAYQAQ#v=onepage&q=Triac%20RC%20glied&f=false

stefanbode, 2020-06-30:
I will try to upload the reference sketch first and check if this in principal works. The bulb is a pure ohm-60W-lamp. Just to ensure not to make it to complicated fist. Anyhow this part of hardware seems more complicated than expected. SSR? Never heard. Need to google it. Will upload some pictures and videos today.

stefanbode, 2020-06-30:
Ok, SSR solid-state-relay. I assume this will not solve the Pwm problem that you need to start with the zero crossing. Just putting 100HZ PWM will result in pulsating lamp because the RMS changes if you start in different times of the sinus. The frequency of the main slightly changes and also the esp is not 100% accurate.

Jason2866, 2020-06-30:
Yes, a SSR is just easier to use. Zero Cross Detection is still needed and needs to be insync with switching the Relay on/off

stefanbode, 2020-06-30:
I got it working without flickring and in most situations. Support only PWM1 with COUNTER1. All changed though DIMMER command or the Web-Slider.

Remove the "core...." files from source.
Use latest PWM module (platform.ino)
```
[tasmota_core]
; *** Esp8266 Arduino core 2.7.1
platform                  = espressif8266@2.5.3
platform_packages         = https://github.com/Jason2866/platform-espressif8266/releases/download/2.8.0/esp8266-2.8.0.zip
build_flags               = ${esp82xx_defaults.build_flags}
```

Add xns_01_counter.ino line 79ff
```
    if bitRead(Counter.pin_state, index) {
      //stb mod
      // PWMfrequency 100
      // restart PWM 4-times a second (german 50Hz has to much deviation)
      // set COUNTERDEBOUNCELOW 1 to catch the raising edge
      // Zero-HIGH is typical 2ms
      if (RtcSettings.pulse_counter[index]%25 == 0 && PinUsed(GPIO_PWM1, index)) {
        // stop pwm on PIN to start in Sync with rising edge
        // calculate timeoffset to fire PWM
        uint16_t dimm_time= 10000 / Settings.pwm_frequency * (100 - light_state.getDimmer(index)) ;
        analogWrite(Pin(GPIO_PWM1, index), 0);
        delayMicroseconds(dimm_time);
        // fire small PWM signal to start TRIAC/SSR. Kill on next
        // zero phase automatic.
        analogWrite(Pin(GPIO_PWM1, index), 100);
      }
      return;
```

Current findings:
The often sold preconfigured Items are are cutting leading edge. Therefore you must start dependent on DIMMER late in the sinus wave the PWM. The PWM just needs 1% and can be fixed. The light intensity is defined by the delay after the zero cross before you trigger the Triac. The Triac stay open until the next zero phases. Nothing to do to stop.

Open issues:
need a setoption for the mess. Theoretically, with some changes it should be possible to drive up to FOUR Dimmer (setoption68 1). On DIMMER=0 or other low values I have exceptions. Still something wrong
Not clear if counter needs a reset from time to time.

stefanbode, 2020-07-01:
@Jason2866, @arendst : Where can I find the source files core_esp8266_waveform... core_esp8266_wiring.... that you use in the ZIP file of core 2.8.0. These work fine and I can influence the start of the PWM. The files in TASMOTA do not have this functionality.

stefanbode, 2020-07-01:
Done. Problem fixed!

Jason2866, 2020-07-01:
@stefanbode  Tasmota has the same files as my compilation Core "2.8"  for PWM since the shutter fix. 
The files are in source folder "Tasmota"

stefanbode, 2020-07-01:
The PR is now submitted. But I need more control over the PWMfrequency. Currently implemented by adding an arduino standard class. Any help is appreciated. Without this change there is a very visible pumping of the light intensity visible. Especially at low light. This is because electricity companies uses the frequency to manage their network load. You ave to follow this to have a steady light. If implementing more AC dimmer this will only work on the same AC phase. They must all share the same zero-counter. Let's see how and ifthe community likes it and what feature request are comming

TrDA-hab, 2020-07-03:
Will "robotdyn dimmer" work correctly now? And can I buy it?
https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html
https://aliexpress.ru/item/32802025086.html

stefanbode, 2020-07-07:
Yes, this will work. I tested it exactly with this device. Made some changes today to fix accidental reboot when you change dimmer on the webpage.

closed 2020-07-07 by stefanbode

martin072, 2022-02-14:
> Yes, this will work. I tested it exactly with this device. Made some changes today to fix accidental reboot when you change dimmer on the webpage.

@stefanbode Can you tell me how to get this working? I have an AC dimmer from Robotdyn, connected it to a Wemos D1 

ZC to D7 
PWM to D6
Powered from 3.3v
Configured D6 as PWM 1 and D7 as ZC Pulse
Nothing works, light just goes on and off. If I play with the PWM Frequency I get some strange and non consistent  dimming (dims, then brightens, sometimes switches off and on). 

I tried setOption99 1 but then nothing works.

Any help would be much appreciated :)

schejbity, 2022-02-27:
> > Yes, this will work. I tested it exactly with this device. Made some changes today to fix accidental reboot when you change dimmer on the webpage.
> 
> @stefanbode Can you tell me how to get this working? I have an AC dimmer from Robotdyn, connected it to a Wemos D1
> 
> ZC to D7 PWM to D6 Powered from 3.3v Configured D6 as PWM 1 and D7 as ZC Pulse Nothing works, light just goes on and off. If I play with the PWM Frequency I get some strange and non consistent dimming (dims, then brightens, sometimes switches off and on).
> 
> The little blue led on the robotdyn board seem to show the correct pwm, ie. dims and brightens when changing the dimmer
> 
> I tried setOption99 1 but then nothing works. (little blue led on the robotdyn board does nothing)
> 
> Any help would be much appreciated :)

I have same problem.

stefanbode, 2022-02-28:
The esp32 is currently not supported. Just esp8266 because here I have native access to the PWM curve

stefanbode, 2022-02-28:
I have a day off today. Maybe I can find a comparable function for the esp32. I need to change frequency e.g. Fromm 100.000Hz to 100.008Hz. Not tested how to do this on esp32

stefanbode, 2022-02-28:
I just committed a new version to DEVELOPMENT branch. I hope this will enable it for the ESP32, but currently I'm a little less on usable devices to test. If you can give feedback this would be great

stefanbode, 2022-02-28:
Please all read the updated documentation. There is a major and breaking change. All in all now we support 3 dimmer in parallel and the sync is much better now. Also all crashes are addressed and fixed as far as I can see

schejbity, 2022-03-01:
> I just committed a new version to DEVELOPMENT branch. I hope this will enable it for the ESP32, but currently I'm a little less on usable devices to test. If you can give feedback this would be great

Reflashed ESP32 with DEV version 11.0.0.3
Using this configuration and wiring:
[https://tasmota.github.io/docs/Lights/#1-channel-dimmable-light](url)

SetOption99 = 1
PWMFREQUENCY = 100
LEDTABLE = 0

My pinout:
GPIO5 - Zero Cross - Counter4 - I see, that counter is counting zero crosses
GPIO16 - Triak gate - PWM1
GPIO17 - Feedback - Counter1 - Counting, when bulb lights.

Testing with normal bulb.
Bulb is blinking and dimming, like disco mode.
With changing dimmer value disco mode "confusedly changing frequency".
The toggle button does nothing.
When I shift the dimmer to zero, the light bulb is still in disco mode.

BusinessClaes, 2022-05-06:
> > Please all read the updated documentation. There is a major and breaking change. All in all now we support 3 dimmer in parallel and the sync is much better now. Also all crashes are addressed and fixed as far as I can see
> 
> 
> 
> @stefanbode could you please advise what document you are refering ?
> 
> I'm failing to find anything new in the doc regarding new requirements.
> 
> Thanks

https://tasmota.github.io/docs/Lights/#1-channel-dimmable-light

:)

barbudor, 2022-05-06:
So if I get it correctly we can support only 3 PWM with ZC
Each PWM_[1..3] need to be fedback to COUNTER[1..3] and COUNTER4 is used for ZC detection.
So on the 4 channels RoboDyn, only 3 PWM would be supported (and eventually the 4th as a relay on/off only)
Is that correct ?

stefanbode, 2022-05-09:
Yes. And if you switch on one Pwm there is a short time and small influence on the other lights if they are on and dimmed to a low value.

cacengineering, 2023-06-20:
How did you eventually synch the Zero crossing to the PWM - i.e start the pwm cycle ?

cacengineering, 2023-06-20:
Appreciate the response. I did try to use a interrupt based on the Tasmota configurations options but could ot get it working - so went back Counter.  If you can help with the following two issues:
1) How can we synchronise the pwm with counter ? Would still like to play with it.  I am more a hardware expert and can possibly make it happen.
2)   Any suggestions to setting up interrupt - ideally to trigger a variable timer ?

By the way - the Gui for pwn goes really flaky when option99 set.  Did you have that problem ?

stefanbode, 2023-06-21:
The gui problems haven't been noticed. But also using the esp32 240mhz. It could be because the interrupt scratches what is technical possible. Zero cross dimmer will not work with high cpu load coming from other sensors. Tested with 1-4 dimmer and mqtt to control. No additional sensors connected.

nccchirag, 2024-09-18:
> The gui problems haven't been noticed. But also using the esp32 240mhz. It could be because the interrupt scratches what is technical possible. Zero cross dimmer will not work with high cpu load coming from other sensors. Tested with 1-4 dimmer and mqtt to control. No additional sensors connected.

may I request you to share the template and/or user_config_override.h ... also which hardware did you use for the test (1-4 dimmer)?