arendst/Tasmota #8790: Add support for  IEC 62056 / EN 62056-21 utility meters

ilgrank, 2020-06-26:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
not related to a problem

**Describe the solution you'd like**  
I would like to be able to read smart Gas/Energy meters using a phototransistor.
HopenHAB seems to have a basic implementation already ( [https://www.openhab.org/addons/bindings/smartmeter/](https://www.openhab.org/addons/bindings/smartmeter/) ) , but it would be nice to be able to read values from Tasmota:

**Describe alternatives you've considered**  
I've tried the code here on Arduino: [http://tunn.us/arduino/landisgyr.php](http://tunn.us/arduino/landisgyr.php)

**Additional context**  
A sample of utility meter using the port: [http://www.metersit.com/images/pdf/TECHNICAL_DATA_MM.pdf](**url**)

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-06-26:
Look in the wiki. Perhaps that helps you: https://tasmota.github.io/docs/Smart-Meter-Interface/

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ilgrank, 2020-06-26:
Thanks @kugelkopf123 , I had already checked it, and currently support includes:
ASCII OBIS / SML OBIS / Binary EBUS / Binary MODBUS / Binary RAW 
but does not include **IEC62056**, unless I'm missing something of course

kugelkopf123, 2020-06-26:
Oh sorry. Didn’t saw it. Perhaps you could ask for this feature on gemus fork https://github.com/gemu2015/Sonoff-Tasmota
He did the sml driver. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ilgrank, 2020-06-26:
@kugelkopf123 no problem :)
Didn't know of Gemus fork either.. but I don't see any change in his readme.md.. what is his fork about?

meingraham, 2020-06-26:
@gemu2015 is the developer of Tasmota scripting and the smart meter interface for Tasmota.

gemu2015, 2020-06-27:
@ilgrank 
while it is simply a copy and paste to install an existing SML driver descriptor, it is not easy for a beginner to design one for a new meter. You must invest some time and effort to explore your meters responses. from the above links i am optimistic that it is possible with my driver to read your meter.

how to start:

first compile a tasmota version with scripting and sml enabled.
then type the following basic script.
\>D
\>B
=>sensor53 r
\>M 1
; OBIS uses serial 7E1 here an GPIO pin 3 for receive with 300 baud
+1,3,o,300,OBIS
1,1-0:1.8.0*255(@1,Test,KWh,Total,4

then in tasmota console type:  sensor53 d1
if your meter emits data from itself you should now see the incoming data

if your meter needs a telegram to start you have to change the descriptor line e.g.
here with GPIO 1 for the transmit line
the driver sends this message every 3 seconds to the meter
+1,3,o,300,OBIS,1,30,0x2F,0x3F,0x21,0x0D,0x0A

at least now you should get an answer in console every 3 seconds.

if you get responses, post them here and we can proceed.

meingraham, 2020-06-27:
@ilgrank 

Once you are able to successfully capture the meter responses using the procedure @gemu2015 provided, it would be helpful to future users if you would add this procedure and any other "novice" explanation details you learn to the Tasmota docs - https://tasmota.github.io/docs/Smart-Meter-Interface/

kugelkopf123, 2020-06-29:
At the moment, the Dev Branch from tasmota is in sync with Gemus Fork. So you can try to compile the tasmota one.

meingraham, 2020-06-29:
You can also try to compile @gemu2015 fork using Gitpod. Replace the Tasmota repository URL with Gerhard's repository.

closed 2020-08-15 by stale[bot]

ldoench, 2022-12-19:
@gemu2015 I just found this tread and hope it is ok to revive it.
I have a Iskra AM550 with only P0/IR and there is very little information about it online. Only that it should work according to IEC 62056-21 standard as well as DLMS (IEC 62056-46).

The response I get at 9600 baud is not quite SML:
```
21:31:38.231 : 77 9b 12 42 6c 69 52 e0 0c fd 89 ca a4 8e b6 dc e8 5a 38 c9 ae 7f 58 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 
21:31:48.247 : 77 53 db 2f a1 52 bf 6a ad 71 0e 23 90 40 ed 11 4e 41 da ce 0f b0 44 e8 82 9d f2 81 00 32 2b 50 4c f8 85 e4 ff 05 43 a8 
21:31:49.273 : 77 68 27 8d 79 5b 5d bd 74 5a 91 46 91 86 ac c7 8c 35 ec 91 3d 63 de 18 90 e8 3d 1f 7b 67 2c c5 45 9e 93 5a a7 69 95 89 
21:31:53.246 : 77 de b7 46 ae 70 e1 50 4c 92 70 12 4a 65 1a 79 b3 e0 99 92 d6 b3 cc 3e 96 e2 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 
21:31:55.243 : 77 86 ce 64 43 99 15 a4 04 42 b1 7e ee 40 11 e4 40 89 8d b5 e6 4c a4 f1 e4 0a 0f cc 6d 2a ba ad 56 89 fe 96 c1 60 df c0 
21:31:56.277 : 77 89 3b 8b 30 0f 83 27 9f 10 de 2a 51 d3 ee 46 8d b2 8e 30 72 88 8d c3 35 76 b2 8f 7f a5 6b 96 d7 00 c4 ed 98 7a ce 52 
21:31:57.264 : 77 5d 88 6f 21 b0 89 7b de 49 71 c0 19 fd f7 50 ba 20 78 ea e8 83 ad 14 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 
21:31:58.239 : 77 b9 78 f4 1f 4c 24 04 64 d1 fd 1b 76 d0 04 dd fa 15 db 24 94 e3 67 1f 96 87 e8 e3 7c de 93 5f af 75 85 7e 7e a0 67 cf 
21:32:04.260 : 77 a3 33 24 4f 93 ff 3a b7 12 e6 2c aa 13 d9 4e df 7c 38 f3 cd f8 08 cd e7 a4 45 6d 1a 27 be 4d 03 39 01 e9 49 8e b4 84 
21:32:05.292 : 77 7f ff a5 34 c5 42 4d 20 57 17 9d f5 89 03 c2 62 21 fb d0 78 95 5b b4 cc d0 d4 d5 0c cc 4d 9b 3b d1 92 ff e1 8d c8 86 
21:32:07.221 : 77 35 ea 4d ae dd ee f8 c8 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 14 4d 76 28 ef 
21:32:09.246 : 77 0d 0b a4 e6 d4 af 33 b4 e4 2f 69 2a f1 f1 a9 da cb cf 37 9f b5 fc 56 ea 6d ba f6 f1 41 ae 4d 2f 11 cc f6 69 f8 ae 8f 
21:32:14.274 : 77 d6 9e 6e 88 eb d2 75 97 e9 93 05 5a 0b 2e 8e 33 c9 bd 26 41 db d6 cc 84 86 63 f0 b0 85 23 93 dc c2 1b db 48 e7 c3 1a 
21:32:15.256 : 77 17 95 2f 52 99 83 2f 01 f5 ff 72 0e 78 73 93 ab 91 49 70 44 c9 6b 74 b0 9d 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 
21:32:19.264 : 77 e8 21 5c c0 ed 8f 92 02 1a f0 19 a4 2e e4 3b 86 f4 cc ad 76 51 99 34 a7 7e 58 51 7f a9 32 96 53 ab 6f eb 6d 9c 25 f8 
21:32:20.251 : 77 1a 6d be c1 dd 8b 6a eb 47 bb cc ee 59 96 e9 2f e5 17 27 e5 fb 6e 85 7a eb 1d 1e 76 7e 7e a0 67 cf 02 23 13 fb f1 e6 
21:32:28.272 : 77 1e a8 57 21 c5 fe 45 4a 4e 4d 91 16 7d 93 4f 81 3b fc 2b 86 bc 91 52 63 cf 51 35 ea 31 52 8b b3 c7 02 41 49 e5 b0 73 
21:32:38.284 : 77 dd 62 81 0b ad 4f 1d a2 11 c0 5c 5d 57 0a 2f 30 4d 80 2c cc 1f e1 fe 08 04 71 bc 02 7e 7e a0 67 cf 02 23 13 fb f1 e6 
21:32:42.228 : 77 55 da d8 bc 41 15 7b e1 3c 42 3d 65 31 a4 f7 ad 54 7e 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 
```

Could it be somehow encrypted? Since the only Information, that I got from my grid operator was a 32 bit AES key.

gemu2015, 2022-12-19:
yes this looks like its encrypted. there is currently no support for encrypted meters.
if you happen to find an implementation (source code) of a decryptor for that meter i might have a look at it.

ldoench, 2022-12-19:
Thanks for your response. 
I found this python script, that promises to decrypt the messages: https://gist.github.com/pocki80/941fa090a8d6269a9b3b68c195f8750f
 
Strangely my received messages don't start with 7E as they are in the example.

gemu2015, 2022-12-19:
this will not work. you must find a software that decodes your meter thus must have the same start character.

ldoench, 2022-12-21:
Pocki80 just mentioned, that it seems to be the right data, but by selecting SML in tasmota it added the 77 at the beginning. By selecting raw i was able to get the 7E A0 ... 7E packages:

```
09:32:20.360 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 30 c4 74 eb 7a a7 3c 74 22 81 57 81 74 
09:32:20.402 : b7 44 32 17 b1 5b 70 ec bb 7a 6f dc 78 de 2e 64 9f f4 e0 c1 d7 ac 19 1d 56 44 8e 96 d2 0c f7 19 cf f8 f0 a9 5a 59 7b e7 
09:32:20.445 : 7f 6c 59 6b c7 b6 72 43 38 9b 61 da ad 01 09 06 60 73 3d 7c 40 f3 2d 39 8b 7e 
09:32:23.429 : 12 fb 80 bf f6 65 65 b7 b4 cd 66 bb 0a 53 fc b5 96 18 a3 48 59 4e 8d 61 3e e7 c6 90 0c e3 ea 38 77 76 ad e9 89 79 a0 cf 
09:32:23.471 : cc 33 4a 7e 
09:32:24.349 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 
09:32:24.392 : 20 00 1e 30 c8 6a 2e 27 45 2d 31 8c 86 5d e3 a9 41 d7 b0 63 2f fd e5 a5 f9 1b d9 88 32 c9 cc a7 85 ef 1e 6d b0 e7 17 4f 70 
09:32:24.434 : 13 5b 07 5c bb f3 f1 86 b2 4e 60 ac 27 49 59 f8 05 a0 ab c0 a8 ea 26 53 49 42 e2 03 ed b5 15 c3 b4 d4 09 0e 2d 54 b2 ce 7e 
09:32:25.332 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 
09:32:25.375 : db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 30 c9 73 db b0 dd 59 03 64 87 46 d9 09 21 26 0c 46 42 73 0e 87 a0 30 9e b9 1d fb 
09:32:25.417 : e6 7c 56 3a a7 f1 bc 34 f0 d3 9a 8b 72 34 78 94 89 fc be c9 a0 09 e4 72 fb 66 72 6f e4 8a 95 f1 41 b1 30 7d 13 ec 03 fa 
09:32:25.461 : 02 31 c5 5d 4e 9e bf ff ee 61 f3 7e 
09:32:27.436 : a1 5a ad 6c f7 58 2a 8a cd b9 cc 9e 46 18 80 a5 ec 2e 52 d8 2e 90 9b a3 b3 8e 62 b1 5c 16 5b e9 a0 63 77 6a db 84 7e 
```

Would it be possible to align them by this pattern and then decode it with the algorithm from the python script?

gemu2015, 2022-12-21:
i normally only add things i use myself and don't add things on user request which are a lot of work.
i will have a look at this data when i find some time probably in christmas holidays 
however you must post your KEY to test the decoding and possibly a larger data set.

ldoench, 2022-12-21:
I would really appreciate it, if you'll take a look. I also found another [thread](https://github.com/arendst/Tasmota/discussions/14610), where someone had the same problem with an encrypted smart meter.

```
13:18:17.309 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 
13:18:17.353 : 20 00 1e 65 b9 da 54 eb a6 b5 bf 51 2a 7a a5 1d 85 5a 7c c9 16 24 de 22 b3 39 08 9f cb 2e e8 bc ab 83 b8 ba cb 2b 10 15 6e 7a 
13:18:17.397 : 13 62 18 50 2e 14 d9 bc cb 5f 91 0d a0 65 8a dc 08 3e 81 ae 42 fa 1e 51 a1 f8 ed 3d d3 e1 6f bc 3d cc 8a ae f7 75 9e 7e 
13:18:18.298 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 
13:18:18.340 : 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 ba 59 28 4c ef 11 f5 d3 98 e7 a5 e9 56 fe 84 bb 19 e1 a8 e3 6c e4 25 1e 2e 9f 
13:18:18.384 : 74 9e fe 1f 8d f8 2d ba 22 2d 86 13 37 67 5c 27 81 d1 91 5a e6 73 57 f2 2b 59 d3 6a 24 5a 6a 19 c8 93 6f 10 0e 8a 03 ca 4f 27 
13:18:18.427 : 94 3f 4f 58 93 6a a1 50 37 7e 
13:18:19.317 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 bb b9 a1 5b 72 1a 07 f0 85 b3 17 c7 da 
13:18:19.360 : 39 8b 97 e6 0e bb 46 39 e4 0c 55 7d 84 81 81 ae 3e aa 93 27 2d c3 3f ca 29 00 3b f9 19 d3 66 f5 f1 ae a8 a8 15 be ce c5 b9 
13:18:19.403 : 63 87 1e 77 ad 65 33 b1 59 2d 64 5a a7 62 49 2f 6e b1 5c 77 51 d3 55 7e 
13:18:20.320 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 bc 97 4f 32 86 f7 81 71 c3 1d 
13:18:20.363 : 54 b7 67 20 eb 7d f8 15 03 37 3e e3 17 94 7c 53 85 d3 6c 3b b4 f1 a2 56 7d 15 2f aa 5b 8e 01 1a f0 5d e7 5d 11 34 28 da 36 
13:18:20.405 : 8f 1f d5 b5 48 ae b9 13 42 c2 07 0b 93 6f 7f e3 f0 52 c6 ec f1 16 11 6c 25 8d 7e 
13:18:21.300 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 
13:18:21.343 : 08 4f 20 00 1e 65 bd a5 23 5a 31 a3 53 28 b7 78 56 71 04 38 01 ef b9 1e 7d f1 62 34 ff 44 aa 62 d1 d8 5e 08 c9 74 bf 08 f5 9a 
13:18:21.386 : b3 70 91 33 9f 58 75 5e fe 2f 32 4e ba 8e 8d 34 e6 4a 64 7f 18 44 40 47 ac b2 ab 24 1c 49 cc 76 c1 d5 b0 4b af 9e 8d 2f 8f 
13:18:21.428 : 7e 
13:18:22.309 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 be 
13:18:22.352 : c9 68 1f e7 e1 f4 99 93 d4 d5 ad 3c 1b aa 3e 01 8f a1 26 57 3d fd d4 03 e1 47 00 d1 8b db a5 c1 48 9a aa ee fd ed ab 4a 86 a2 
13:18:22.395 : 6a da 54 0d 5f 1a fe 6a a9 8d 99 fb a6 84 41 10 5a 3c c2 c2 96 df 9e ad 4c 27 39 c8 37 7b 51 f7 9b 74 7e 
13:18:23.302 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 
13:18:23.345 : db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 bf 3a e8 d4 8c 79 ae 07 bc 88 cb 13 b2 57 35 4f a4 8c 93 bd 3b bd b4 b5 3f f8 
13:18:23.388 : 81 59 9e a1 bc b5 46 b0 bb 14 33 0a d2 e5 ba 11 c6 52 ee 51 46 a3 e5 f8 65 41 57 5e 1e 3c 72 9a 32 3a 48 a3 04 c3 bf ee 26 b5 
13:18:23.430 : a1 6d 72 56 9f cd 28 0c 1a 7e 
13:18:24.317 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 c0 11 3b 7b fa 63 e6 ec f2 df e3 24 31 
13:18:24.360 : f3 9a 37 62 9d 6f af b9 37 13 89 3d 30 f4 70 cc be 27 1e 2f 31 6b ef 56 ee 9f d5 16 0e 30 eb 6a 05 8d b7 7b 1a a7 b7 3f 03 
13:18:24.402 : 8c bf 2f de 84 20 c9 7b 48 5c 00 99 ff 69 5a c0 40 66 47 c7 96 5d 6f 7e 
13:18:25.283 : 7e a0 67 
13:18:25.325 : cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 c1 21 0c c4 5c 76 7a a7 3e 43 ee 56 9f 4c f6 29 a3 
13:18:25.369 : 7d 9e ab 0c 42 17 3d 4c 97 93 8b 8c 75 ab f6 b4 97 7d b5 b2 f5 74 bc 85 7d 56 42 0c bb 73 20 44 71 65 2b 48 fc 04 88 ce cc 4a 
13:18:25.411 : bc fb 51 41 14 ea e4 96 fa 45 7c cd 39 22 16 55 1a ff 7e 
13:18:26.310 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 c2 d7 ca 
13:18:26.353 : e0 ef 68 a7 a2 91 15 d2 e2 fe 8d f9 05 0f e1 9f e3 bf ea 32 2c b6 af 5e 62 1b 6d b0 b2 73 78 65 0c a9 98 a2 c9 cb d2 8a ce 
13:18:26.396 : 0d 8c c2 e8 67 5d 11 91 b8 50 2f c6 51 ec 39 bb c7 0f 5a 3b 50 7b 07 d5 e0 7b 45 1e d7 f9 d8 b8 5b 7e 
13:18:27.301 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 
13:18:27.343 : 68 74 99 f5 08 4f 20 00 1e 65 c3 12 a7 df 8b 81 ca 58 a0 62 07 5e 07 9a 39 51 2e 40 9c ce d4 dc 62 76 3c 20 f7 05 53 d2 48 
13:18:27.387 : 8d da b1 61 9e ea 0c bd 85 7a d4 8c 73 5e 23 11 e4 2f 37 10 b3 66 dd ff 58 08 5c 49 c2 ea 48 ef 67 12 88 c1 27 01 3c ed 36 af 
13:18:27.429 : 73 8f 69 51 7e 
13:18:30.409 : 72 f7 95 b2 fe 49 ef 9d 1c 17 12 45 13 c3 35 e1 43 54 b2 1e 20 bb 7e 
13:18:31.323 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 c7 a9 af 9c 7d c9 0d ad 89 60 b1 2f 
13:18:31.366 : 3a f1 8c 3a 2c d9 3b 9d 36 ba cc 39 35 29 4f 0d 29 56 69 4c ac 19 57 f4 93 3b b8 d1 f3 aa 4b 46 02 29 08 ae c8 3a b2 26 e8 c6 
13:18:31.408 : 3b 8e 53 9c 7e e3 59 dc 1a 2e 2d c4 cc bd 96 7c 86 b9 96 a4 4a 48 79 7e 
13:18:32.312 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 65 c8 3b 
13:18:32.354 : b6 15 ed 5f ce 2b 55 1f a4 60 6c 38 4f 29 76 be 18 eb c4 84 f9 b4 d1 bb 20 ea d2 b4 4c 74 f7 29 62 78 b2 04 f5 c2 b1 9a f6 
13:18:32.396 : d6 16 c1 dc 7e d7 e6 ad 55 41 3d c3 6c fb 4d a4 ef 32 60 90 67 9e f9 f9 c4 03 47 ea 51 cd 49 f9 58 d2 7e 
13:18:33.301 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 
13:18:33.343 : 4b 68 74 99 f5 08 4f 20 00 1e 65 c9 26 60 5b 7c 90 9d 5b bc 9a e5 de fe c7 2c ed 5c 52 43 a2 28 61 95 90 75 d9 80 c6 28 6e 
13:18:33.386 : 0d f6 bf 7f 4e a5 40 10 84 08 61 ab c5 61 d4 ca cb 6e 61 41 92 f8 81 a7 5c ba 85 8d 23 c3 0b 04 26 78 0c ec a3 44 f7 1a dc 
13:18:33.428 : 99 af e8 70 ae b4 7e 
[... cut ...]
13:19:44.386 : c1 68 63 58 dd c4 e1 b9 74 6d bd 91 03 7c be 9c ff 5c 8e c1 71 4e f3 7c c6 9d 2f f5 65 68 e9 8a db 19 9e 08 11 9f 21 ce 31 
13:19:44.429 : c0 55 46 01 be c9 7e 
13:19:45.314 : 7e a0 67 cf 02 23 13 fb f1 e6 e7 00 db 08 49 53 4b 68 74 99 f5 08 4f 20 00 1e 66 11 b2 
13:19:45.358 : 7d a1 66 83 94 6e 48 f1 04 73 ce db 0a 58 10 b7 5a df 24 31 c7 58 28 18 a0 94 0f 79 6d f5 0a 5a 76 43 1a 5b 28 63 fa 44 e6 8e 
13:19:45.401 : 66 55 7b 23 54 64 b0 24 3b 24 f8 5d f7 72 7b 96 41 13 54 45 59 fc 8c 83 c0 68 69 28 4c 6e 19 5d 51 7e 
```

The key is 69C3B089DC1EE19E5981444085447123

gemu2015, 2022-12-26:
i tested your data with the python code and it delivers strange results:

{"datetime": "2022-12-21T13:18:17", "L1": {"v": 3079, "a": 2.55}, "L2": {"v": 3093, "a": 0.06}, "L3": {"v": 3346, "a": 7.97}, "actual": {"in": 100663296, "out": 100663297}, "total": {"in": 100663744, "out": 100663296}}

as long as i don't have a working code source with reasonable result i can not port that to tasmota

gemu2015, 2022-12-26:
this seems to be a better result:
Output:    204.058kWh,      0.000kWh,      0.348kvarh,    114.898kvarh,   149W,     0W,     0var,   123var at 21.12.2022-13:18:17

i will have the driver ready in the next days.

speedymk1, 2022-12-26:
I have also a crypted meter and got the key. How can I assist?

gemu2015, 2022-12-27:
@ldoench

driver working

decrypted dump from sensor53 d1

0f 00 1e 65 ba 0c 07 e6 0c 15 03 0d 12 11 00 ff 
c4 00 02 09 09 0c 07 e6 0c 15 03 0d 12 11 00 ff 
c4 00 06 00 03 1d 1a 06 00 00 00 00 06 00 00 01 
5c 06 00 01 c0 d2 06 00 00 00 95 06 00 00 00 00 
06 00 00 00 00 06 00 00 00 7b 

from python script these should be your offsets:
        a=bytes_to_int(s[35:39])/1000.000  #+A Wh
        b=bytes_to_int(s[40:44])/1000.000  #-A Wh
        c=bytes_to_int(s[45:49])/1000.000  #+R varh
        d=bytes_to_int(s[50:54])/1000.000  #-R varh
        e=bytes_to_int(s[55:59])           #+P W
        f=bytes_to_int(s[60:64])           #-P W
        g=bytes_to_int(s[65:69])           #+Q var
        h=bytes_to_int(s[70:74])           #-Q var
        yyyy=bytes_to_int(s[22:24])
        mm=bytes_to_int(s[24:25])
        dd=bytes_to_int(s[25:26])
        hh=bytes_to_int(s[27:28])
        mi=bytes_to_int(s[28:29])
        ss=bytes_to_int(s[29:30])
        print ("Output: %10.3fkWh, %10.3fkWh, %10.3fkvarh, %10.3fkvarh, %5dW, %5dW, %5dvar, %5dvar at %02d.%02d.%04d-%02d:%02d:%02d" %(a,b,c,d,e,f,g,h, dd,mm,yyyy,hh,mi,ss))
		
		Output:    204.058kWh,      0.000kWh,      0.348kvarh,    114.898kvarh,   149W,     0W,     0var,   123var at 21.12.2022-13:18:17

our offsets are 4 bytes less because of trailer => 0f001e65 taken from first 4 bytes of dump

this is your initial descriptor just decoding  year and value a from above

```
>D

>B
=>sensor53 r

>M 1
+1,3,r,0,9600,CRYPT
1,=so4,69C3B089DC1EE19E5981444085447123
1,0f001e65x18UUuu@1,year,,year,0
1,0f001e65x31UUuuUUuu@1000,A,kWh,A,3
#
```

=so4 => your key

@speedymk1 
only 7ea0 frames are supported, make a dump of your meters output to check if your data is compatible

offsets to values are specific to each meter

needs #define USE_SML_DECRYPT

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/10306563/xsns_53_sml.ino.zip)

australien340, 2023-01-21:
Hello all!

@gemu2015 did now the encryption, great!
Is there a possibility to download an allready compiled tasmota file for the ESP32? Or do I have to do this at my own?

barbudor, 2023-01-21:
There's no prebuilt with USE_SM_DECRYPT
But that could be added in tasmota32-scripting "unofficial" binary : https://github.com/Jason2866/Tasmota-specials/tree/firmware/firmware/tasmota32/other

@gemu2015 May be it could be interresting to PR to Jason's Specials so that tasmota*-scripting includes everuthing needed for SML ?
The user_config_override is there: https://github.com/Jason2866/Tasmota-build/blob/684b9f47d2f009365b92ca1fbe6eca79e6baa3df/tasmota/user_config_override.h#L1332-L1678

gemu2015, 2023-01-21:
meanwhile i reverted most compile flags to its opposite
so nearly all options are default ON now
like
#ifndef NO_USE_SML_DECRYPT
// allows 256 bit AES decryption
#define USE_SML_DECRYPT
#endif

some other define options were moved to descriptor variables.

see docs

as far as i know Jasons scripting build includes the SML driver

gemu2015, 2023-01-21:
@australien340 
keep in mind that currently only HDLC frames (7e A0) are supported.
i am working on MBUS (68 01 01 68) frames and others.

gismo2004, 2023-01-22:
@gemu2015 
I have a "7e A0" HDLC frame providing SmartMeter. But it seems that either I am too stupid to set it up correctly, or my data is different to the others here. 
It seems like my data is split into three blocks. When putting the raw data into Gurux DLMS Translator the values are extracted correctly.  [smartmeterData.txt](https://github.com/arendst/Tasmota/files/10474269/smartmeterData.txt)
It would be highly appreciated if you could guide me on how to proceed. 

`>D

>B
=>sensor53 r

>M 1
+1,17,r,0,2500,CRYPT
1,=so3,256
1,=so4,C4E1450D4AC04E0BD0F4468C660ED2BD
1,1-0:1.8.0*255(@1000,consumption,KWh,Total_in,3)
#`

gemu2015, 2023-01-22:
sorry no segment support yet. 
do you know this very nice project : https://github.com/gskjold/AmsToMqttBridge
i am currently porting parsers from this library but don't expect it very soon.

try if it works with AMS Bridge, if it does it has a chance it will run with future sml driver too.

gismo2004, 2023-01-22:
@gemu2015 AMS Bridge is working "plug&play" :-) THX for the hint. If you need something where I can help to test, please let me know! I still prefer to use Tasmota, since I have two meters and it would be cool to have just one ESP running.

gemu2015, 2023-01-25:
@gismo2004 
i have now ported AMS Bridge parsers to tasmota. it decodes among others your stream data from above.
however the concept of the SML driver is a bit different than AMS Bridge.
you only get the decrypted block and have to construct  a decoder list for your meter.

adapt the descriptor with your decryption key and meter serial baud rate etc

the current version still contains a lot of debug console messages
in dump mode (sensor53 d1) you should then get raw decrypted blocks
copy one of them to a text editor.

like this
```
08:21:33.981 SML: Received valid DLMS at 45
08:21:33.982 SML: Using application data:
08:21:33.983 >> decrypted block: 224 bytes
08:21:33.984 :>02 21 09 0c 07 e7 01 16 07 0c 0d 0a ff 80 00 00 
08:21:33.985 :>09 06 01 00 01 08 00 ff 06 00 89 0c 51 09 06 01 
08:21:33.986 :>00 01 08 01 ff 06 00 68 cd 77 09 06 01 00 01 08 
08:21:33.987 :>02 ff 06 00 20 3e da 09 06 01 00 01 07 00 ff 06 
08:21:33.987 :>00 00 01 c1 09 06 01 00 02 08 00 ff 06 00 00 00 
08:21:33.988 :>00 09 06 01 00 02 08 01 ff 06 00 00 00 00 09 06 
08:21:33.989 :>01 00 02 08 02 ff 06 00 00 00 00 09 06 01 00 02 
08:21:33.990 :>07 00 ff 06 00 00 00 00 09 06 01 00 03 08 00 ff 
08:21:33.991 :>06 00 00 4a 9a 09 06 01 00 03 08 01 ff 06 00 00 
08:21:33.992 :>4a 1f 09 06 01 00 03 08 02 ff 06 00 00 00 7b 09 
08:21:33.993 :>06 01 00 03 07 00 ff 06 00 00 00 00 09 06 01 00 
08:21:33.994 :>04 08 00 ff 06 00 2c b9 b7 09 06 01 00 04 08 01 
08:21:33.995 :>ff 06 00 1e 7e e2 09 06 01 00 04 08 02 ff 06 00 
08:21:33.995 :>0e 3a d5 09 06 01 00 04 07 00 ff 06 00 00 00 db 
08:21:34.030 :>0e 3a d5 09 06 01 00 04 07 00 ff 06 00 00 00 db 
```

now you can determine the entries of your meters data.

find a unique byte sequence at the beginning of the decrypted dump
then determine the offset to your desired value (x30 in this fictive example)
then give the decode sequence unsigned 32 bit here (see sml docs)

like this:

1,0221x30UUuuUUuu@1000,total,kWh,total,3

```
>D
>B
=>sensor53 r

>M 1
+1,3,r,0,9600,CRYPT
1,=so3,512
1,=so4,69C3B089DC1EE19E5981444085447123
1,0221x30UUuuUUuu@1000,A,kWh,A,3
#
```

this is work in progress, testers welcome

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/10497093/xsns_53_sml.ino.zip)

gemu2015, 2023-01-25:
put the lib into /lib/div
[ams.zip](https://github.com/arendst/Tasmota/files/10497150/ams.zip)

Noschvie, 2023-01-25:
> put the lib into /lib/div [ams.zip](https://github.com/arendst/Tasmota/files/10497150/ams.zip)

will test this soon...

gismo2004, 2023-01-25:
@gemu2015 AWSOME!!! --> works out of the box.

Since all this L&G E450 should provide the same output, I will work on decoding all provided messages, and then we can add this to the WIKI I think.

gismo2004, 2023-01-25:
@gemu2015 Reading data works with the following code. The question to you: why can't we use OBIS codes with this approach?
Data is definitely OBIS

```
>D

>B
=>sensor53 r

>M 1
+1,17,r,0,2500,Meter_A
1,=so3,512
1,=so4,GUEK
1,=so5,GAK
1,0221x23UUuuUUuu@1000,1.0.1.8.0.255,kWh,1.0.1.8.0.255,3
1,0221x36UUuuUUuu@1000,1.0.1.8.1.255,kWh,1.0.1.8.1.255,3
1,0221x49UUuuUUuu@1000,1.0.1.8.2.255,kWh,1.0.1.8.2.255,3
1,0221x62UUuuUUuu@1000,1.0.1.7.0.255,kW,1.0.1.7.0.255,3
1,0221x75UUuuUUuu@1000,1.0.2.8.0.255,kWh,1.0.2.8.0.255,3
1,0221x88UUuuUUuu@1000,1.0.2.8.1.255,kWh,1.0.2.8.1.255,3
1,0221x101UUuuUUuu@1000,1.0.2.8.2.255,kWh,1.0.2.8.2.255,3
1,0221x114UUuuUUuu@1000,1.0.2.7.0.255,kW,1.0.2.7.0.255,3
1,0221x127UUuuUUuu@1000,1.0.3.8.0.255,kvarh,1.0.3.8.0.255,3
1,0221x140UUuuUUuu@1000,1.0.3.8.1.255,kvarh,1.0.3.8.1.255,3
1,0221x153UUuuUUuu@1000,1.0.3.8.2.255,kvarh,1.0.3.8.2.255,3
1,0221x166UUuuUUuu@1000,1.0.3.7.0.255,kvar,1.0.3.7.0.255,3
1,0221x179UUuuUUuu@1000,1.0.4.8.0.255,kvarh,1.0.4.8.0.255,3
1,0221x192UUuuUUuu@1000,1.0.4.8.1.255,kvarh,1.0.4.8.1.255,3
1,0221x205UUuuUUuu@1000,1.0.4.8.2.255,kvarh,1.0.4.8.2.255,3
1,0221x218UUuuUUuu@1000,1.0.4.7.0.255,kvar,1.0.4.7.0.255,3
#
```

gemu2015, 2023-01-25:
Currently there is no pattern match in crypto mode
I will check if this can be implemented

gemu2015, 2023-01-25:
ok, now with pattern match:

1,pm(00010801ff)UUuuUUuu@1000,1.0.1.8.0.255,kWh,1.0.1.8.0.255,3

[Archiv.zip](https://github.com/arendst/Tasmota/files/10498661/Archiv.zip)

gismo2004, 2023-01-25:
@gemu2015 
Seems like the thing with AuthKey is broken now because i get: “Decrypt authentication failed”.

gemu2015, 2023-01-25:
Do you need an auth. key ?
Since most meters don’t need it it is default off
#define USE_SML_AUTHKEY

gemu2015, 2023-01-25:
try removing the auth definition in descriptor

gemu2015, 2023-01-25:
the first version above didn't support auth, the last version does, so it seems your meter either doesn't support auth or your auth is wrong

gismo2004, 2023-01-25:
That explains why it worked before. Now that I have deactivated USE_SML_AUTHKEY it's working as expected. I was not aware of that it is working without GAK --> in AMS bridge I have added the key, and it worked (also working without it after this discussion :-) ) that's why I have added the option to use it. 

Following config, is working:

```
>D

>B
smlj=0
=>sensor53 r
>R
smlj=0
>S
if upsecs>22
then
smlj|=1
endif
>M 1
+1,17,r,0,2500,Meter_A
1,=so3,512
1,=so4,GUEK
1,pm(0100010800FF06)UUuuUUuu@1000,1.0.1.8.0.255,kWh,1.0.1.8.0.255,3
1,pm(0100010801FF06)UUuuUUuu@1000,1.0.1.8.1.255,kWh,1.0.1.8.1.255,3
1,pm(0100010802FF06)UUuuUUuu@1000,1.0.1.8.2.255,kWh,1.0.1.8.2.255,3
1,pm(0100010700FF06)UUuuUUuu@1000,1.0.1.7.0.255,kW,1.0.1.7.0.255,3
1,pm(0100020800FF06)UUuuUUuu@1000,1.0.2.8.0.255,kWh,1.0.2.8.0.255,3
1,pm(0100020801FF06)UUuuUUuu@1000,1.0.2.8.1.255,kWh,1.0.2.8.1.255,3
1,pm(0100020802FF06)UUuuUUuu@1000,1.0.2.8.2.255,kWh,1.0.2.8.2.255,3
1,pm(0100020700FF06)UUuuUUuu@1000,1.0.2.7.0.255,kW,1.0.2.7.0.255,3
1,pm(0100030800FF06)UUuuUUuu@1000,1.0.3.8.0.255,kvarh,1.0.3.8.0.255,3
1,pm(0100030801FF06)UUuuUUuu@1000,1.0.3.8.1.255,kvarh,1.0.3.8.1.255,3
1,pm(0100030802FF06)UUuuUUuu@1000,1.0.3.8.2.255,kvarh,1.0.3.8.2.255,3
1,pm(0100030700FF06)UUuuUUuu@1000,1.0.3.7.0.255,kvar,1.0.3.7.0.255,3
1,pm(0100040800FF06)UUuuUUuu@1000,1.0.4.8.0.255,kvarh,1.0.4.8.0.255,3
1,pm(0100040801FF06)UUuuUUuu@1000,1.0.4.8.1.255,kvarh,1.0.4.8.1.255,3
1,pm(0100040802FF06)UUuuUUuu@1000,1.0.4.8.2.255,kvarh,1.0.4.8.2.255,3
1,pm(0100040700FF06)UUuuUUuu@1000,1.0.4.7.0.255,kvar,1.0.4.7.0.255,3
#
```

Maybe off-topic: is it a configuration thing to get a unit for a sensor signal over MQTT, or is this not supported by tasmota?

gemu2015, 2023-01-25:
normally units are not transferred via MQTT with exception of Units that are user defined as temperature like Celsius versus Fahrenheit.
btw you may reduce the pattern match to less info if the remaining bytes clearly define the target.
so removing 0100 would probably be OK.
1,pm(010800FF06)

gismo2004, 2023-01-25:
Of course, will let it run with following config:

```
>D

>B
smlj=0
=>sensor53 r
>R
smlj=0
>S
if upsecs>22
then
smlj|=1
endif
>M 2
+1,17,r,0,2500,Heizung
1,=so3,512
1,=so4,GUEK
1,pm(010800FF06)UUuuUUuu@1000,1-0:1.8.0,kWh,  1-0:1.8.0_kWh,3 ;Wirkenergie Lieferung (+A) **
1,pm(010801FF06)UUuuUUuu@1000,1-0:1.8.1,kWh,  1-0:1.8.1_kWh,3 ;Wirkenergie Lieferung (+A) Tarif 1
1,pm(010802FF06)UUuuUUuu@1000,1-0:1.8.2,kWh,  1-0:1.8.2_kWh,3 ;Wirkenergie Lieferung (+A) Tarif 2
1,pm(010700FF06)UUuuUUuu@1000,1-0:1.7.0,kW,   1-0:1.7.0_kW,3 ;Momentane Wirkleistung Bezug (+A)
;1,pm(020800FF06)UUuuUUuu@1000,1-0:2.8.0,kWh,  1-0:2.8.0_kWh,3 ;Wirkenergie Bezug (-A) ***
;1,pm(020801FF06)UUuuUUuu@1000,1-0:2.8.1,kWh,  1-0:2.8.1_kWh,3 ;Wirkenergie Bezug (-A) Tarif 1
;1,pm(020802FF06)UUuuUUuu@1000,1-0:2.8.2,kWh,  1-0:2.8.2_kWh,3 ;Wirkenergie Bezug (-A) Tarif 2
;1,pm(020700FF06)UUuuUUuu@1000,1-0:2.7.0,kW,   1-0:2.7.0_kW,3 ;Momentane Wirkleistung Lieferung (-A)
1,pm(030800FF06)UUuuUUuu@1000,1-0:3.8.0,kvarh,1-0:3.8.0,3_kvarh ;Blindenergie Lieferung (+R)
1,pm(030801FF06)UUuuUUuu@1000,1-0:3.8.1,kvarh,1-0:3.8.1,3_kvarh ;Blindenergie Lieferung (+R) Tarif 1
1,pm(030802FF06)UUuuUUuu@1000,1-0:3.8.2,kvarh,1-0:3.8.2,3_kvarh ;Blindenergie Lieferung (+R) Tarif 2
1,pm(030700FF06)UUuuUUuu@1000,1-0:3.7.0,kvar, 1-0:3.7.0,3_kvar ;Momentane Blindleistung Bezug (+R)
1,pm(040800FF06)UUuuUUuu@1000,1-0:4.8.0,kvarh,1-0:4.8.0,3_kvarh ;Blindenergie Bezug (-R)
1,pm(040801FF06)UUuuUUuu@1000,1-0:4.8.1,kvarh,1-0:4.8.1,3_kvarh ;Blindenergie Bezug (-R) Tarif 1
1,pm(040802FF06)UUuuUUuu@1000,1-0:4.8.2,kvarh,1-0:4.8.2,3_kvarh ;Blindenergie Bezug (-R) Tarif 2
1,pm(040700FF06)UUuuUUuu@1000,1-0:4.7.0,kvar, 1-0:4.7.0,3_kvar ;Momentane Blindleistung Lieferung (-R)

+2,16,r,0,2500,Haus
2,=so3,512
2,=so4,GUEK
2,pm(010800FF06)UUuuUUuu@1000,1-0:1.8.0,kWh,  1-0:1.8.0_kWh,3 ;Wirkenergie Lieferung (+A) **
2,pm(010801FF06)UUuuUUuu@1000,1-0:1.8.1,kWh,  1-0:1.8.1_kWh,3 ;Wirkenergie Lieferung (+A) Tarif 1
2,pm(010802FF06)UUuuUUuu@1000,1-0:1.8.2,kWh,  1-0:1.8.2_kWh,3 ;Wirkenergie Lieferung (+A) Tarif 2
2,pm(010700FF06)UUuuUUuu@1000,1-0:1.7.0,kW,   1-0:1.7.0_kW,3 ;Momentane Wirkleistung Bezug (+A)
;2,pm(020800FF06)UUuuUUuu@1000,1-0:2.8.0,kWh,  1-0:2.8.0_kWh,3 ;Wirkenergie Bezug (-A) ***
;2,pm(020801FF06)UUuuUUuu@1000,1-0:2.8.1,kWh,  1-0:2.8.1_kWh,3 ;Wirkenergie Bezug (-A) Tarif 1
;2,pm(020802FF06)UUuuUUuu@1000,1-0:2.8.2,kWh,  1-0:2.8.2_kWh,3 ;Wirkenergie Bezug (-A) Tarif 2
;2,pm(020700FF06)UUuuUUuu@1000,1-0:2.7.0,kW,   1-0:2.7.0_kW,3 ;Momentane Wirkleistung Lieferung (-A)
2,pm(030800FF06)UUuuUUuu@1000,1-0:3.8.0,kvarh,1-0:3.8.0,3_kvarh ;Blindenergie Lieferung (+R)
2,pm(030801FF06)UUuuUUuu@1000,1-0:3.8.1,kvarh,1-0:3.8.1,3_kvarh ;Blindenergie Lieferung (+R) Tarif 1
2,pm(030802FF06)UUuuUUuu@1000,1-0:3.8.2,kvarh,1-0:3.8.2,3_kvarh ;Blindenergie Lieferung (+R) Tarif 2
2,pm(030700FF06)UUuuUUuu@1000,1-0:3.7.0,kvar, 1-0:3.7.0,3_kvar ;Momentane Blindleistung Bezug (+R)
2,pm(040800FF06)UUuuUUuu@1000,1-0:4.8.0,kvarh,1-0:4.8.0,3_kvarh ;Blindenergie Bezug (-R)
2,pm(040801FF06)UUuuUUuu@1000,1-0:4.8.1,kvarh,1-0:4.8.1,3_kvarh ;Blindenergie Bezug (-R) Tarif 1
2,pm(040802FF06)UUuuUUuu@1000,1-0:4.8.2,kvarh,1-0:4.8.2,3_kvarh ;Blindenergie Bezug (-R) Tarif 2
2,pm(040700FF06)UUuuUUuu@1000,1-0:4.7.0,kvar, 1-0:4.7.0,3_kvar ;Momentane Blindleistung Lieferung (-R)
#

;Landis+Gyr E450
;https://www.e-netze.at/downloads-data/pdf.aspx?pdf=EN_Update%20Kundenschnittstelle%20Smart%20Meter_ID3282_WEB_RGB.pdf
```

Also, it will be integrated in HomeAssistant, to check MQTT stability

Thank you very much for making this possible :-)

gemu2015, 2023-01-25:
I wonder why your baud rate is 2500 ?
Standard would be 2400

gismo2004, 2023-01-25:
> I wonder why your baud rate is 2500 ? Standard would be 2400

Wtf, me too... I have set this while testing and obviously never changed back.

australien340, 2023-01-25:
This sounds good to me as an beginner. 

So can I hope for a ready fw. 

@gismo2004 can you help me personally?

gemu2015, 2023-01-26:
one step further, now obis auto types (currently numbers only)
put a 't' in front of the matching bytes

example:
1,pm(t010800FF)@1000,kWh_IN,kWh,kWh_IN,3;Wirkenergie Lieferung (+A)

[Archiv.zip](https://github.com/arendst/Tasmota/files/10508543/Archiv.zip)

gismo2004, 2023-01-29:
@gemu2015 
Great work! Works as expected and is definitely much better to read/setup! Thank you again (sorry for the delayed feedback!)

gemu2015, 2023-01-29:
meanwhile i changed syntax to pure obis code
1,pm(1.8.1)@1000,kWh_IN,kWh,kWh_IN,3;Wirkenergie Lieferung (+AFF)

i am still waiting for feedback from some users before doing a pr

so fetch latest version here:

https://github.com/gemu2015/Sonoff-Tasmota

gismo2004, 2023-01-29:
Awesome! 

Current Config:

```
>D

>B
smlj=0
=>sensor53 r
>R
smlj=0
>S
if upsecs>22
then
smlj|=1
endif
>M 2
+1,17,r,0,2400,Heizung
1,=so3,512
1,=GUEK
1,pm(1.8.0)@1000,kWh_IN,kWh,kWh_IN,3;Wirkenergie Lieferung (+A)
1,pm(1.8.1)@1000,kWh_IN_T1,kWh,kWh_IN_T1,3;Wirkenergie Lieferung (+A) Tarif 1
1,pm(1.8.2)@1000,kWh_IN_T2,kWh,kWh_IN_T2,3;Wirkenergie Lieferung (+A) Tarif 2
1,pm(1.7.0)@1000,kW_IN,kW,kW_IN,3;Momentane Wirkleistung Lieferung (+A)
;1,pm(2.8.0)@1000,kWh_OUT,kWh,kWh_OUT,3;Wirkenergie Bezug (-A)
;1,pm(2.8.1)@1000,kWh_OUT_T1,kWh,kWh_OUT_T1,3;Wirkenergie Bezug (-A) Tarif 1
;1,pm(2.8.2)@1000,kWh_OUT_T2,kWh,kWh_OUT_T2,3;Wirkenergie Bezug (-A) Tarif 2
;1,pm(2.7.0)@1000,kW_OUT,kW,kW_OUT,3;Momentane Wirkleistung Bezug (-A)
1,pm(3.8.0)@1000,kvarh_IN,kvarh,kvarh_IN,3;Blindenergie Lieferung (+R)
1,pm(3.8.1)@1000,kvarh_IN_T1,kvarh,kvarh_IN_T1,3;Blindenergie Lieferung (+R) Tarif 1
1,pm(3.8.2)@1000,kvarh_IN_T2,kvarh,kvarh_IN_T2,3;Blindenergie Lieferung (+R) Tarif 2
1,pm(3.7.0)@1000,kvar_IN,kvar,kvar_IN,3;Momentane Blindleistung Lieferung (+R)
1,pm(4.8.0)@1000,kvarh_OUT,kvarh,kvarh_OUT,3;Blindenergie Bezug (-R)
1,pm(4.8.1)@1000,kvarh_OUT_T1,kvarh,kvarh_OUT_T1,3;Blindenergie Bezug (-R) Tarif 1
1,pm(4.8.2)@1000,kvarh_OUT_T2,kvarh,kvarh_OUT_T2,3;Blindenergie Bezug (-R) Tarif 2
1,pm(4.7.0)@1000,kvar_OUT,kvar,kvar_OUT,3;Momentane Blindleistung Bezug (-R)

+2,16,r,0,2400,Haus
2,=so3,512
2,=so4,GUEK
2,pm(1.8.0)@1000,kWh_IN,kWh,kWh_IN,3;Wirkenergie Lieferung (+A)
2,pm(1.8.1)@1000,kWh_IN_T1,kWh,kWh_IN_T1,3;Wirkenergie Lieferung (+A) Tarif 1
2,pm(1.8.2)@1000,kWh_IN_T2,kWh,kWh_IN_T2,3;Wirkenergie Lieferung (+A) Tarif 2
2,pm(1.7.0)@1000,kW_IN,kW,kW_IN,3;Momentane Wirkleistung Lieferung (+A)
;2,pm(2.8.0)@1000,kWh_OUT,kWh,kWh_OUT,3;Wirkenergie Bezug (-A)
;2,pm(2.8.1)@1000,kWh_OUT_T1,kWh,kWh_OUT_T1,3;Wirkenergie Bezug (-A) Tarif 1
;2,pm(2.8.2)@1000,kWh_OUT_T2,kWh,kWh_OUT_T2,3;Wirkenergie Bezug (-A) Tarif 2
;2,pm(2.7.0)@1000,kW_OUT,kW,kW_OUT,3;Momentane Wirkleistung Bezug (-A)
2,pm(3.8.0)@1000,kvarh_IN,kvarh,kvarh_IN,3;Blindenergie Lieferung (+R)
2,pm(3.8.1)@1000,kvarh_IN_T1,kvarh,kvarh_IN_T1,3;Blindenergie Lieferung (+R) Tarif 1
2,pm(3.8.2)@1000,kvarh_IN_T2,kvarh,kvarh_IN_T2,3;Blindenergie Lieferung (+R) Tarif 2
2,pm(3.7.0)@1000,kvar_IN,kvar,kvar_IN,3;Momentane Blindleistung Lieferung (+R)
2,pm(4.8.0)@1000,kvarh_OUT,kvarh,kvarh_OUT,3;Blindenergie Bezug (-R)
2,pm(4.8.1)@1000,kvarh_OUT_T1,kvarh,kvarh_OUT_T1,3;Blindenergie Bezug (-R) Tarif 1
2,pm(4.8.2)@1000,kvarh_OUT_T2,kvarh,kvarh_OUT_T2,3;Blindenergie Bezug (-R) Tarif 2
2,pm(4.7.0)@1000,kvar_OUT,kvar,kvar_OUT,3;Momentane Blindleistung Bezug (-R)
#

;Landis+Gyr E450
;https://www.e-netze.at/downloads-data/pdf.aspx?pdf=EN_Update%20Kundenschnittstelle%20Smart%20Meter_ID3282_WEB_RGB.pdf
```
@australien340 This is for the new config :-)
[tasmota32.zip](https://github.com/arendst/Tasmota/files/10530237/tasmota32.zip)

gismo2004, 2023-01-30:
@gemu2015 for some reason, some signals have reported zero overnight. I am quite sure, that this is an error/problem on SmartMeter side, but unfortunately this leads to bad statistics in HomeAssistant. Do you see any chance to prevent reading/sending values lower/higher than xy via “limit” set in signal description line? 

```
1,pm(1.8.0,lowerLim,upperLim)@1000,kWh_IN,kWh,kWh_IN,3;Wirkenergie Lieferung (+A)
```

[image]

gemu2015, 2023-01-30:
try setting the median filter (flag = 16)
+1,17,r,16,2400,Heizung

m1k3f15h, 2023-02-03:
Hello,

I tried do compile @gemu2015 last version of tasmota but with SML but:

```
c:/users/michael/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\tasmota-sml\libc77\libams.a(han_Parser.cpp.o):(.text._ZN10Han_Parser10unwrapDataEPhR17DataParserContext+0x24): undefined reference to `GCMParser::GCMParser(unsigned char*, unsigned char*)'
c:/users/michael/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\tasmota-sml\libc77\libams.a(han_Parser.cpp.o):(.text._ZN10Han_Parser10unwrapDataEPhR17DataParserContext+0x28): undefined reference to `GCMParser::parse(unsigned char*, DataParserContext&)'
c:/users/michael/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\tasmota-sml\libc77\libams.a(han_Parser.cpp.o): in function `Han_Parser::unwrapData(unsigned char*, DataParserContext&)':
han_Parser.cpp:(.text._ZN10Han_Parser10unwrapDataEPhR17DataParserContext+0x12a): undefined reference to `GCMParser::GCMParser(unsigned char*, unsigned char*)'
c:/users/michael/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: han_Parser.cpp:(.text._ZN10Han_Parser10unwrapDataEPhR17DataParserContext+0x13a): undefined reference to `GCMParser::parse(unsigned char*, DataParserContext&)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tasmota-sml\firmware.elf] Error 1
```

Someone knows about this?

mike

gemu2015, 2023-02-03:
#define USE_TLS
or use latest dev from Tasmota

m1k3f15h, 2023-02-03:
Thank you @gemu2015 :)
compile ok.. now testing :)

m1k3f15h, 2023-02-03:
I see now data:

`13:17:50.097 decrypted block: 512 bytes
13:17:50.099 :>e6 01 01 68 53 00 00 00 01 67 1b 00 08 53 41 47 
13:17:50.101 :>59 05 e9 46 11 01 00 f8 20 00 4d 09 00 23 00 16 
13:17:50.103 :>14 00 01 00 18 06 00 6a 09 32 00 06 9d a7 0a 00 
13:17:50.105 :>7d 7f b6 a7 43 1b d0 7e 38 1d 00 91 00 1d 00 94 
13:17:50.107 :>cb 18 77 e6 5b 55 1c 00 0c 00 3b 00 01 0f 00 32 
13:17:50.109 :>8f 0c 00 bc c1 3f 73 2a 00 2f 00 d9 a7 51 cb 2b 
13:17:50.112 :>24 3d 58 16 21 37 85 c8 21 f1 62 6e 0b 00 39 fe 
13:17:50.114 :>17 13 00 02 00 df 0e 00 3e 13 00 32 0f 03 00 3b 
13:17:50.116 :>11 ef 7f 54 25 34 00 07 00 67 29 09 00 4d 36 71 
13:17:50.118 :>e9 77 1b e9 55 8f 25 01 00 28 12 00 ef b5 d3 71 
13:17:50.120 :>b0 24 d6 1c 00 30 0f 00 0c 3d bc da ef cd 79 01 
13:17:50.122 :>00 4e ba 32 00 34 6c 2e a7 3b 04 69 04 00 6a 20 
13:17:50.124 :>a1 78 20 00 09 15 00 fd 3e 00 3f 12 10 80 0c 11 
13:17:50.126 :>00 01 00 1c 00 3f 01 00 3b 00 72 07 00 51 97 b6 
13:17:50.128 :>f7 58 7c b3 7a 10 02 07 00 ba 8f 06 57 ec 0b 00 
13:17:50.131 :>77 fd 85 1a 00 0f 00 ae 47 6f 98 02 00 80 38 e0 
13:17:50.133 :>33 33 1d 33 01 00 8f 00 00 00 00 00 00 00 00 00 
13:17:50.135 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.137 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.139 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.141 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.143 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.145 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.147 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.149 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.151 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.153 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.155 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.158 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.160 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.162 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:17:50.164 :>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 `

but i dont know what to write in the script / describer:
https://www.netz-noe.at/Download-(1)/Smart-Meter/218_9_SmartMeter_Kundenschnittstelle_lektoriert_14.aspx

@gemu2015 can you give me a hint?

mike

gismo2004, 2023-02-03:
@m1k3f15h --> it's all in the document

This should be "Momentanleistung"
```
1,pm(2.7.0)@1,kW_IN,kW,kW_IN,3;
```

m1k3f15h, 2023-02-03:
Yess rlthis works perfect

gismo2004, 2023-02-03:
Ok, what is your current config?

gemu2015, 2023-02-03:
switch to console debug mode to see what is decoded in detail

Noschvie, 2023-02-03:
@m1k3f15h : Hi Mike, which type of smart meter do you have ? Kaifa MA309 or Sagemcom T210 ?

m1k3f15h, 2023-02-03:
@Noschvie SagemcomT210

@gemu2015:

`13:44:53.595 SML: sync
13:44:54.555 SML: MBUS frame:
13:44:54.557 SML: Boundry flag missing
13:44:54.607 SML: Ended up in default case while unwrapping...(tag 3B)
13:44:54.609 SML: Unknown data payload:
13:44:54.656 SML: Ended up in default case while unwrapping...(tag DA)
13:44:54.658 SML: Unknown data payload:
13:44:54.708 SML: Ended up in default case while unwrapping...(tag 72)
13:44:54.710 SML: Unknown data payload:
13:44:54.758 SML: Ended up in default case while unwrapping...(tag AE)
13:44:54.760 SML: Unknown data payload:
13:44:54.810 SML: Ended up in default case while unwrapping...(tag 0D)
13:44:54.812 SML: Unknown data payload:
13:44:54.859 SML: Ended up in default case while unwrapping...(tag 26)
13:44:54.861 SML: Unknown data payload:`

[111 more replies]

PR #17828 (2023-01-30): Sml ams
## Description:

**Related issue (if applicable):** fixes #8790
support for encrypted advanced metering systems

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributo [... cut]