arendst/Tasmota #8792: Add option for mired min and mired max

crazifyngers, 2020-06-26:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
in home assistant you can send a "kelvin" variable to a light.  this translates to mired.  If the min and max mired do not match the warmwhite and cool white led then the scale is off and the wrong temperature is displayed.

**Describe the solution you'd like**  
an option to set the minimum mired and maximum mired.

**Describe alternatives you've considered**  
I asked on a discord server if this was possible to accomplish in tasmota.  I was told no, but an alternative would be to use a template light in home assistant.

**Additional context**  
Tasmota has a min mired of 153 (~6500k) and a maximum mired of 500 (2000k).  Most color temperature bulbs cannot attain that range without using the rgb led.  Not all CT lights also have RGB leds and calibrating the CT bulbs with the RGB would not be ideal for most.  I propose a simpler solution would be to allow a configurable mired range.  Most CT bubs I have seen have a 2700k-6500k range but have also seen some with 2200k-4000k.  

Thank you for the consideration.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-06-27:
There is already `SetOption82 1` which reduces the range to 200-380, so that's 2600K-5000K. It allows to take into account the range used by Alexa.

Is there a need to have a narrower range?

crazifyngers, 2020-06-27:
It is not required to have a narrower range.  What I'm asking for is a configurable range, setoption82 is a fixed range.

If I use setoption82 (2600-5000) but my warm white is 2700 and cool white is 6500, then when you call for 2600 it will actually display 2700 and when you call for 5000 it will actually display 6500.  This will happen throughout the CT range.

s-hadinger, 2020-06-27:
What bulb model do you have?

Actually Tasmota does not know about Mireds. It just use this range as a convention to split power between CW and WW leds. I don't really see the value of calibrating the Mireds value for each bulb model, taking into account that unless you know the value from the manufacturer, it's hard to measure visually.

markianwallace, 2020-06-27:
Spooky, I just came to open the same request. Hard limits on min and max mireds would be fantastic.

Existing code could be essentially reused, just need to push the integer values to the existing function

`  void setAlexaCTRange(bool alexa_ct_range) {
    // depending on SetOption82, full or limited CT range
    if (alexa_ct_range) {
      _state->setCTRange(CT_MIN_ALEXA, CT_MAX_ALEXA);   // 200..380
    } else {
      _state->setCTRange(CT_MIN, CT_MAX);               // 153..500
    }
  }`

crazifyngers, 2020-06-27:
I wasn't looking for any calibration, just the option to manually set the min/max.  I also understand that it is an arbitrary indicator to split the power between the ww/cw.  However, when used in conjunction with home assistant the min)max mireds is exposed and used by home assistant.  This means if I tell home assistant to set the Kelvin to 2700 it will be wrong.

I'm not looking for any smarts or calibration.  That is way to big an ask and impossible really since led can change from lot to lot.

crazifyngers, 2020-06-27:
Here is a screenshot of what is reported.  I would just like these reported numbers to be configurable in tasmota.
[image]

blakadder, 2020-06-27:
but the highlighted values are already configurable in Home Assistant! why do you need them configurable in Tasmota?

markianwallace, 2020-06-28:
In HA they are not hard limits. Just as with SetOption82 this is another case where I would like to specify these limits to control the correct hardware mireds range.

Could I suggest just making it 

SetOption82 min max

and specifying in the instruction what those values are for the existing case?

with a note as to the correct

Jason2866, 2020-06-28:
@markianwallace  you find a good explanation here why it makes no sense
https://discord.com/channels/530777070435827722/530862923363516417/726574846829658156

crazifyngers, 2020-06-28:
yep that was me who was having the conversation with blakadder after he commented here.  I am closing this request as I now understand it to not fix the issue I was trying to solve.

closed 2020-06-28 by crazifyngers

srnoth, 2020-11-09:
@crazifyngers did you find a solution to this?

I've managed to make this work using the Template schema in HASS, but surely this would be better to set in Tasmota itself?

The problem is that MIREDS are not a random scale like 0 to 100%. Mireds represent actual white color temperatures that need to comparable across multiple light integrations. E.g. I have both zigbee and Tasmota CCT bulbs. During the day I have them set to 5000 kelvin (200 mireds) and at night 2700 kelvin (370 mireds).

However as you said:
`If I use setoption82 (2600-5000) but my warm white is 2700 and cool white is 6500, then when you call for 2600 it will actually display 2700 and when you call for 5000 it will actually display 6500. This will happen throughout the CT range.`

This means if I call Color Temp directly (e.g. using the new Tasmota HASS integration), when my Zigbee lights are displaying 2700k (370 mireds), my Tasmota lights will be displaying a much cooler light (I'd have to call 500 mireds to actually get only the 2700k LEDs in Tasmota).

I've solved this using the Template schema and some mathematical calculations to convert the correct Mireds value in HASS to the scaled value in Tasmota:
```
command_on_template: >-
        {%- if color_temp is defined -%}
        CT {{ ((((color_temp | float)-153)*1.599078)+153) | round(0) }}; 
        {%- endif -%}
color_temp_template: '{{ ((((value_json.CT | float)-153)*0.62536)+153) | round(0) }}'
```

But that seems like a work around whether than the proper fix, which would be to specify in Tasmota the "Warm" and "Cold" temperature of a particular CCT bulb, so that the Tasmota mireds scale actually corresponds to the correct Kelvin temperature.

This is especially important for using the built in HASS Tasmota integration that is coming out.