arendst/Tasmota #8794: Get no Sound from Buzzer

ChrisVonFrosta, 2020-06-26:
### PROBLEM DESCRIPTION
an installed piezo buzzer is setup as Buzzer. If send Command Buzzer 2,3,4 i can recognize super silent knocks. but no beeps.
When set this Buzzer/Port as PWM i can contoll is channel to do beeps.
What ist to do to get the piezo buzzer works as "Tasmota's Buzzer"?

### REQUESTED INFORMATION
- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): NodeMCU V3 ESP 8266
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-28:
You may have a power issue with your buzzer. Depending on the buzzer voltage you may need an amplifier to use. A simple transister could do the trick.

ChrisVonFrosta, 2020-06-28:
But it do sounds if i configure the output as pwm. So i thought it shouldn’t  be a voltage issue.

Lawrencezarb, 2020-06-30:
you probably have the wrong type of buzzer. You are using one that requires a square wave/PWM signal. You should be using one that only requires the trigger pin to go high to sound, ie a KY-012 buzzer.

ascillato2, 2020-07-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-01 by ascillato2

ChrisVonFrosta, 2020-07-01:
I could be wrong but this is not described in the wiki anyway. Maybe a little hint could be helpfull there.

Btw.  It would be a nice feature to let the buzzer output “speak” pwm beeps.

hello-world-dot-c, 2020-09-08:
I am working on it:
https://github.com/arendst/Tasmota/pull/8994

BroBan, 2020-12-01:
@ChrisVonFrosta Could you please tell me how you setup a buzzer with the sonoff basic? Like what GPIO pins you used, did you need to add a resistor? Is the 3.3V out pin from the sonoff enough to power the active piezzo buzzer? Thanks!

ChrisVonFrosta, 2020-12-01:
@BroBan 
hard do remamber.
the result was that i use a ws28xx for indication.
But for tests my buzzer was one with intergrated transistor i think.
so it could  powered 5 V and was able to switch "activ low" with gpio.
But u need one which work without pwm signal.

Bobatwork, 2021-11-11:
Found this thread when searching myself. You can use the PWM Buzzer, just do setoption111 1. Works like a charm. 
If you need more voltage than 3.3V from your ESP, you can use an NPN transistor, with a approx. 220Ohm resistor before the base and a flyback diode across the buzzer, if you really want to get serios about it.

thomas374b, 2023-11-28:
I had the same Problem and I dug deeper.

It seems to be that the function 
   analogWrite(pin, duty);
in method
   BuzzerSet()
in file
  xrdv_24_buzzer.ino
is wrapping the duty to 8 bits.

I verified it with a scope and customized self-compiled firmware
with
   duty := 254  I get sound with a 8µs long L-Pulse
   duty := 255  I get nothing, but one click
   duty := 256  I get nothing, but one click
   duty := 257  I get sound  with a 4µs long H-pulse

The argument in the original code id pwm_range/2.  But pwm_range is 1023 and 511 folded to 8bit is 255
Thats why, the buzzer is only silent clicking instead of buzzing

I reworked the buzzer code already for me and added some alarm-noise patterns with changing frequencies and allowed
the user to change frequency and duty (for loudness) via extra commands.
The code is all ready but not yet under git control. 

I might give you a pull request if I find the time to cleanup and checkin.