arendst/Tasmota #8798: KNX issue on iFan02

nitinuae, 2020-06-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Hello

Can we have fanspeed (or switch fan on off) by using KNX slots?
Thank you.

ascillato, 2020-06-27:
Hi,

What you have tried?

I don't have an iFAN02 to test, but checking the template https://templates.blakadder.com/sonoff_ifan02.html it seems that you should have relays 1 to 4 to select from the KNX menu right? And you can select the speed from each relay right?

nitinuae, 2020-06-27:
I have tried the following - (and found)
Output1 = Light = KNX1
Output2 = FanSpeed1 = Knx2 (just realised I can do this)

Others - I can also enable output 3, 4 - But they dont do anything (Dont even switch on fan).
The web server though says fan at 2,3 or even 4 !!! (this is bug)
Thank you.

nitinuae, 2020-06-27:
Adding (For other for info)
Starting from fan off, -

When I switch on fan3, I can hear the relays. The relay seem to suggest that fan starts at 1, then after 1 second, 2 more relay are heard- making the fan go to 3.

When I switch on fan2, I can hear first relay and then after 1 second, only one relay.

Writing directly to the KNX or output slots (fan speed 2 or fan speed 3) will make webserver show the speed, but the fan will not move.
(and that bug still remains - You can make fan speed 4 also)

ascillato, 2020-06-27:
Ok. Thanks for the test. So what you need is just use knx RX slots for receiving the speed and set rules to run the command fanspeed accordingly.

ascillato2, 2020-06-28:
Hi,

Were you able to get this working? 

Which approach do you use? 
You want a different address for each speed or you want to receive the speed value in just one address?

nitinuae, 2020-06-28:
No- I was not.
Light works, fan speed doesnot.
I will send screenshots etc a few hours later.

ascillato2, 2020-07-04:
Hi, any news on this?

nitinuae, 2020-07-07:
Hello
Sorry for delay -
I was not able to make the fan speed work on KNX.
I tried to switch the relay individually, but it doesnot work (As per some other thread - MCU is the king on this controller)

The controller is OFF now (so I cannot post the pages as I had written earlier) - Will try to post later when its on.
Summary - Individual relay DOESNOT bring the fan to speed. 2 relays donot work (though I can hear them switching on off). Only one relay works - which brings the relay on.

I can bring some relay on which says speed 4 (this should be pointed as bug in web display)
Regards
Nitin

ascillato, 2020-08-01:
testing...

ascillato2, 2020-08-24:
Hi, 

Sorry for the delay.

A rework of the driver is needed to support IFAN.

Changing this issue to a Feature Request.

ascillato, 2020-11-26:
Hi,

Revisiting this issue, I see now that in KNX protocol there isn't any DPT for this type of FANs (https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf).

So, the implementation for this is not going to follow the KNX standard.

How you was going to control it from your KNX network? Which GA type you want to use?

As it is going to be a custom implementation, I see that we can set a some **rules** to control your IFAN.

As I don't have an IFAN, I took the information from https://tasmota.github.io/docs/devices/Sonoff-iFan02/ and https://tasmota.github.io/docs/Home-Assistant/#device-specific

We can set:

* a KNX TX slot1 for sending the actual State of the FAN (ON/OFF)
* a KNX RX slot1 for receiving Turn ON and OFF commands
The 2 above can have the same GA
* a KNX TX slot2 for sending the actual speed of the FAN 0 to 3
* a KNX RX slot2 for receiving speed commands (0 to 3)
The 2 above can have the same GA

```lua
rule1 1
rule1
  on fanspeed#data do KnxTx_Val2 %value% endon
  on power1#state do KnxTx_cmnd1 %value% endon
  on event#knxrx_val2 do fanspeed %value% endon
  on event#KnxTx_Cmnd1 do power1 %value% endon
```

All together for copy and paste:

```lua
rule1 1
rule1 on fanspeed#data do KnxTx_Val2 %value% endon on power1#state do KnxTx_cmnd1 %value% endon on event#knxrx_val2 do fanspeed %value% endon on event#KnxTx_Cmnd1 do power1 %value% endon
```

Hope this helps.

ascillato2, 2020-11-26:
If this was useful for you, please consider helping us and add this to the docs. Thanks.

closed 2020-11-26 by ascillato2