arendst/Tasmota #8803: Running Time

bug84, 2020-06-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

————————————————————
Hello, on my Sonoff 4ch I see the running time (uptime) of the device, but I can’t see the running time from my 4 ports? Can u add the information please? 
I want to know, how long was port 1 on “true” and how long the others.

best regards

ascillato, 2020-06-27:
Just for reference, this request has been already requested before a couple of times but nobody spend time on working on this due to it is already implemented by default in any home automation software. https://github.com/arendst/Tasmota/issues/8054

bug84, 2020-06-27:
however, when i install it in my smart home control, i am not protected against disconnections. that's why I want to measure it directly on the sonoff and then just call it up

curzon01, 2020-06-28:
If you implement it in Tasmota, then it is not protected against Tasmota failures (power failure, etc.). Which errors occur more often is very user-specific. To prevent such cases, you have to write the runtime into the flash - a value that changes every second! - this is detrimental to the life of the flash and reducing the flash store interval leads back to the first-mentioned problem.
In addition saving runtimes costs some of the valuable flash space, a solution should not handle only 4 channels but all possible, currently 8 - I suspect 32 byte (8 x 32bit)

arendst, 2020-06-28:
As said above get a decent HA tool and you'll be able to see a lot more:

[image]

It's a waste of code space and won't be implemented any time soon.

closed 2020-06-28 by ascillato2