arendst/Tasmota #8805: Add support for Pressure and Level for Analog input.

eXCepT74, 2020-06-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hello . I just wanted to propose adding support for pressure and level sensor for analog input.
Many sensors ( even in industry ) sending analog signal instead of digital. And those sensors are very exact.

Most of those  which i working with - can send analog signal 0-10v or 0-20mA. ( voltage or current - it can be set manually or  automatically after measure load resistance ).
0-10V can be easily convert to ESP levels with resistance divider.

Im interested in option for analog input which can give reading pressure in >Bar< with one or two digit after dot exact and level in >mm<.
Im not sure but there is probably already some calibration command in tasmota for analog input . 
I wanted to use following sensors. Both of those sending voltage 0-10v . Level sensor is fully configurable and i will use in range 0 - 1500 or 2000 mm. It would be great if we could configure that range in free way. Pressure sensor sending 0-10v in fixed range 0-6 Bar.

lfp cubic sensor from Sick ( level in mm )

https://www.google.com/url?sa=i&url=https%3A%2F%2Fplcwest.com%2Fproduct%2Fsick-lfp-cubic-level-sensor%2F&psig=AOvVaw2ytNNOvrwv52aRbuLWw6Jw&ust=1593383613720000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCOi77deGo-oCFQAAAAAdAAAAABAQ

and Pressure sensor from Huba Control

https://www.hubacontrol.com/en/products/pressure-transmitter/pressure-sensor-501

ascillato, 2020-06-28:
Hi

To make the analog input as generic as it could be, Tasmota already supports to scale any signal from the adc pin. There are several options in order to choose the best scenario for you and adapt the signal to the value you need in order to send it by mqtt. 

If, besides sending the scaled adc value, you want to add any name, you can use the rules feature of Tasmota and publish the mqtt message you need.

eXCepT74, 2020-06-28:
Ok. Thanks for answer. So i have to take look on that scale commands and rules.
Few more examples in cookbook for A0 could be useful.
If nobody else has problem with that it can be close.
Cheers

ascillato2, 2020-06-28:
Great.

If you need further assistant on making those rules, please, just ask. 

Thanks

closed 2020-06-28 by ascillato2