arendst/Tasmota #8808: global_temperature is used inconsistently (almost always as Celsius, once as Fahrenheit)

mhansen, 2020-06-28:
### PROBLEM DESCRIPTION

It's unclear what unit `global_temperature` is.

Forked off an issue from https://github.com/arendst/Tasmota/pull/8799 as this comment was starting to get a bit long.

I'm taking a look through the code and getting a bit confused trying to figure out what unit `global_temperature` is...

## The places that treat global_temperature as Celsius

I see only one non-NAN assignment to global_temperature, in [`support.ino`](https://github.com/arendst/Tasmota/blob/development/tasmota/support.ino#L604):

```
float ConvertTemp(float c)
{
  float result = c;

  global_update = uptime;
  global_temperature = c;

  if (!isnan(c) && Settings.flag.temperature_conversion) {    // SetOption8 - Switch between Celsius or Fahrenheit
    result = c * 1.8 + 32;                                    // Fahrenheit
  }
  result = result + (0.1 * Settings.temp_comp);
  return result;
}
```

Looks like global_temperature is set to 'c' (short for Celsius?), and then `Settings.flag.temperature_conversion` optionally converts the return value to Fahrenheit for display -- doing the `F = C * 9/5 + 32` equation.

And I see that `global_temperature` is used unconditionally as Celsius in [`xdrv_03_energy.ino`](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_03_energy.ino#L462):

```
    if (power && !isnan(global_temperature) && (global_temperature > (float)Settings.param[P_OVER_TEMP])) {  // Device overtemp, turn off relays
```
where Settings.param[`P_OVER_TEMP`] is set to `ENERGY_OVERTEMP` in [`settings.ino`](https://github.com/arendst/Tasmota/blob/faecd38006836018926186a7a6f39adadfe845f9/tasmota/settings.ino#L935):

```
  Settings.param[P_OVER_TEMP] = ENERGY_OVERTEMP;
```

Where [`ENERGY_OVERTEMP` is in Celsius:(https://github.com/arendst/Tasmota/blob/f4adce5fb15c4aa2375b8558ee66a73087a0d0b4/tasmota/tasmota_globals.h#L205)

```
#define ENERGY_OVERTEMP             90         // Overtemp in Celsius
```

And in [`support_command.ino`](https://github.com/arendst/Tasmota/blob/f4adce5fb15c4aa2375b8558ee66a73087a0d0b4/tasmota/support_command.ino) I see they convert Fahrenheit to Celsius before calling `ConvertTemp`:

```
void CmndGlobalTemp(void)
{
  if (XdrvMailbox.data_len > 0) {
    float temperature = CharToFloat(XdrvMailbox.data);
    if (!isnan(temperature) && Settings.flag.temperature_conversion) {    // SetOption8 - Switch between Celsius or Fahrenheit
      temperature = (temperature - 32) / 1.8;                             // Celsius
    }
    if ((temperature >= -50.0f) && (temperature <= 100.0f)) {
      ConvertTemp(temperature);
      global_update = 1;  // Keep global values just entered valid
    }
  }
  ResponseCmndFloat(global_temperature, 1);
}
```

[`xsns_31_ccs811.ino`](
https://github.com/arendst/Tasmota/blob/0abfcf1954fa6d60fab5a63d04cfde8d4477a32f/tasmota/xsns_31_ccs811.ino#L69) treats `global_temperature` as Celsius, passing it to `ccs.setEnvironmentalData`, which expects Celsius:
```
          ccs.setEnvironmentalData((uint8_t)global_humidity, global_temperature);
```

So all those so far are treating `global_temperature` as Celsius.

## The place that treats global_temperature as Celsius or Fahrenheit

There's just one dissenter: [`xsns_21_sgp30.ino`](https://github.com/arendst/Tasmota/blob/0abfcf1954fa6d60fab5a63d04cfde8d4477a32f/tasmota/xsns_21_sgp30.ino#L92) seems to treat global_temperature as either being Celsius or Fahrenheit depending on the TempUnit:
```
    sgp30_abshum=sgp30_AbsoluteHumidity(global_temperature,global_humidity,TempUnit());
```

## Conclusion

Given all that, I propose:
* changing `xsns_21_sgp30.ino` to always treat `global_temperature` as Celsius
* renaming `global_temperature` to `global_temperature_celsius` to make it very clear what units the field has

What do you think?

_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mhansen, 2020-06-28:
I should note, I'm happy to prepare a pull request to address this, just want to make sure I'm not missing anything?

arendst, 2020-06-28:
Thx for your research. The thing is different users support Tasmota with their PR's and sometimes they don't see the whole picture.

I'll have a look at the fahrenheit implementation.

My goal of Tasmota was using ISO standard units. As the USA guys can't get rid of their non-Iso standards I joyfully support some non-Iso units too. The global variables were ment to be used internally by incidental drivers needing them. If you dive deeper in the global variables you'll notice they won't stick to a valid global temperature if more than one sensors are connected. This is no problem as long as the user of the global variable knows this; mostly only one temp sensor is connected where the global temp is needed for calculation.

In case of Prometheus I don't think you should use the global vars for reporting temps/hums and pressures but should catch the individual sensors and report their values. In the end of the prometheus code I provided a framework to catch sensor JSON messages that a programmer can use to send individual prometheus values.

Knowing this I don't think renaming global variables is needed but as it can be done easily I will do it anyway ;-)

closed 2020-06-28 by ascillato2

mhansen, 2020-06-28:
Thanks for the guidance, indeed it sounds like a bad idea to use global
temperature for Prometheus exports, sounds better to loop through the
sensors and export each one individually.

Thanks for renaming the variable, this will make bugs jump out and be much
more obvious :-)