arendst/Tasmota #8810: Document units of global_pressure

mhansen, 2020-06-28:
### PROBLEM DESCRIPTION
It's unclear from the comments what units `float global_pressure` is in.

Background: I'm adding some Prometheus exporting code to Tasmota and I want to export in Pascals, but I don't know how to convert `global_pressure` into Pascals if the units are undocumented.

## How is global_pressure set?

Looks like `global_pressure` is only ever assigned to one non-zero value in [`support.ino`](https://github.com/arendst/Tasmota/blob/development/tasmota/support.ino#L664):

```
float ConvertPressure(float p)
{
  float result = p;

  global_update = uptime;
  global_pressure = p;

  if (!isnan(p) && Settings.flag.pressure_conversion) {  // SetOption24 - Switch between hPa or mmHg pressure unit
    result = p * 0.75006375541921;                       // mmHg
  }
  return result;
}
```

For the maths to work here, the input argument `p` must always be in hPa, and the return value will depend on `Settings.flag.pressure_conversion`.

ConvertPressure is called with hPa from [xsns_73_hp303b.ino](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_73_hp303b.ino#L69):

```
  hp303b_sensor[hp303b_idx].pressure = (float)ConvertPressure(p / 100);  // Conversion to hPa
``` 

And ConvertPressure is also called from [`xsns_09_bmp.ino`](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_09_bmp.ino#L526)
```
      float bmp_pressure = ConvertPressure(bmp_sensors[bmp_idx].bmp_pressure);
```

where bmp_pressure seems to always [be in millibars (another way to say hPa)](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_09_bmp.ino#L196
):
```
  bmp_sensors[bmp_idx].bmp_pressure = (float)p / 100.0;  // convert to mbar
```

The [25600 divisor seems to be a way to convert to mBar/hPa too](https://github.com/kriswiner/nRF51822_IoT_Breakout/blob/master/nRF52832/main.cpp#L306) in other projects:
```
  bmp_sensors[bmp_idx].bmp_pressure = (float)p / 25600.0;
```

This looks like converting Pa to hPa:
```
      bmp_sensors[bmp_idx].bmp_pressure = data.pressure / 100.0;
```

So I think `global_pressure` is in hPa.

## How is global_pressure used?

Looks like it's used in `xrdv_10_scripter.ino`, which doesn't care about units, and in `xsns_75_prometheus.ino` which is also noncommittal about units.

## Conclusion

I propose, to make the units clearer:

* Renaming `bmp_sensors[bmp_idx].bmp_pressure` to `bmp_sensors[bmp_idx].bmp_pressure_hpa`
* Renaming `global_pressure` to `global_pressure_hpa`

What do you think?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mhansen, 2020-06-28:
I should note, I'm happy to prepare a pull request to address this, just want to make sure I'm not missing anything?

arendst, 2020-06-28:
See https://github.com/arendst/Tasmota/issues/8808#issuecomment-650728767

It's in hPa indeed.

closed 2020-06-28 by ascillato2

mhansen, 2020-06-28:
Thank you! This is much clearer.