arendst/Tasmota #8811: Westinghouse Ceiling Fan IR Raw Unknown Raw Output

Cosmicbase, 2020-06-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS = {"Status":{"Module":0,"DeviceName":"IR-Fernbedienung","FriendlyName":["IR-Fernbedienung"],"Topic":"IR-Schlafzimmer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-ir.bin","RestartReason":"Power On","Uptime":"1T15:48:42","StartupUTC":"2020-06-26T18:05:04","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-06-26T13:50:16","SaveCount":47,"SaveAddress":"FB000"}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS2 = {"StatusFWR":{"Version":"8.3.1(ir)","BuildDateTime":"2020-05-18T15:42:24","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"450/699"}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Villa dei Toscana",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000100","00000000"]}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8F5A2583","040003A0","000000C1","010000C0","00000000","00000020"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":"1,2,5,6"}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS5 = {"StatusNET":{"Hostname":"IR-Fernbedienung-Schlafzimmer","IPAddress":"192.168.2.24","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"BC:DD:C2:A7:6C:F4","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.66","MqttPort":1883,"MqttClientMask":"IR-Schlafzimmer","MqttClient":"IR-Schlafzimmer","MqttUser":"admin","MqttCount":12,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS7 = {"StatusTIM":{"UTC":"2020-06-28T09:53:46","Local":"2020-06-28T11:53:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:49","Sunset":"21:57"}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS10 = {"StatusSNS":{"Time":"2020-06-28T11:53:46"}}
11:53:46 MQT: stat/IR-Schlafzimmer/STATUS11 = {"StatusSTS":{"Time":"2020-06-28T11:53:46","Uptime":"1T15:48:42","UptimeSec":143322,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":12,"Wifi":{"AP":1,"SSId":"Villa dei Toscana","BSSId":"F0:B0:14:B9:F1:0D","Channel":11,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

When I press the light on and light off button on the remote control, I get the following output.
 IRsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1FB}

 IRsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1EF}

 When I press the Fan Off button, I get the following output.

 IRsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1F7}

 When I press the fan level 1 button.

 12:32:45 MQT: tele / IR bedroom / RESULT = {"IrReceived": {"Protocol": "UNKNOWN", "Bits": 88, "Hash": "0xC5EE6423", "Repeat": 0, "  RawData ": [1318,400,1286,402,442,1216,472,1244,442,1244,466,1222,442,1214,472,1248,464,1194,472,1216,470,1216,1316,7122,  1,318,402,1286,400,442,1218,492,1194,468,1218,470,1218,470,1244,442,1216,472,1216,470,1246,442,1216,1314,7122,1318,400,  1,286,402,466,1222,442,1244,442,1246,442,1216,496,1190,472,1246,442,1216,472,1212,474,1216,1312,7124,1340,378,1286,400,442,  1212,500,1190,474,1216,494,1192,472,1214,472,1214,474,1214,472,1246,464,1198,1336,7096,1320,402,1286,402,468,1190,472,  1216,472,1216,472,1214,472,1216,472,1216,472,1218,470,1216,472,1214,1316,7124,1340,378,1286,402,442,1216, ...

 12:32:51 MQT: tele / IR bedroom / RESULT = {"IrReceived": {"Protocol": "UNKNOWN", "Bits": 66, "Hash": "0x268C17C3", "Repeat": 0, "  RawData ": [1340,378,1306,380,466,1192,470,1218,494,1220,444,1216,468,1218,472,1244,442,1218,468,1218,494,1220,1286,7124,  1,316,402,1286,402,442,1216,470,1216,472,1244,440,1246,442,1244,442,1216,472,1244,444,1242,466,1222,1284,7124,1316,402,  1310,378,442,1218,492,1196,468,1218,470,1246,442,1218,470,1244,442,1246,466,1220,444,1216,1338,7098,1318,402,1284,402,466,  1220,442,1216,470,1216,470,1220,492,1192,472,1218,468,1216,472,1216,494,1222,1286,7122,1342,376,1286,402,440,1216,474,  1216,470,1216,472,1246,442,1214,472,1214,474,1214,472,1246,442,1214,1316,7122,1342,376,1286,402,442,1246, ...

 12:34:49 MQT: tele / IR bedroom / RESULT = {"IrReceived": {"Protocol": "UNKNOWN", "Bits": Taste3, "Hash": "0x25AE7EE0", "Repeat": 0, "  RawData ": [88,33324,86,33324,90]," RawDataInfo ": [5,5,0]}}

 Fan level 2.

 IRsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1FD}

 And fan level 3.

 12:26:02 MQT: tele / IR bedroom / RESULT = {"IrReceived": {"Protocol": "UNKNOWN", "Bits": 57, "Hash": "0xE970CDA4", "Repeat": 0, "  RawData ": [1342,348,1336,380,466,1190,474,1218,470,1242,470,1190,470,1244,466,1222,468,1218,470,1186,498,1190,1338,7100,  1340,376,1310,376,468,1220,468,1218,466,1222,466,1220,466,1194,470,1216,470,1244,468,1220,468,1220,1308,7098,1320,372,  1336,378,444,1216,494,1220,466,1220,468,1220,468,1220,470,1190,496,1218,444,1214,494,1194,1338,7098,1342,350,1336,376,468,  1188,496,1222,466,1192,498,1218,470,1218,468,1218,468,1218,466,1220,468,1190,1338,7098,1320,398,1288,400,466,1190,500,  1190,496,1220,466,1192,498,1218,468,1220,468], "RawDataInfo": [113,113,0]}}

 12:27:48 MQT: tele / IR bedroom / RESULT = {"IrReceived": {"Protocol": "UNKNOWN", "Bits": 41, "Hash": "0x3DC8DA51", "Repeat": 0, "  RawData ": [1340,378,1308,378,468,1190,472,1214,472,1216,470,1246,466,1222,454,1204,472,1218,492,1220,468,1220,1286,7122,  1,318,402,1286,400,468,1190,472,1216,470,1246,466,1222,442,1214,472,1218,494,1192,494,1196,492,1192,1314,7122,1340,378,  1,284,402,442,1214,472,1216,472,1246,440,1222,490,1222,440,1246,444,1218,492,1194,494,1192,1314,7122,1316,404,1310,376,468.  1220,440,1218,470], "RawDataInfo": [81,81,0]}}

 My goal is to use buttons 1 and 3 from the fan that have a raw value.
 How can I use the raw values ​​to send.
 I just do not understand.
 Sometimes I get the RawDataInfo at the end and sometimes ...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

My goal is to use buttons 1 and 3 from the fan that have a raw value.
 How can I use the raw values ​​to send.
 I just do not understand.
 Sometimes I get the RawDataInfo at the end and sometimes ...
[image]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Cosmicbase, 2020-06-28:
I have found a solution. I limited the raw output to 47 blocks of numbers. 
Here is my solution: 

**Light on** 
irsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1FB} 
**Light off** 
irsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1EF} 
**fan off** 
irsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1F7} 
**fan level 1** 
irsend 0.1306,382,1302,384,464,1222,486,1204,460,1250, 438,1224,460,1224,464,1224,466,1222,462,1250,440,1218,1308,7156,1310,354,1304,380,462,1226,460,1228,462,1248,434,1230, 462,1248,440,1222,462,1226,462,1224,496,1190,1306 
**or** 
irsend 0.1314,376,1308,376,464,1222,518,1170,492,1192,468,1220,464,1222 , 462,1224,438,1250,466,1220,468,1220,1306,7132,1304,382,1308,380,462,1224,464,1222,486,1202,460,1226,466,1222,462,1224 , 464,1224,440,1248,462,1224,1330 
**fan level 2** 
irsend {"Protocol": "SYMPHONY", "Bits": 11, "Data": 0x1FD} 
**fan level 3** 
irsend 0.1279,383,1301 , 362,469,1194,469,1194,447,1215,1279,383,447,1194,469,1194,469,1194,447,1215,1301,362,1279,7125,1279,383,1301,362,447,1215,447 , 1215,447,1215,1301,362.46 9,1194,469,1194,469,1194,469,1194,1301,362,1301

Jason2866, 2020-06-28:
Tasmota uses this lib https://github.com/crankyoldgit/IRremoteESP8266 to decode
IR Signals. No IR decoding is done in Tasmota. 
For adding your code you have to open a issue in the IRremoteESP8266 github

closed 2020-06-28 by ascillato2

aserraric, 2020-07-04:
Here are the IR codes I am using successfully for a Westinghouse fan, provided in the hope that they will be useful to someone in the future (these are MQTT commands, but the values are the same if used directly).

Power-Button:
`"cmnd/{devicename}/IRSend",` `"0,1298,371,1298,397,450,1219,450,1219,477,1192,477,1192,450,1192,477,1192,1298,371,477,1192,477,1192,477,8003,1298,371,1298,397,450,1219,477,1192,477,1192,450,1219,450,1219,450,1192,1298,397,477,1192,477,1192,477,26"`

Dim-Button: 
`"cmnd/{devicename}/IRSend",` `"0,1298,397,1298,371,477,1192,477,1192,477,1192,450,1192,1298,371,477,1192,477,1192,503,1192,477,1192,477,8003,1298,397,1298,397,477,1192,477,1192,477,1192,450,1192,1298,397,477,1192,450,1219,477,1192,450,1192,477,26"`

Note: For reliable switching of the light (regardless of what state it is in currently) I send `Dim` followed by `Power` to turn the light off, and `Power` followed by `Dim` to turn it on. A delay of of a few 100 milliseconds is necessary between commands for the receiver to distinguish them.

Fan-Off:
`"cmnd/{devicename}/IRSend",` `"0,1298,371,1298,371,450,1192,477,1192,477,1192,477,1192,477,1192,1298,371,450,1219,477,1192,503,1192,477,8003,1298,371,1298,397,477,1192,477,1192,477,1192,477,1192,450,1192,1325,371,450,1219,477,1192,477,1192,477,26"`

Fan-Level1:
`"cmnd/{devicename}/IRSend",` `"0,1298,397,1272,397,450,1219,477,1192,477,1192,477,1192,503,1192,503,1192,477,1192,477,1192,503,1192,1298,7155,1298,397,1298,371,450,1219,477,1192,477,1192,477,1192,477,1192,477,1192,477,1192,477,1192,477,1192,1298,26"`

Fan-Level2:
`"cmnd/{devicename}/IRSend",` `"0,1298,397,1298,371,477,1192,477,1192,477,1192,450,1219,477,1192,477,1192,477,1192,1298,371,477,1192,477,8003,1325,371,1298,371,477,1192,477,1192,424,1219,477,1192,450,1219,450,1219,450,1219,1272,397,477,1192,450,26"`

Fan-Level3:
`"cmnd/{devicename}/IRSend",` `"0,1298,397,1298,371,477,1192,477,1192,450,1219,1298,397,450,1192,477,1192,477,1192,450,1219,1298,371,1298,7181,1298,397,1298,371,450,1219,450,1219,450,1219,1298,371,477,1192,477,1192,477,1192,477,1192,1298,371,1298,26"`

Cosmicbase, 2020-07-05:
@aserraric 

Thanks for your information.
What type of fan do you have?

aserraric, 2020-07-05:
@Cosmicbase 

My fan (a Westinghouse Kingston) didn't come with a remote, I used this kit to retrofit remote control: https://www.conrad.de/de/p/westinghouse-78792-deckenventilator-620621.html

Cosmicbase, 2020-07-05:
OK Thanks,
i have the same IR Remote and the same KIT. But my IR Kit is included.