arendst/Tasmota #8813: Tasmota doesn't start on power-up with connected PN532

ChrisVonFrosta, 2020-06-28:
### PROBLEM DESCRIPTION
Tasmota on NodeMCU doesn't start on power up if the PN532 Board ist connected. Reset button doesn't help. i have to disconnect the modul before power up. Than Tasmote starts. Connect the PN532, but since the PN532 Module was disconected on power up, i need to restart. than it works fine. 
But every time the NodeMCU was power off, it will not start if the PN532 is connected on power on.

I tested several power supplies. All the same. I am not sure if this was the case before V8.3.1.  At the begin of testing this PN532 Modul some days ago, i did an update on latest.

The Pn532 module get power from the 3V Pin on the NodeMCU. This should be a pretty small current.

I checked the serial output.  If the module is connected (doesn't start) there is no sensefull output

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NodeMCU V3
- [ ] Tasmota binary firmware version number used:  8.3.1.6(tasmota)
- [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: plattformIO
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:04:02 CMD: Backlog Template; Module; GPIO 255
21:04:02 SRC: WebConsole from 192.1xxxxxx
21:04:02 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
21:04:02 SRC: Backlog
21:04:02 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
21:04:02 MQT: stat/tasmota_5FB8B8/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:04:03 SRC: Backlog
21:04:03 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
21:04:03 MQT: stat/tasmota_5FB8B8/RESULT = {"Module":{"18":"Generic"}}
21:04:03 SRC: Backlog
21:04:03 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
21:04:03 MQT: stat/tasmota_5FB8B8/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"21":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Buzzer"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"139":"PN532 Rx"},"GPIO14":{"7":"WS2812"},"GPIO15":{"138":"PN532 Tx"},"GPIO16":{"0":"None"}}
21:04:03 CMD: _

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2;`:
```
 21:06:19 CMD: Backlog Rule1; Rule2; Rule3
21:06:19 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Rule1; Rule2; Rule3"
21:06:19 SRC: Backlog
21:06:19 CMD: Gruppe 0, Index 1, Befehl "RULE", Daten ""
21:06:19 MQT: stat/tasmota_5FB8B8/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":59,"Free":452,"Rules":"ON EVENT#PN532_UID DO Backlog Color1 12; ruletimer1 1 ENDON"}
21:06:19 SRC: Backlog
21:06:19 CMD: Gruppe 0, Index 2, Befehl "RULE", Daten ""
21:06:19 MQT: stat/tasmota_5FB8B8/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"ON rules#timer=1 DO Color1 3 ENDON"}
21:06:20 SRC: Backlog
21:06:20 CMD: Gruppe 0, Index 3, Befehl "RULE", Daten ""
21:06:20 MQT: stat/tasmota_5FB8B8/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
Output power-on disconnected PN532
[image]

Output power-on connected PN532
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-28:
Hi,

Sorry but it is not a Tasmota issue. Your PN532 is connected to a GPIO that is used for boot mode selection of the ESP8266 chip.

Check the docs at https://tasmota.github.io/docs/Peripherals/#restrictions

and the guide to the ESP8266 chip at https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html

Fast solution: avoid using GPIOs: 0, 1, 2, 6-11, 15 and 16, or make the electronic adaptation for using them.

closed 2020-06-28 by ascillato2

ChrisVonFrosta, 2020-06-29:
Don’t say sorry, you show me the way ;)

I will check this soon.

Why you mark this as dublicate?

ascillato2, 2020-06-29:
> Why you mark this as dublicate?

Because this was addressed before in an older issue but I couldn't find it either. So, the label was removed.