arendst/Tasmota #8820: Make PWM CT (setoption92) available for 5-channel PWM RGBCCT bulbs

curlyel, 2020-06-29:
**Have you looked for this feature in other issues and in the docs?** 

According to the docs, `setoption92 `provides the feature of controlling color temperature CT in addition to brightness instead of controlling a cold white _and_ a warm white channel independently. This feature "PWM CT" is currently available for 2-channel CCT bulbs.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

The same feature PWM CT is required for RGBCCT bulbs without separate cold white/warm white channels on the "white part" but with a CT channel + brightness channel instead. (Example: https://templates.blakadder.com/lvwit_A70_1521lm.html). The CT of this bulb is controlled by GPIO13 while brightness for white is controlled by GPIO5.

This means, the bulb can at least be controlled e.g. via:

`color #RRGGBBCxWW`

But "`dimmer`" or "`ct`" commands will lead to unexpected results, since it changes PWM4 and PWM5 simultanously

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I guess, for such devices the functionality of `setoption92` needs to be extended to RGBCCT too.
Means, the second PWM for the white part (PWM5) should be treated as  as CT (Color Temp) instead of warm white.

eg.:
```
setoption92 0 ... separate cold/warm channels
setoption92 1 ... treat PWM1 as brightness and PWM2 as CT
setoption92 2 ... treat PWM4 as brightness and PWM5 as CT
```

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Well, the RGB part _and_ the White/CCT part _can_ be controlled by the use of the "`color`" command.
So the pain here is limited somehow. Though, it would be nice, if the other commands like "`dimmer`", "`white`" and "`ct`" can be used as well...

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-06-29:
I don't understand your problem. `SetOption92 1` does work for 5-channel bulbs. PMW4 is brightness and PWM5 is CT.

So in your example, set GPIO13 to PWM4, and GPIO5 to PWM4. If CT is reversed, set GPIO13 to PWM4i.

curlyel, 2020-06-29:
Just found a remark:

... or SetOption92 1 (**supported since v.8.2.0.5**)

Seems that I simply need to upgrade ;-)
Thanks for pointing this out!

closed 2020-06-29 by curlyel

curlyel, 2020-06-30:
> `SetOption92 1` does work for 5-channel bulbs

Just upgraded to 8.3.1.6 and can confirm it's working.
And indeed: CT was reversed, so I needed to configure GPIO3 as PWM5i.

Again, thanks for pointing me in the right direction...

poseidogithub, 2021-01-04:
> 
> 
> > `SetOption92 1` does work for 5-channel bulbs
> 
> Just upgraded to 8.3.1.6 and can confirm it's working.
> And indeed: CT was reversed, so I needed to configure GPIO3 as PWM5i.
> 
> Again, thanks for pointing me in the right direction...

Hello

do you mind helpme with the configuration? Im really lost.
according to blackadder, the template is: 
- https://templates.blakadder.com/lvwit_A70_1521lm.html
- SetOption92 1 to enable PWM CT
- and you say that I have to change GPIO3 as PWM5i.

this is not working for me, or I'm doing it wrong... this is my template config
[image]

can you please help me?

curlyel, 2021-01-05:
Sorry, there was a typo in my former post. I've corrected that above.... 

Just checked, **GPIO13 is configured as PWM5i(50)**  on my side here. This is my full config:
[image]

And as said, SetOption92=1:

```
07:49:05 CMD: SetOption92
07:49:05 RSL: stat/tasmota/RESULT = {"SetOption92":"ON"}
```

To be complete, I'm on 8.3.1.6 with these bulbs:

[image]

s-hadinger, 2021-01-05:
Use latest 9.2.0.2 and see here for VirtualCT: https://github.com/arendst/Tasmota/pull/10311