arendst/Tasmota #8826: Rule memX/varX not triggered for X>8

curzon01, 2020-06-30:
### PROBLEM DESCRIPTION
`Mem<x>#State` / `Var<x>#State` in rules do not trigger if `<x>` > 8

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1
- [x] Tasmota binary firmware version number used: 8.3.1.6(a44f82d-tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
{"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
11:31:33 MQT: stat/tasmota_2B3341/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":213,"Free":298,"Rules":"on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon"}
11:31:33 MQT: stat/tasmota_2B3341/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:31:34 MQT: stat/tasmota_2B3341/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
11:31:56 MQT: stat/tasmota_2B3341/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_2B3341","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:21","StartupUTC":"2020-06-30T09:28:35","Sleep":50,"CfgHolder":4617,"BootCount":63,"BCResetTime":"2020-05-04T13:43:23","SaveCount":325,"SaveAddress":"F7000"}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(a44f82d-tasmota)","BuildDateTime":"2020-06-30T09:00:52","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"419/699"}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"syslog.int.prsolution.eu","LogPort":514,"SSId":["<_=_>","<*_*>"],"TelePeriod":300,"Resolution":"558180E4","SetOption":["0020C229","2805C8000100060000005A00000000000000","400000CC","00006800"]}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS5 = {"StatusNET":{"Hostname":"tasmota_2B3341-4929","IPAddress":"172.18.10.235","Gateway":"172.18.10.4","Subnetmask":"255.255.255.0","DNSServer":"172.18.10.4","Mac":"3C:71:BF:2B:33:41","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.int.prsolution.eu","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2B3341","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS7 = {"StatusTIM":{"UTC":"2020-06-30T09:31:56","Local":"2020-06-30T11:31:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:17","Sunset":"21:37"}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS10 = {"StatusSNS":{"Time":1593509516,"Shutter1":{"Position":100,"Direction":0,"Target":100}}}
11:31:56 MQT: stat/tasmota_2B3341/STATUS11 = {"StatusSTS":{"Time":1593509516,"Uptime":"0T00:03:21","UptimeSec":201,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"<*_*>","BSSId":"9C:C7:A6:46:A8:A3","Channel":4,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
12:40:52 CMD: Weblog 4
12:40:52 MQT: stat/tasmota_2B3341/RESULT = {"WebLog":4}
12:40:52 CFG: Saved to flash at FB, Count 329, Bytes 4096
12:40:55 CMD: rule1 on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon
12:40:55 SRC: WebConsole from 172.18.10.42
12:40:55 CMD: Group 0, Index 1, Command "RULE", Data "on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon"
12:40:55 RUL: Stored uncompressed, would compress from 213 to 75 (-65%)
12:40:55 MQT: stat/tasmota_2B3341/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":213,"Free":298,"Rules":"on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon"}
12:40:58 WIF: Checking connection...
12:40:59 CMD: Backlog Rule1 1;Rule1 4;Rule2 0;Rule3 0
12:40:59 SRC: WebConsole from 172.18.10.42
12:40:59 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1 1;Rule1 4;Rule2 0;Rule3 0"
12:40:59 SRC: Backlog
12:40:59 CMD: Group 0, Index 1, Command "RULE", Data "1"
12:40:59 MQT: stat/tasmota_2B3341/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":213,"Free":298,"Rules":"on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon"}
12:40:59 SRC: Backlog
12:40:59 CMD: Group 0, Index 1, Command "RULE", Data "4"
12:40:59 MQT: stat/tasmota_2B3341/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":213,"Free":298,"Rules":"on mem1#state DO var1 %mem1% endon on mem4#state DO var1 %mem4% endon on mem8#state DO var1 %mem8% endon on mem9#state DO var1 %mem9% endon on mem12#state DO var1 %mem12% endon on mem16#state DO var1 %mem16% endon"}
12:40:59 SRC: Backlog
12:40:59 CMD: Group 0, Index 2, Command "RULE", Data "0"
12:40:59 MQT: stat/tasmota_2B3341/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:40:59 SRC: Backlog
12:40:59 CMD: Group 0, Index 3, Command "RULE", Data "0"
12:40:59 MQT: stat/tasmota_2B3341/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:41:14 CMD: Var1 "
12:41:14 SRC: WebConsole from 172.18.10.42
12:41:14 CMD: Group 0, Index 1, Command "VAR", Data """
[... 69 lines cut ...]
```
Test changing mem1, mem4, mem8, mem9, mem12, mem16
only changes on Mem with `<x>` <= 8 is reflected in Var1

Same when using `Var<x>` as trigger
```
Rule1 
	on var1#state DO var2 %var1% endon
	on var4#state DO var2 %var4% endon
	on var8#state DO var2 %var8% endon
	on var9#state DO var2 %var9% endon
	on var12#state DO var2 %var12% endon
	on var16#state DO var2 %var16% endon
```
Test changing var1, var4, var8, var9, var12, var16
Only changes on Var with `<x>` <= 8 is reflected in Var2

### EXPECTED BEHAVIOUR
Trigger for all possible Var/Mem 1 to 16

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-30:
Nice one!

I think I've found a solution. Working on it.

arendst, 2020-06-30:
Pls retry.

curzon01, 2020-06-30:
unfortunately not, that would be too easy :wink:  - this happens also for VarX triggers X>8 and this was already uint16_t

curzon01, 2020-06-30:
solved - (it works better with enabled rules :wink:)

closed 2020-06-30 by curzon01