arendst/Tasmota #8827: Supported I2C device TSL 2561 / GY-2561 (Lux Sensor)

cniweb, 2020-06-30:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
not related to a problem

**Describe the solution you'd like**  
I would like to be able to read the I2C device TSL 2561 / GY-2561

**Describe alternatives you've considered**  
Self programmed with Arduino Library:
https://github.com/adafruit/Adafruit_TSL2561

**Additional context**  
Datasheet:
https://cdn-shop.adafruit.com/datasheets/TSL2561.pdf

Thanks,
Chris

sfromis, 2020-06-30:
I'm already using that sensor with Tasmota, no problem. You just have to flash tasmota-sensors.bin, or build your own binary if you  have special needs.

closed 2020-06-30 by ascillato2

cniweb, 2020-07-01:
@sfromis How can I use it? Unfortunately I didn't find any documentation.

sfromis, 2020-07-01:
There are no special steps to use it, other than the very general setup of having flashed with tasmota-sensors.bin, configuring two pins for the I2C bus (I used the pretty standard gpio4=sda and gpio5=scl, but other pins mostly work too), and wiring the sensor appropriately. The reading appears on the Tasmota front page like `TSL2561 Illuminance 13.650 lx` , and in the published MQTT messages, along any other sensors present.

cniweb, 2020-07-01:
[image]
I configured, but no Sensor output is on he Tasmota front page  :worried:

sfromis, 2020-07-02:
If you can't get it to work with right binary, you may have to do the usual stuff to hunt down issues, like triple-checking wiring and swap parts with known good parts, like if you have another I2C sensor to test with.

Command I2CScan allows you to check what devices Tasmota sees on the I2C bus, and command I2CDriver lets you check what drivers are supported by your binary. Number 16 has to be in the list for the TSL2561 to be supported. Right after boot, Tasmota will also report what devices it found on the I2C bus, check the console. There may be some risk of conflict if you also have other I2C devices wired up.

cniweb, 2020-07-02:
OK, I check...

Binary is tasmato-sensor.bin (Ver. 8.3.1)

**I2CScan**
```
21:13:14 CMD: I2CScan
21:13:14 MQT: stat/multisensor/RESULT = {"I2CScan":"Device(s) found at 0x39"}
```
**I2CDriver**
```
21:13:39 CMD: I2CDriver
21:13:39 MQT: stat/multisensor/RESULT = {"I2CDriver":"7"}
```
Tasmota load I2C Driver number 7:

 xnrg_07 | ADE7953 | 0x38 | Energy monitor
-- | -- | -- | --

:raised_eyebrow:

sfromis, 2020-07-02:
You have absolutely not flashed tasmota-sensors.bin. You may have tried, but failed. Read this:
https://tasmota.github.io/docs/Upgrading

cniweb, 2020-07-02:
OK, I flashed tasmota-sensors.bin and it works! 👍 

[image]

Thanks

TheBurgundyBear, 2021-11-01:
> I2CDriver

@sfromis, I'm having a similar issue as @cniweb. I2CDriver outputs:

`{"I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48"}`

How do I get 16??

sfromis, 2021-11-01:
In the long time since then, the precompiled setup for [builds](https://github.com/arendst/Tasmota/blob/development/BUILDS.md) has changed to no longer have TSL2561 included in tasmota-sensors. To use the sensor with current Tasmota, you'll have to [compile](https://tasmota.github.io/docs/Compile-your-build/) your own binary adding `#define USE_TSL2561` to the base "tasmota" build environment, do not try to add to tasmota-sensors.