arendst/Tasmota #8828: Restart Reason: Exception with 8.3.1.6

richard-scott, 2020-07-01:
### PROBLEM DESCRIPTION

After upgrading a Sonoff RF Bridge from 8.3.1 to 8.3.1.6 I get a restart due to an exception at around 9 seconds after booting.
Boot Count doesn't go up either. It just restarts.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RF Bridge, and a Wemos D1 Mini
- [x] Tasmota binary firmware version number used: tasmota.bin.gz on 8.3.1.6 on the Sonoff, and tasmota-ir.bin.gz on the Wemos
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Upgrade from 8.3.1 to 8.3.1.6

### EXPECTED BEHAVIOUR

System should not restart

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
I get this in the console:
```
00:00:00 CFG: Loaded from flash at FB, Count 4934
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota IR 01 Version 8.3.1.6(8ff609b-ir)-2_7_1
00:00:00 WIF: Checking connection...
...
00:00:05 WIF: Connected
00:00:06 HTP: Web server active on tasmota-ir-01 with IP address 192.168.0.92
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tasmota-ir-01/tele/LWT = Online (retained)
00:00:07 MQT: tasmota-ir-01/cmnd/POWER = 
00:00:07 MQT: Subscribe to tasmota-ir-01/cmnd/#
00:00:07 MQT: Subscribe to tasmotas/cmnd/#
00:00:07 MQT: Subscribe to cmnd/DVES_7058A4_fb/#
00:00:07 MQT: tasmota-ir-01/tele/INFO1 = {"Module":"Generic","Version":"8.3.1.6(8ff609b-ir)","FallbackTopic":"cmnd/DVES_7058A4_fb/","GroupTopic":"tasmotas/cmnd/"}
00:00:07 MQT: tasmota-ir-01/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-ir-01","IPAddress":"192.168.0.92"}
00:00:07 MQT: tasmota-ir-01/tele/INFO3 = {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["fffefcf8","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40281539","4021c719","40100f34","40249891","4024c7ed","402498d4","40100a62","40249948","40240745","40210bb9","40249948","40244676","40216812","40249948","40204c24","4021c37c","40216812","40202364","40279224","40206e0e","4021bb40","40204c24","4021bb40","4024aa29","4024a0f2","4024a028","4024ab54","4024a0f2","4028422c","4028422c","4028422d"]}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-01:
Can not reproduce.

```00:00:00 CFG: Loaded from flash at FB, Count 13
00:00:00 Project tasmota Tasmota Version 8.3.1.6(8ff609b-tasmota)-2_7_1
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN Channel 9 BSSId 8C:53:C3:85:17:96 in mode 11N as sonoff-D43BE2-7138...
00:00:02 WIF: Connected
12:18:33 HTP: Web server active on sonoff-D43BE2-7138 with IP address 192.168.2.149
12:18:34 MQT: Attempting connection...
12:18:34 MQT: Connected
12:18:34 MQT: tele/sonoff-D43BE2/LWT = Online (retained)
12:18:34 MQT: cmnd/sonoff-D43BE2/POWER = 
12:18:34 MQT: tele/sonoff-D43BE2/INFO1 = {"Module":"Generic","Version":"8.3.1.6(8ff609b-tasmota)","FallbackTopic":"cmnd/sonoff-D43BE2_fb/","GroupTopic":"cmnd/sonoffs/"}
12:18:34 MQT: tele/sonoff-D43BE2/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-D43BE2-7138","IPAddress":"192.168.2.149"}
12:18:34 MQT: tele/sonoff-D43BE2/INFO3 = {"RestartReason":"External System"}
12:18:38 MQT: tele/sonoff-D43BE2/STATE = {"Time":"2020-07-01T12:18:38","Uptime":"0T00:00:10","UptimeSec":10,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"8C:53:C3:85:17:96","Channel":9,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:04"}}
12:19:46 CMD: teleperiod 60
12:19:46 MQT: stat/sonoff-D43BE2/RESULT = {"TelePeriod":60}
12:19:47 MQT: tele/sonoff-D43BE2/STATE = {"Time":"2020-07-01T12:19:47","Uptime":"0T00:01:19","UptimeSec":79,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"8C:53:C3:85:17:96","Channel":9,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
```

Jason2866, 2020-07-01:
You get exception 9 if no NTP server can be reached. Since you have no valid time, i think thats the reason

richard-scott, 2020-07-01:
Hmmm NTP... so why does it work fine in 8.3.1, but not with 8.3.1.6?

richard-scott, 2020-07-01:
Actually NTP wasn't working on 8.3.1 either, but at least it didn't restart.
Anyway, a "restart 6" fixed it :-)

richard-scott, 2020-07-03:
Actually, my "wipe and restart" didn't help.

arendst, 2020-07-03:
Cannot reproduce with NTP:
```
00:00:00 CFG: Loaded from flash at FB, Count 367
00:00:00 SER: Set to 8N1 19200 bit/s
00:00:00 Project tasmota Bridge2 Version 8.3.1.6(theo)-STAGE
00:00:04 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as bridge2...
00:00:05 WIF: Connected
11:15:57 HTP: Web server active on bridge2 with IP address 192.168.2.169
11:15:58 MQT: Attempting connection...
11:15:58 MQT: Connected
11:15:58 MQT: tele/bridge2/LWT = Online (retained)
11:15:58 MQT: cmnd/bridge2/POWER = 
11:15:58 MQT: tele/bridge2/INFO1 = {"Module":"Sonoff Bridge","Version":"8.3.1.6(theo)","FallbackTopic":"cmnd/DVES_825686_fb/","GroupTopic":"cmnd/sonoffs/"}
11:15:58 MQT: tele/bridge2/INFO2 = {"WebServerMode":"Admin","Hostname":"bridge2","IPAddress":"192.168.2.169"}
11:15:58 MQT: tele/bridge2/INFO3 = {"RestartReason":"Software/System restart"}
11:16:02 MQT: tele/bridge2/STATE = {"Time":"2020-07-03T11:16:02","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-15,"LinkCount":1,"Downtime":"0T00:00:07"}}
```
or without NTP:
```
00:00:00 CFG: Loaded from flash at F6, Count 373
00:00:00 SER: Set to 8N1 19200 bit/s
00:00:00 Project tasmota Bridge2 Version 8.3.1.6(theo)-STAGE
00:00:04 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as bridge2...
00:00:05 WIF: Connected
00:00:06 HTP: Web server active on bridge2 with IP address 192.168.2.169
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/bridge2/LWT = Online (retained)
00:00:07 MQT: cmnd/bridge2/POWER = 
00:00:07 MQT: tele/bridge2/INFO1 = {"Module":"Sonoff Bridge","Version":"8.3.1.6(theo)","FallbackTopic":"cmnd/DVES_825686_fb/","GroupTopic":"cmnd/sonoffs/"}
00:00:07 MQT: tele/bridge2/INFO2 = {"WebServerMode":"Admin","Hostname":"bridge2","IPAddress":"192.168.2.169"}
00:00:07 MQT: tele/bridge2/INFO3 = {"RestartReason":"Software/System restart"}
00:00:11 MQT: tele/bridge2/STATE = {"Time":"1970-01-01T00:00:11","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-17,"LinkCount":1,"Downtime":"0T00:00:07"}}
```
Your exception 0 might be caused by a power supply issue.

arendst, 2020-07-03:
I did some further testing and received exception 3 in some cases where PROGMEM data was checked for valid JSON included recently.

The latest commit should solve this. Pls have a try if this solves your exception 0 too.

richard-scott, 2020-07-03:
OK, I've tested the [archives ](http://thehackbox.org/tasmota/archive/) and my issues start somewhere after [cc301e9](http://thehackbox.org/tasmota/archive/20200702-190128-cc301e9-tasmota.bin)

I'll need to wait for that site to update to test the next commit.

Jason2866, 2020-07-03:
You find the firmware files here too https://github.com/arendst/Tasmota/tree/firmware
NO OTA update

richard-scott, 2020-07-03:
I've tried the ones from the tree folder you suggested and I get a restart due to the Hardware Watchdog this time.
If I downgrade back to 8.3.1 again, all is stable. Same Wemos D1 mini board and same Amazon 9W USB Mains power adapter.

Jason2866, 2020-07-03:
Wemos D1 mini ?  Most of them have a weak 3.3V vreg. Many boards do not run reliable since Tasmota generates now and than heavy cpu load. The vreg cant deliver the needed current / voltage 
Do you have a NodeMCU? There is in general a vreg which can deliver enough current on the 3.3V path.
Unstable 3.3V results in various strange errors / issues

meingraham, 2020-07-03:
@richard-scott 

Your question as to why one version is stable and the newer version is not could be due to the "load" the firmware logic differences place on the device by having added new functionality to the Tasmota code.

Perhaps as a test you can try a binary with only the required options needed for operation. Try using the latest `tasmota-lite.bin` on the Wemos D1 Mini to see if the problem remains. For the RF Bridge, compile tasmota-lite but add back `#define USE_SONOFF_RF` in `user_config_override.h`.

This isn't a foolproof test for power issues as not only is it reducing the features, but it may also be eliminating any new code that could be the source of your issue. But at least it will begin to try to isolate where your problem is coming from.

arendst, 2020-07-03:
If your watchdog appears after cc301e9 it may be solved with the latest commit.

richard-scott, 2020-07-04:
@Jason2866 I have a LoLin NodeMCU v3, and with the latest [e5127d8 ](https://github.com/arendst/Tasmota/commit/e5127d8e8ed7de4ff3f2521d273116cc1d344183) firmware I get a restart reson of External System and not Power On as I used to get. This is powered from a 1A USB Mains adapter.

As for the Wemos D1 Mini... That appears to be stable now :-)

richard-scott, 2020-07-04:
I've just Tazmotized my NodeMCU with 8.3.1 and I get the same restart reson at boot time... so must be an internal issue?

Jason2866, 2020-07-04:
Do you have still execptions? Restart reason power on or external system is not a issue caused from Tasmota. This is a info that power was switched off. Whatever was the reason for

richard-scott, 2020-07-06:
Yes, i'm still getting exceptions with 8.3.1.6. I've re-flashed all my units, and configured nothing but WiFi and MQTT... i've not configured external devices. I have Sonoff Basics, they are all OK. I have Wemos D1 Mini's, all from the same batch. The D1's running ESPHome are ok, all the Tasmota ones have exceptions.

I have two PZEM-016T's connected to its own D1 Mini via a RS485->TTL board monitoring power. They are OK. All from the same batch as the ones with exceptions.

My main issue was a Wemos D1 Mini running as an IR Sender. IR Led's on their own 5V rail being triggered by a 2n222a transistor. That crahses around 10 sec after booting, even when the Wems isn't in the PCB headers for the IR Led's i.e. with no devices connected to it other than USB power.

I have just upgraded everything to 8.4.0 and everything is much much quicker, and seems to be stable so far.
I'll keep an eye on it.

richard-scott, 2020-07-06:
Ok, small update. The Sonoff RF Bridge seems stable, so we can forget about that...

However, the Wemos D1 running IR still has an exception issues with 8.4.0 running tasmota-ir.bin.
This unit runs OK, with 8.4.0 tasmota-lite.bin and seems stable with tasmota.bin.

This is with the same Wemos D1, flashed/erased via Tasmota with 8.3.1 tasmota.bin and the upgraded to 8.4.0 OTA to tasmota-lite.bin, and then tasmota.bin. When I then use tasmota-ir.bin I get exception.

```
11:34:01 MQT: tasmota-ir-01/tele/INFO1 = {"Module":"Generic","Version":"8.4.0(d612d2b-ir)","FallbackTopic":"cmnd/DVES_7058A4_fb/","GroupTopic":"tasmotas/cmnd/"}
11:34:01 MQT: tasmota-ir-01/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota-ir-01","IPAddress":"192.168.0.92"}
11:34:01 MQT: tasmota-ir-01/tele/INFO3 = {"RestartReason":{"Exception":2,"Reason":"Exception","EPC":["3ffe9297","00000000","00000000"],"EXCVADDR":"3ffe9294","DEPC":"00000000","CallChain":["40281519","4021c701","40100f34","40249879","4024c7d5","402498bc","40100a62","40249930","4023ff81","40210bb5","40249930","4024465e","402167fe","40249930","40204c24","4021c364","402167fe","40202364","4027920c","40206e0a","4021bb28","40204c24","4021bb28","4024aa11","4024a0da","4024a010","4024ab3c","4024a0da","4028420c","4028420c","4028420d"]}}
11:34:01 HTP: Console
```

Jason2866, 2020-07-06:
You have a hardware problem. The D1 mini is unreliable. Probably the onboard 3.3 Vreg.
Tasmota with less functions compiled in needs less CPU power (= less current).
This behaviour is well known...

richard-scott, 2020-07-06:
ok, thanks for the update.

closed 2020-07-06 by richard-scott

richard-scott, 2020-07-15:
Hmmm, same hardware with v8.4.0 is working just fine today... well, apart from this #8913 issue.

richard-scott, 2020-08-06:
Doesn't seem to have been a hardware problem after all... v 8.4.0 is rock solid on my "problem" devices :-)