arendst/Tasmota #8831: Reboot by hardware watchdog when using both switches and a counter and a rule on tele-counter#c1>0

FransOv, 2020-07-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Robotdyn mini D1
- [ x] Tasmota binary firmware version number used: Version 8.3.1(tasmota)-2_7_1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tasmota-PyFlasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:47:21 MQT: stat/tasmota-s4/RESULT = {"NAME":"Generic","GPIO":[255,255,56,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:47:21 MQT: stat/tasmota-s4/RESULT = {"Module":{"0":"Generic"}}
15:47:21 MQT: stat/tasmota-s4/RESULT = {"GPIO0":{"14":"Switch6"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"9":"Switch1"},"GPIO5":{"10":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"11":"Switch3"},"GPIO13":{"12":"Switch4"},"GPIO14":{"13":"Switch5"},"GPIO15":{"42":"Counter1"},"GPIO16":{"57":"Led2i"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
15:48:27 MQT: stat/tasmota-s4/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":210,"Free":301,"Rules":"on switch1#state do status 8 endon on switch2#state do status 8 endon on switch3#state do status 8 endon on switch4#state do  status 8 endon on switch5#state do status 8 endon on switch6#state do status 8 endon"}
15:48:27 MQT: stat/tasmota-s4/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":42,"Free":469,"Rules":"on tele-counter#c1>0>0 do counter1 0 endon"}
15:48:27 MQT: stat/tasmota-s4/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:49:01 MQT: stat/tasmota-s4/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-s4","FriendlyName":["Tasmota-s4"],"Topic":"tasmota-s4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[2,2,2,2,2,2,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:49:01 MQT: stat/tasmota-s4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:07:11","StartupUTC":"2020-07-01T13:41:50","Sleep":50,"CfgHolder":4617,"BootCount":67,"BCResetTime":"2020-04-03T10:34:45","SaveCount":183,"SaveAddress":"F9000"}}
15:49:01 MQT: stat/tasmota-s4/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"401/699"}}
15:49:01 MQT: stat/tasmota-s4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OverkleeftAdvies_Nomap",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:49:01 MQT: stat/tasmota-s4/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
15:49:01 MQT: stat/tasmota-s4/STATUS5 = {"StatusNET":{"Hostname":"tasmota-s4","IPAddress":"192.168.2.225","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:60:67:A8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:49:01 MQT: stat/tasmota-s4/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.33","MqttPort":1883,"MqttClientMask":"tasmota-s4","MqttClient":"tasmota-s4","MqttUser":"hamqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:49:01 MQT: stat/tasmota-s4/STATUS7 = {"StatusTIM":{"UTC":"2020-07-01T13:49:01","Local":"2020-07-01T15:49:01","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:51","Sunset":"21:56"}}
15:49:01 MQT: stat/tasmota-s4/STATUS10 = {"StatusSNS":{"Time":"2020-07-01T15:49:01","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","Switch5":"OFF","Switch6":"OFF","COUNTER":{"C1":0}}}
15:49:01 MQT: stat/tasmota-s4/STATUS11 = {"StatusSTS":{"Time":"2020-07-01T15:49:01","Uptime":"0T00:07:11","UptimeSec":431,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"OverkleeftAdvies_Nomap","BSSId":"00:0B:82:AF:D3:91","Channel":4,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect any of the switches to ground. The status led on GPIO2 lights and after disconnecting the switch from ground the module reboots. Restart reason in information says: Hardware Watchdog
When rule 2 is disabled, teh switches work as expected
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A status8 message is send via MQTT whenever a switch is connected to ground. The switches are connected to an optocoupler.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-01:
Your config is faulty.
```"GPIO0":{"14":"Switch6"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"9":"Switch1"},"GPIO5":{"10":"Switch2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"11":"Switch3"},"GPIO13":{"12":"Switch4"},"GPIO14":{"13":"Switch5"},"GPIO15":{"42":"Counter1"},"GPIO16":{"57":"Led2i"}```
What should the switches do? Makes no sense. A switch on Gpio 0 is always a bad idea

curzon01, 2020-07-01:
Rule2 `on tele-counter#c1>0>0 do counter1 0 endon` - find the mistake (hint: greater than greater then...)

FransOv, 2020-07-01:
Already saw the mistake in the rule, copy paste error. That was not the cause however. 

The switches sense whether the valves and relay of the garden irrigation system are activated as they can be operated both from the home automation system and the irrigation computer. Switch6 is not really needed.
I just tried the same configuration on a NodeMcu V3, without any peripherals attached. Withe rule2 (corrected) enabled the system reboots triggered by the hardware watchdog whenever I connect one of the switches to ground. Actually after I break the  connection to ground. Rule 1 is not triggered. With rule2 disabled, rule1 triggers whenever I connect one of the switches to ground an also when I disconnect it.

curzon01, 2020-07-01:
also tried the latest dev bin 8.3.1.6?

JackNJack, 2020-07-02:
I have similar problem, I have button#state rule trigger. My dev bin were #Ifthenelse enabled and Setoption73 enabled. Whenever I enable Rule the device intermittently reboots whenever the button#state triggers. Sometimes it dumps stack, sometime it will only reboot due to watchdog reset = 4.  If rule is disabled the device will normally works without issue or reboot.

FransOv, 2020-07-02:
Tried with yesterdays version 8.3.1.6(618d860-tasmota) file: 20200701-110120-618d860-tasmota.bin. Looks like problem solved, no more spontaneous reboots. Will test some more and report on this later today.

FransOv, 2020-07-02:
Tested in the original configuration with optocoupler attached. Works OK with version 8.3.1.6. Will close the issue.

closed 2020-07-02 by FransOv

JackNJack, 2020-07-02:
my issue also resolved by version 8.3.1.6... Thank you guys!