arendst/Tasmota #8832: Can

wassy92x, 2020-07-01:
### PROBLEM DESCRIPTION
Hello, I have a problem with the DevGroupSend-Command. If I try to send for example the following command: DevGroupSend 193=AnyCmd the receiving device reboots sometimes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): MagicHome
- [x] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VS Code and Platformio
- [x] Flashing tools used: Original Platformio and updated by ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
16:01:53 MQT: stat/Schrankbeleuchtung/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:01:53 MQT: stat/Schrankbeleuchtung/RESULT = {"Module":{"34":"MagicHome"}}
16:01:53 MQT: stat/Schrankbeleuchtung/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"39":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"40":"PWM4"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS = {"Status":{"Module":34,"DeviceName":"Schrankbeleuchtung Wohnzimmer","FriendlyName":["Schrankbeleuchtung Wohnzimmer"],"Topic":"Schrankbeleuchtung","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"BadStimmungslicht","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:56","StartupUTC":"2020-07-01T14:57:39","Sleep":200,"CfgHolder":4617,"BootCount":40,"BCResetTime":"2020-06-28T14:12:37","SaveCount":5626,"SaveAddress":"FB000"}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020.06.29 12:13:42","Boot":7,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"435/699"}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wassy-IoT-WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","02000200","00000008"]}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":448,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","87C06597","042E83A1","000000CD","010013C0","C000D981","00000024"],"Drivers":"1,2,3,4,8,9,10,12,16,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS5 = {"StatusNET":{"Hostname":"Schrankbeleuchtung-3740","IPAddress":"*******","Gateway":"******","Subnetmask":"255.255.255.0","DNSServer":"*****","Mac":"*******","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:02:35 MQT: stat/Schrankbeleuchtung/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.30.4","MqttPort":1883,"MqttClientMask":"Schrankbeleuchtung","MqttClient":"Schrankbeleuchtung","MqttUser":"*******","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:02:36 MQT: stat/Schrankbeleuchtung/STATUS7 = {"StatusTIM":{"UTC":"2020-07-01T15:02:35","Local":"2020-07-01T16:02:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:51","Sunset":"20:56"}}
16:02:36 MQT: stat/Schrankbeleuchtung/STATUS10 = {"StatusSNS":{"Time":"2020-07-01T16:02:36"}}
16:02:36 MQT: stat/Schrankbeleuchtung/STATUS11 = {"StatusSTS":{"Time":"2020-07-01T16:02:36","Uptime":"0T00:04:57","UptimeSec":297,"Heap":22,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":27,"MqttCount":1,"POWER":"OFF","Dimmer":8,"Color":"00000014","HSBColor":"0,0,0","White":8,"Channel":[0,0,0,8],"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"******","BSSId":"*******","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:00:00 Projekt tasmota Schrankbeleuchtung Wohnzimmer Version 8.3.1(tasmota)-2_7_1
00:00:00 WIF: verbinden mit AP1 Wassy-IoT-WLAN Channel 11 BSSId 02:9F:C2:F4:48:B6 in Modus 11N als Schrankbeleuchtung-3740...
00:00:01 WIF: verbunden
00:00:01 HTP: Web-Server aktiv bei Schrankbeleuchtung-3740 mit IP-Adresse 10.11.30.21
16:05:06 MQT: Verbindungsversuch...
16:05:06 MQT: verbunden
16:05:06 MQT: tele/Schrankbeleuchtung/LWT = Online (beibehalten)
16:05:06 MQT: cmnd/Schrankbeleuchtung/POWER = 
16:05:06 MQT: tele/Schrankbeleuchtung/INFO1 = {"Module":"MagicHome","Version":"8.3.1(tasmota)","FallbackTopic":"cmnd/Schrankbeleuchtung_fb/","GroupTopic":"cmnd/BadStimmungslicht/"}
16:05:06 MQT: tele/Schrankbeleuchtung/INFO2 = {"WebServerMode":"Admin","Hostname":"Schrankbeleuchtung-3740","IPAddress":"10.11.30.21"}
16:05:06 MQT: tele/Schrankbeleuchtung/INFO3 = {"RestartReason":"Software/System restart"}
16:05:06 MQT: stat/Schrankbeleuchtung/RESULT = {"POWER":"ON"}
16:05:06 MQT: stat/Schrankbeleuchtung/POWER = ON
```

### TO REPRODUCE
- Set DevGroupName on both devices
- SetOption85 1

### EXPECTED BEHAVIOUR
Device shouldn't reboot.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-07-01:
Please, update to latest version from the development branch and try again.

ascillato2, 2020-07-09:
Hi, Any news on this?

wassy92x, 2020-07-09:
I found the issue, it's based on another issue. I have setup a script and if the command is received another command is directly executed. If I execute the second command directly based on the other command, the esp8266 reboots. If I store the second command and execute it in the next 100ms-loop it works.

ascillato2, 2020-07-09:
Glad you found a solution. Please, could you share it? Thanks.

closed 2020-07-09 by ascillato2