arendst/Tasmota #8834: Sonoff S31 not showing power info

tarasglek, 2020-07-01:
Hi,
I have a sonoff s31 that I purchased a few years ago. It has been running tasmota 8.1.0 perfectly. Yesterday I received a new S31 unit and it's not showing power usage.
I tried downgrading to 8.1.0 to make sure it's not a regression, but same result.

Then I noticed that the info for device is different:
old:
[image]
new:
[image]

I set logging to 4, no errors.
Status 8 shows:
`19:36:34 RSL: stat/tasmota_1B3169/STATUS8 = {"StatusSNS":{"Time":"2020-07-01T19:36:34","ENERGY":{"TotalStartTime":"2020-07-01T18:22:37","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}`

How do I troubleshoot this further? I have 3 of these new units, they were really hard to flash, hope they can be made to work :(

closed 2020-07-01 by close-issue-app[bot]

meingraham, 2020-07-01:
There is the S31 plug with power monitoring and the S31 Lite that does not have power monitoring (only Wi-Fi switching). Form factor is identical.

https://sonoff.tech/product/wifi-smart-plugs/s31-s31lite

tarasglek, 2020-07-01:
oh that sucks :(, thank you