arendst/Tasmota #8836: SIPAILING Smart WL

SVLoneStar, 2020-07-01:
**Have you looked for this feature in other issues and in the docs?**  
I have been looking at the Wiki and have been searching Google for a solution, but without success.

**Is your feature request related to a problem? Please describe.**  
I successfully use Bitzwolf and Gosund Smart Plugs already, and now have purchased 4 plugs of type 'Sipailing Smart WL'. I was able to flash Tasmota on the plugs, I can access them via web interface, but I am not able to control them...switching ON/OFF doesn't work.

**Describe the solution you'd like**  
I'd like to know which template to assign to those plugs so that Tasmote can control them.

**Describe alternatives you've considered**  
I have been testing most of the templates included, without success.

**Additional context**  
Please let me know what any other information is required to get those plugs going...thanks!

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-07-02:
Hi SVLoneStar,
I would suggest to go to the "Unknown Device" procedure.
[See](https://tasmota.github.io/docs/Configuration-Procedure-for-New-Devices/)
After identifying the right template, it would be great if you publish the new template at Blackadder.
Good luck
Joerg

blakadder, 2020-07-02:
Answered and wasn't a bug or issue with Tasmota in the first place

closed 2020-07-02 by blakadder

SVLoneStar, 2020-07-02:
The Sipailing Smart WL is (also visually) identical to the Aisirer SWA11, found [here](https://templates.blakadder.com/aisirer_SWA11.html). Thanks for pointing me in the right direction, @schrej-zz .