arendst/Tasmota #8838: Inconsistencies in the build environment for IR firmare

eku, 2020-07-02:
### PROBLEM DESCRIPTION
I'm trying to build my own firmware for infrared remotes, tailored to my needs. Please correct the inconsistencies in the build configuration described in the following, so that other users are able to compile a ready-made firmware.

Basically there are two flavours: ```tasmota-ir``` and ```tasmota-ircustom``` (ESP32 times out of order).

In ```platformio_tasmota_env.ini``` both use the flags from ```irremoteesp_full```, where ```tasmota-ir``` still defines ```FIRMWARE_IR```. At this point one would expect ```tasmota-ircustom``` to define ```FIRMWARE_IR_CUSTOM```, as mentioned in ```my_user_config.h``` as an example. However, this define is not checked anywhere in the code and should therefore be removed from ```my_user_config.h```.

By the way, there is no ```irremoteesp``` in ```platformio.ini```. Strange. ```irremoteesp_full``` defines ```USE_IR_REMOTE_FULL```, which means that all protocols are always activated in both flavors.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): IR UFO 1
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: platformio
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Look at the files I mentioned.

### EXPECTED BEHAVIOUR
I would expect the configuration to be clearly separated and documented, so that a consistent configuration can be created with little effort.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
If you are still unsure, I recommend a ```git grep``` to ```USE_IR_REMOTE```, ```USE_IR_REMOTE_FULL```, ```irremoteesp``` and ```FIRMWARE_IR```.

I would like to submit a PR, but I don't feel comfortable with the mess myself and don't know how it should be done.

Thanks for the attention.

Jason2866, 2020-07-02:
There is nothing wrong or inconsistent in the platformio definitions. `tasmota-ir` will result in a firmware with all IR codes enabled and disabled nearly all other drivers. Only section 1 from `user_config_override.h` will be used. `tasmota-ircustom` will have all IR codes in the resulting firmware and all settings from `user_config_override.h` will be in

eku, 2020-07-03:
@Jason2866 Thanks for pointing out the CI Buldpipeline. However, I have never questioned its function in any way.

My point is, if you please read my text above again, that the used defines and build configurations are at least misnamed. Why, for example, is there no ```FIRMWARE_IR``` for tasmota-ir*?

Jason2866, 2020-07-03:
There is no need for littering code with variants which differs only in a few settings.
The end user build process is the same for all variants. There is a PlatformIO `env` for every version
[image]

eku, 2020-07-03:
@Jason2866 Would you agree that in ```platformio_tasmota_env.ini``` the entry for ```env:tasmota-ircustom``` should look like this

```
[env:tasmota-ircustom]
build_flags = ${common.build_flags} ${irremoteesp_full.build_flags} -DFIRMWARE_IRCUSTOM
```

so that you have the chance to distinguish between ```tasmota-ir``` and ```tasmota-ircustom``` in ```user_config_override.h``` and make your own settings for the latter.

Jason2866, 2020-07-04:
agree. PR done.

closed 2020-07-04 by arendst

eku, 2020-07-04:
ty

PR #8845 (2020-07-04): Add build flag -DFIRMWARE_IR_CUSTOM
to env tasmota-ir. 

## Description:

**Related issue (if applicable):** fixes #8838

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Y [... cut]