arendst/Tasmota #8840: Sonoff D1 dimmer troble vith relay

NightR1der, 2020-07-03:
### PROBLEM DESCRIPTION
I flash tasmota on sonoff D1 dimmer. I want to use GPIO5 as a relay. I make template, add user(255) on GPIO5. In configuraton mode I set GPIO5 as relay1. But I can't use relay1. When the power1 command is issued, the dimmer state changes. However, when the power2 command is given, the state of GPIO5 does not change. Sorry for my bad English.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff D1 Dimmer
- [ ] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
03:02:48 MQT: BathRoomDNode/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["BathRoomDNode","Tasmota2"],"Topic":"BathRoomDNode","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:02:48 MQT: BathRoomDNode/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-RU.bin","RestartReason":"Software/System restart","Uptime":"0T00:37:23","StartupUTC":"2020-07-02T23:25:25","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-07-03T00:19:59","SaveCount":58,"SaveAddress":"FA000"}}
03:02:48 MQT: BathRoomDNode/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:50:55","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"351/699"}}
03:02:48 MQT: BathRoomDNode/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["nr4ts",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
03:02:48 MQT: BathRoomDNode/stat/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000419","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
03:02:48 MQT: BathRoomDNode/stat/STATUS5 = {"StatusNET":{"Hostname":"BathRoomDNode-5281","IPAddress":"192.168.3.1","Gateway":"192.168.3.200","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.200","Mac":"C8:2B:96:5A:34:A1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
03:02:48 MQT: BathRoomDNode/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A34A1","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
03:02:48 MQT: BathRoomDNode/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-07-03T00:02:48","Local":"2020-07-03T03:02:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"06:52","Sunset":"22:56"}}
03:02:48 MQT: BathRoomDNode/stat/STATUS10 = {"StatusSNS":{"Time":"2020-07-03T03:02:48","Switch1":"ON","Switch2":"ON","Switch3":"OFF"}}
03:02:48 MQT: BathRoomDNode/stat/STATUS11 = {"StatusSTS":{"Time":"2020-07-03T03:02:48","Uptime":"0T00:37:23","UptimeSec":2243,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":102,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Dimmer":29,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"nr4ts","BSSId":"BA:69:F4:CC:92:82","Channel":9,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
03:04:05 CMD: power1 0
03:04:05 SRC: WebConsole from 192.168.1.11
03:04:05 CMD: Группа 0, Индекс 1, Команда "POWER", Данные "0"
03:04:05 MQT: BathRoomDNode/stat/RESULT = {"POWER1":"OFF"}
03:04:05 MQT: BathRoomDNode/stat/POWER1 = OFF
03:04:05 CFG: Сохранено во флэш-память F8, Подсчет 60, Байт 4096
03:04:11 WIF: Проверка соединения...
03:04:12 CMD: power1 1
03:04:12 SRC: WebConsole from 192.168.1.11
03:04:12 CMD: Группа 0, Индекс 1, Команда "POWER", Данные "1"
03:04:12 MQT: BathRoomDNode/stat/RESULT = {"POWER1":"ON"}
03:04:12 MQT: BathRoomDNode/stat/POWER1 = ON
03:04:12 CFG: Сохранено во флэш-память F7, Подсчет 61, Байт 4096
03:04:20 CMD: power2 1
03:04:20 SRC: WebConsole from 192.168.1.11
03:04:20 CMD: Группа 0, Индекс 2, Команда "POWER", Данные "1"
03:04:20 MQT: BathRoomDNode/stat/RESULT = {"POWER2":"ON"}
03:04:20 MQT: BathRoomDNode/stat/POWER2 = ON
03:04:26 CMD: power2 0
03:04:26 SRC: WebConsole from 192.168.1.11
03:04:26 CMD: Группа 0, Индекс 2, Команда "POWER", Данные "0"
03:04:26 MQT: BathRoomDNode/stat/RESULT = {"POWER2":"OFF"}
03:04:26 MQT: BathRoomDNode/stat/POWER2 = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
changing the relay1 status

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-04:
Any additional functionality to a Sonoff D1 is NOT supported.

Power1 is hardcoded to control D1 power.

NightR1der, 2020-07-04:
Ok, thank you.

closed 2020-07-04 by NightR1der