arendst/Tasmota #8841: Modbus slave

jeffersonpimenta, 2020-07-03:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I need a function that allows my scada software access tasmota via modbus tcp and link inputs/outputs in some address

**Describe the solution you'd like**  
A function that create a server in tasmota to access/control it via modbus slave

**Describe alternatives you've considered**  
I've considerate develop a custom software in arduino, but a function in tasmota would be nice since it contains a lot of cool functions a allow a single firmware for a multiple deploys

**Additional context**  
I have plans to use it on ScadaLTS

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-08-08 by stale[bot]

Gifford47, 2026-09-10:
Hey @jeffersonpimenta @ascillato2 !
Is this already possible or merged? I need also a master/slave function in tasmota ...

s-hadinger, 2026-09-10:
Hey @Gifford47 I received my HCP Bridge but haven't tried it yet.

I will explore what kind of options we can provide for Modbus slave in Tasmota

s-hadinger, 2026-09-10:
I also understand that Modbus slave timing is stretch and needs to be very reactive, so it does require a separate FreeRTOS task to handle Modbus responses in adequate time, which is not the way Tasmota works in most drivers. Nothing blocking but it takes more than the few lines above

Gifford47, 2026-09-10:
Yes, you're right. I think it's not possible with the current driver design. A freertos task would be the one.

s-hadinger, 2026-09-11:
Please follow up here: https://github.com/arendst/Tasmota/discussions/25023