arendst/Tasmota #8843: Please, add support very good small text LCD Winstar WO1602I, based on ST7032i controller!

alex-nemo, 2020-07-03:
**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
Please, add support very good small text LCD Winstar WO1602I, based on ST7032i controller!

**Describe the solution you'd like**  
Need to add support very good small text LCD Winstar WO1602I, based on ST7032i controller.

**Describe alternatives you've considered**  
Standard displays are too big! WO1602I is ideal for DIN-rail enclosures.

**Additional context**  
Docmentations and sample code (Keil, MK51 core) this:
WO1602I https://drive.google.com/file/d/1RXBhvPxgDdE4sLmNdwCDbs9G57Ro-QQ8/view?usp=sharing
Libraries on PlatformIO: https://platformio.org/lib/search?query=keyword%253A%2522st7032%2522
And Many Libs for ST7032i on GitHub.

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-07-04:
Hi,
this part seems to have an I2C connection.
Have you already tried to connect it via standard SDA, SCL and write to the display?
Joerg

Jason2866, 2020-07-04:
@schrej-zz  i2c device will be recognized. Still useless because there is no driver for

schrej-zz, 2020-07-04:
Ah, sorry

closed 2020-08-08 by stale[bot]