arendst/Tasmota #8853: WifiOff when updating from pre WifiOFF to latest Dev Version

kugelkopf123, 2020-07-04:
### PROBLEM DESCRIPTION
When updating from an earlier version of the 8.3.1Dev version to the current version via web interface, the ESP8266 no longer connects to the WIFI. 
It creates an AP but without capative portal. 

If you connect it via serial, it can be brought back to life with the command: "wifi 1". Other than that I haven't found any other option yet.

I think the problem was the introduction of the Wifi off option for the ESP32 with LAN connection.
This option also works for ESP8266.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):SonoffBasic,Shelly1,Sonoff4CH-R2
- [ ] Tasmota binary firmware version number used: latest.
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:14:01 CMD: Backlog Template; Module; GPIO 255
22:14:02 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
22:14:02 RSL: RESULT = {"Module":{"46":"Shelly 1"}}
22:14:02 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"82":"Switch1n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
22:14:27 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
22:14:27 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
22:14:27 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  22:13:27 RSL: STATUS = {"Status":{"Module":46,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B8BCB0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
21:56:52 CMD: ipaddress
21:56:52 RSL: RESULT = {"IPAddress1":"(IP unset) (192.168.1.209)"}
21:58:15 RSL: RESULT = {"POWER":"ON"}
21:58:15 RSL: POWER = ON
21:58:15 RSL: RESULT = {"POWER":"OFF"}
21:58:15 RSL: POWER = OFF
21:58:31 UPL: Datei shelly.bin.gz ...
21:58:35 UPL: erfolgreich 345345 bytes. starte neu
21:58:37 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
@cp:0
ld
00:00:00 CFG: aus Flash geladen am 1FB, z√§hlen 92
00:00:00 Projekt tasmota Tasmota Version 8.3.1(tasmota)-2_7_1
00:00:36 CMD: weblog 4
00:00:36 RSL: RESULT = {"WebLog":4}
00:00:39 CMD: restart 1
00:00:39 RSL: RESULT = {"Restart":"Restarting"}
00:00:41 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: aus Flash geladen am 1F9, z√§hlen 94
00:00:00 Projekt tasmota Tasmota Version 8.3.1(tasmota)-2_7_1
00:00:04 CMD: wifi 1
00:00:04 RSL: RESULT = {"Wifi":"ON"}
00:00:06 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: aus Flash geladen am 1F8, z√§hlen 95
00:00:00 APP: Some settings have been reset (2)
00:00:00 Projekt tasmota Tasmota Version 8.3.1(tasmota)-2_7_1
00:00:00 WIF: verbinden mit AP1 WIFINETWORK Channel 6 BSSId 10:9A:DD:XX:XX:XX in Modus 11N als Shelly1_3...
00:00:03 WIF: verbunden
21:59:37 HTP: Web-Server aktiv bei Shelly1_3 mit IP-Adresse 192.168.1.209
21:59:37 RSL: INFO1 = {"Module":"Shelly 1","Version":"8.3.1(tasmota)","FallbackTopic":"cmnd/DVES_B8BCB0_fb/","GroupTopic":"cmnd/tasmotas/"}
21:59:37 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"Shelly1_3","IPAddress":"192.168.1.209"}
21:59:37 RSL: INFO3 = {"RestartReason":"Software/System restart"}
21:59:37 RSL: RESULT = {"POWER":"OFF"}
21:59:37 RSL: POWER = OFF

```

### TO REPRODUCE
Update from a Version of 8.3.1 before WifiOff was possible.

### EXPECTED BEHAVIOUR
Update without Bricking the device.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

arendst, 2020-07-04:
wifi on/off was introduced in 8.3.1.4 so how come your logging shows 8.3.1?

kugelkopf123, 2020-07-04:
@arendst All devices that i updated with the latest version shows 8.3.1. Dont Know why thats like this.

kugelkopf123, 2020-07-04:
[image]
Another Device but the same thing with the Version Number.

kugelkopf123, 2020-07-04:
Here a log from another device same Problem. Updated, and Wifi Off. 
Connected via Serial:

```
00:00:00 CFG: aus Flash geladen am FB, z√§hlen 1973
00:00:00 SRC: Restart
00:00:00 Projekt tasmota 4ch_OutdoorLights Version 8.3.1(tasmota)-2_7_1
00:00:06 QPC: Reset
00:00:09 APP: Anzahl Startvorg√§nge 13
00:00:09 CFG: in Flash gespeichert am FA, z√§hlen 1974, Bytes 4096
00:00:21 CMD: status 0
00:00:21 SRC: Serial
00:00:21 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
00:00:21 RSL: stat/4ch_OutdoorLights/STATUS = {"Status":{"Module":7,"DeviceName":"4ch_OutdoorLights","FriendlyName":["4ch_OutdoorLights","4ch_OutdoorLights2","4ch_OutdoorLights3","4ch_OutdoorLights4"],"Topic":"4ch_OutdoorLights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,3,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://raspi4/tasmotaUpdates/4ch.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:23","StartupUTC":"","Sleep":250,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-05-19T01:50:41","SaveCount":1974,"SaveAddress":"FA000"}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020.07.04 20:20:52","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"460/699"}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":0,"SysLog":3,"LogHost":"192.168.1.20","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00000000000000","00000088","00000000"]}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":33,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F002596","04000001","00000006","000000C0","00000000","00004000"],"Drivers":"1,2,3,4,9,10","Sensors":"3"}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS5 = {"StatusNET":{"Hostname":"4ch-outdoorlights","IPAddress":"(IP unset)","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:44:20:87","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.73","MqttPort":1883,"MqttClientMask":"4ch_OutdoorLights","MqttClient":"4ch_OutdoorLights","MqttUser":"tasmota","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:00:22","Local":"1970-01-01T00:00:22","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":99,"Sunrise":"20:03","Sunset":"05:19"}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:22","Switch5":"ON"}}
00:00:22 RSL: stat/4ch_OutdoorLights/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:22","Uptime":"0T00:00:23","UptimeSec":23,"Vcc":3.427,"Heap":33,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF"}}
00:00:28 CMD: wifi 1
00:00:28 SRC: Serial
00:00:28 CMD: Gruppe 0, Index 1, Befehl "WIFI", Daten "1"
00:00:28 RSL: stat/4ch_OutdoorLights/RESULT = {"Wifi":"ON"}
00:00:29 CFG: in Flash gespeichert am F9, z√§hlen 1975, Bytes 4096
00:00:30 APP: starte neu

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: aus Flash geladen am F9, z√§hlen 1975
00:00:00 SRC: Restart
00:00:00 Projekt tasmota 4ch_OutdoorLights Version 8.3.1(tasmota)-2_7_1
00:00:00 WIF: Pr√ºfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: verbinden mit AP1 Eiswuerstchen Channel 6 BSSId 10:9A:DD:84:A0:07 in Modus 11N als 4ch-outdoorlights...
00:00:01 WIF: Pr√ºfe Verbindung...
00:00:01 WIF: verbunden
23:26:34 NTP: UTC 2020-07-04T21:26:33, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
23:26:34 HTP: Web-Server aktiv bei 4ch-outdoorlights mit IP-Adresse 192.168.1.199
23:26:34 WIF: Sending Gratuitous ARP
23:26:35 MQT: Verbindungsversuch...
23:26:35 MQT: verbunden
23:26:35 MQT: tele/4ch_OutdoorLights/LWT = Online (beibehalten)

```

ascillato2, 2020-07-04:
Hi, I have tried to reproduce this but I couldn't. Please, can you try to erase all flash and flash the precompiled bins and then update from there?

kugelkopf123, 2020-07-04:
> Hi, I have tried to reproduce this but I couldn't. Please, can you try to erase all flash and flash the precompiled bins and then update from there?

I just need to find another device that is relatively easily accessible and has at least 8.3.1. installed. 
Actually I don't plan to disassemble my roller shutter box just to test this :)! Thats the Reason why it is pretty important that this update work =)!
I think I still have 2x Wemos d1mini laying around, which I flashed last time on this [project](https://github.com/arendst/Tasmota/issues/8399#issuecomment-644434867). 
Will try it tomorow.

Jason2866, 2020-07-05:
@kugelkopf123  your device setup is screwed. If you have flashed latest development version and it still shows 8.3.1 something has gone to the south. Dont expect that this devices act as the should.

kugelkopf123, 2020-07-05:
@Jason2866 You're right.
The problem can be traced back to a wrong setting in Github Desktop. Obviously I didn't always sync everything. Sorry for the trouble.

closed 2020-07-05 by kugelkopf123