arendst/Tasmota #8856: WifiConfig changes to 2 on reboot on iFan02

rishabmehta7, 2020-07-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have 3 iFan02 devices.

All 3 have been set to WifiConfig 5. On a power failure (which I have been seeing a lot recently - our generator is having issues) the WifiConfig changes to 2 on all 3 devices, my other SonOff devices work as expected (SonOff Basic, SonOff 4CH).

I have tried changing the same to 5 multiple times. But on every 2 3rd reboot it Just switches to 2 and then takes forever to connect to MQTT broker.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____SonOff iFan02
- [x] Tasmota binary firmware version number used: _____8.2, 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: _____esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:41:37 MQT: stat/iFan02_02/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:41:37 MQT: stat/iFan02_02/RESULT = {"Module":{"44":"Sonoff iFan02"}}
12:41:37 MQT: stat/iFan02_02/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"15":"Switch7"},"GPIO2":{"0":"None"},"GPIO3":{"14":"Switch6"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:43:11 MQT: stat/iFan02_02/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":475,"Rules":"on switch6#state do Fanspeed + endon"}
12:43:11 MQT: stat/iFan02_02/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on switch7#state do POWER1 TOGGLE endon"}
12:43:12 MQT: stat/iFan02_02/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
12:43:44 MQT: stat/iFan02_02/STATUS = {"Status":{"Module":44,"FriendlyName":["iFan02_02"],"Topic":"iFan02_02","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,3,3,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
12:43:44 MQT: stat/iFan02_02/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:50:58","StartupUTC":"2020-07-05T06:22:46","Sleep":50,"CfgHolder":4617,"BootCount":193,"BCResetTime":"2020-06-02T09:16:54","SaveCount":1332,"SaveAddress":"FB000"}}
12:43:44 MQT: stat/iFan02_02/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"326/699"}}
12:43:44 MQT: stat/iFan02_02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mehta's",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028029","2805C8000100060000005A00000000000000","00008200","00000000"]}}
12:43:44 MQT: stat/iFan02_02/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:43:44 MQT: stat/iFan02_02/STATUS5 = {"StatusNET":{"Hostname":"iFan02_02","IPAddress":"192.168.31.212","Gateway":"192.168.31.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.31.250","Mac":"84:0D:8E:47:A8:9C","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:43:44 MQT: stat/iFan02_02/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.31.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_47A89C","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:43:44 MQT: stat/iFan02_02/STATUS7 = {"StatusTIM":{"UTC":"2020-07-05T07:13:44","Local":"2020-07-05T12:43:44","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+05:30","Sunrise":"09:24","Sunset":"01:25"}}
12:43:44 MQT: stat/iFan02_02/STATUS10 = {"StatusSNS":{"Time":"2020-07-05T12:43:44","Switch6":"ON","Switch7":"ON"}}
12:43:44 MQT: stat/iFan02_02/STATUS11 = {"StatusSTS":{"Time":"2020-07-05T12:43:44","Uptime":"0T00:50:58","UptimeSec":3058,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"Mehta's","BSSId":"78:11:DC:47:28:65","Channel":8,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:17"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set WifiConfig to 5
SetOption65 1
SetOption19 0
SwitchMode6 3
SwitchMode7 3

Reboot 3-4 times and WifiConfig would become 2

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The WifiConfig value should stay at 5 as set.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Rx and Tx pin are connected to capacitive touch switches(TT223), Vcc Gnd pulled from Pins 

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-07-05:
see SetOption65 and SetOption36

rishabmehta7, 2020-07-05:
SO65 is 1 and SO36 is 0

ascillato2, 2020-07-05:
Hi, please, can you try with latest version (thehackbox.org/tasmota/tasmota.bin). And then, after the update, please perform `reset 6` command and configure your module again for retesting? Thanks

rishabmehta7, 2020-07-06:
I have updated to latest 8.3.1, reset the device(1 of them) and reset the config. Will report back in a day or 2.

rishabmehta7, 2020-07-06:
Still reverts to WifiConfig 2

ascillato, 2020-07-06:
8.3.1 is not latest from development branch. Please update to latest Tasmota version from thehackbox.org/tasmota/tasmota.bin

Thanks

rishabmehta7, 2020-07-06:
Upgraded to 8.4 now, reset and reconfigured everything. Will report back.

rishabmehta7, 2020-07-08:
Nope being on the latest dev build did not help
I had a power cut, the power was back in about 45 sec
it was gone again in like 6 sec and then back in another 3-4 sec.

The WifiConfig reverted back to 2 (It was 4 before)

Whats suprising is its only happening on iFan02 devices (All 3 of them), rest of my SonOffs stay fine.

rishabmehta7, 2020-07-08:
Just a thought, I am using switch6 and 7 here in rules, on my other sonoff basic i use switch 1 and 2. could this be related to that?
I used 6 and 7 assuming iFan might be using it internally as it has 4 relays.

ascillato2, 2020-07-09:
Please, post the output of the console when your device boots and resets its wificonfig. Seems to be a retained message.

rishabmehta7, 2020-07-11:
i will try this over weekend

rishabmehta7, 2020-07-12:
So every time this happens is when i have a power cut, not with a power cut my router as well as hassio also restart.
What could be the best way to get console output with rebooted, WiFi and HA (and MQTT)?

arendst, 2020-07-12:
Try on a ifan02 using the default template WITHOUT connected external switches and see what happens.

rishabmehta7, 2020-07-13:
So i updated to latest dev

rishabmehta7, 2020-07-13:
Had closed the issue by mistake...

rishabmehta7, 2020-07-15:
Ok so without the switches the config seems to have stayed put after reboot.
So the touch switches could be a problem?
I have similarly configured switches on sOnOff Basics which work perfectly fine.

Rishab

rishabmehta7, 2020-07-15:
Just to clarify the switches were connected physically but not configured on the module and no rules were created.

Rishab

rishabmehta7, 2020-07-16:
Spoke too soon, the immediate next reboot the device reset the wifi config to 2, no changes were made in any way.

Rishab

rishabmehta7, 2020-07-16:
Every software reboot doesn't seem to reset the config, a power cycle resets the WifiConfig to 2.
Rishab

rishabmehta7, 2020-07-23:
Any update guys?

closed 2020-07-29 by rishabmehta7

thewizard34, 2022-04-10:
How does one set the WiFi Config on ifan02

ascillato, 2022-04-10:
> How does one set the WiFi Config on ifan02

Go to the Tasmota console and type `wificonfig 4`