arendst/Tasmota #8857: Can I combine (group) 2 relays in 1 switch?

antonwantstosleep, 2020-07-05:
### PROBLEM DESCRIPTION

Sorry for my clumsy Russian English!

I`ve got a KS-621 device (smart wall socket).
First I tried to apply a template for the socket from https://templates.blakadder.com/KS-621.html. The template is like:
```
{"NAME":"KS-621","GPIO":[17,0,0,0,133,132,0,0,130,52,0,21,91],"FLAG":15,"BASE":18}
```
This template looks like working (as I remember right) but has an issue: device continously reboots (every 8-10 seconds) with reason "Power ON". It happens only when the socket is in ON (power out) state.

So I decided to "configure unknown device" and investigated that there are two relays for one output. Got this template:
```
{"NAME":"KS-621","GPIO":[52,0,0,0,133,132,0,0,130,21,0,22,17],"FLAG":0,"BASE":18}, where
52 - Led1 - is ON when Relay1 or Relay2 is ON, is OFF when both relays are OFF
21 - Relay1 - needs to be ON (rising edge) to **enable** power out (Relay2 must be OFF), after that can be switched OFF
22 - Relay2 - needs to be ON (rising edge) to **disable** power out (Relay1 must be OFF), after that can be switched OFF
133 - power
132 - voltage
130 - current
17 - Button1 - a physical button (with Led1)
```
I tried to configure the GPIO13 and GPIO15 as Relay1 and Relay1i but this leads to the same issue - continuous reboot every 10 seconds. So I switched back to two separate relays.

Also I`ve made a rule:
```
SetOption21 1   //Energy monitoring when power is off. // Default 0
SetOption73 1   //Enable Buttons decoupling and send multi-press and hold MQTT messages. // Default 0
SetOption32 10  //Number of 0.1 seconds to hold button before sending HOLD action message. // Default 40
LedMask 0x0002  //Only Relay 2 controls the power LED (0x0002 = 0000 0000 0000 0010)
Rule1 ON Button1#State DO Backlog Power2 0; Delay 2; Power1 1 ENDON ON Button1#State=3 DO Backlog Power1 0; Delay 2; Power2 1 ENDON
Rule1 1
```
Now the socket is working very well:
- when I press the button it turns OFF (the led goes OFF)
- when I hold the button it turns ON (the led goes ON)

But my problem is that I got **two separate ON-OFF switches in the webui (for each relay), two friendly names and problems with Home Assistant (too much devices)**. What should I do to have only one switch? 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): KS-621
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:25:01 MQT: stat/tasmota_water_heater/RESULT = {"NAME":"KS-621","GPIO":[52,0,0,0,133,132,0,0,130,21,0,22,17],"FLAG":0,"BASE":18}
08:25:01 MQT: stat/tasmota_water_heater/RESULT = {"Module":{"0":"KS-621"}}
08:25:01 MQT: stat/tasmota_water_heater/RESULT = {"GPIO0":{"52":"Led1"},"GPIO1":{"0":"-нет-"},"GPIO2":{"0":"-нет-"},"GPIO3":{"0":"-нет-"},"GPIO4":{"133":"HLW8012 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"-нет-"},"GPIO10":{"0":"-нет-"},"GPIO12":{"130":"HLWBL SEL"},"GPIO13":{"21":"Relay1"},"GPIO14":{"0":"-нет-"},"GPIO15":{"22":"Relay2"},"GPIO16":{"17":"Button1"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
08:26:12 MQT: stat/tasmota_water_heater/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":125,"Free":386,"Rules":"ON Button1#State DO Backlog Power2 0; Delay 2; Power1 1 ENDON ON Button1#State=3 DO Backlog Power1 0; Delay 2; Power2 1 ENDON"}
08:26:13 MQT: stat/tasmota_water_heater/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
08:26:13 MQT: stat/tasmota_water_heater/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:26:48 MQT: stat/tasmota_water_heater/STATUS = {"Status":{"Module":0,"DeviceName":"Water Heater","FriendlyName":["Water Heater OFF","Water Heater ON"],"Topic":"tasmota_water_heater","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"0002","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-RU.bin","RestartReason":"Software/System restart","Uptime":"3T13:35:06","StartupUTC":"2020-07-01T17:51:42","Sleep":50,"CfgHolder":4617,"BootCount":4549,"BCResetTime":"2020-06-13T15:03:27","SaveCount":4902,"SaveAddress":"F5000"}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:50:55","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"425/699"}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["devices","-"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["002A8009","0A05C8000100060000005A00000000000000","00800200","00000000"]}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000419","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS5 = {"StatusNET":{"Hostname":"tasmota_water_heater-8052","IPAddress":"192.168.3.43","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"***","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.10","MqttPort":1883,"MqttClientMask":"Water Heater","MqttClient":"Water Heater","MqttUser":"device","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS7 = {"StatusTIM":{"UTC":"2020-07-05T07:26:48","Local":"2020-07-05T08:26:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS10 = {"StatusSNS":{"Time":"2020-07-05T08:26:48","ENERGY":{"TotalStartTime":"2020-06-13T14:36:16","Total":81.856,"Yesterday":14.617,"Today":4.479,"Power":2841,"ApparentPower":2841,"ReactivePower":0,"Factor":1.00,"Voltage":251,"Current":11.305}}}
08:26:48 MQT: stat/tasmota_water_heater/STATUS11 = {"StatusSTS":{"Time":"2020-07-05T08:26:48","Uptime":"3T13:35:06","UptimeSec":308106,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"devices","BSSId":"***","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}
08:26:54 MQT: tele/tasmota_water_heater/HASS_STATE = {"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:50:55","Module or Template":"KS-621","RestartReason":"Software/System restart","Uptime":"3T13:35:12","Hostname":"tasmota_water_heater-8052","IPAddress":"192.168.3.43","RSSI":"66","Signal (dBm)":"-67","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:03","MqttCount":1,"LoadAvg":19}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

grob6000, 2020-07-16:
> 21 - Relay1 - needs to be ON (rising edge) to **enable** power out (Relay2 must be OFF), after that can be switched OFF
> 22 - Relay2 - needs to be ON (rising edge) to **disable** power out (Relay1 must be OFF), after that can be switched OFF

Sounds like a latching relay - two coils, pulling the actuator in opposite directions. A pulse to relay1 turns the device on, a pulse to relay2 turns the device off? The resets might be a power supply problem, the device may not be designed to hold the coil current long term. The advantage is minimal parasitic power consumption when on, unlike a normal relay (can be 2-3W in a relay coil).

You will just have to hide the auto-generated controls in HA and wrap it with a template switch I think. Tasmota can help make it a bit easier though:

- `pulsetime1 1` and `pulsetime2 1` (see [here](https://tasmota.github.io/docs/Commands/#pulsetime)) instead of your delay routine in rules. This will make a 0.1 second pulse instead of leaving it on (adjust as needed).
- Give the two relays friendly names of 'on' and 'off' on the 'configure other' page. When you run `setoption 19` in the tasmota console to update the ha autoconfig, and home assistant should show two switches: "switch.tasmota_water_heater_on" and "switch.tasmota_water_heater_off", which is at least a bit clearer.
- Create a rule to do the state updates (as the power1 and power2 states are not very useful to you):
`rule1 on power1#state=1 do publish stat/tasmota_water_heater/LATCH on endon on power2#state=1 do publish stat/tasmota_water_heater/LATCH off endon` 
- In HA, it's quite common to wrap this kind of thing, see [this example](https://www.home-assistant.io/integrations/switch.template/#sensor-and-two-switches). Create a [template switch](https://www.home-assistant.io/integrations/switch.template/) to handle the toggling, and [mqtt sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) to handle the state. The template switch references the mqtt sensor for the power state, and the autodiscovered switches to control it. You can then use the template switch in your front-end, which will work as you expect.

A tasmota-native latching relay support would be interesting. I'd put the latch into settings (use a `setoption`, which would specify a number of latch pairs from 0 to 4, which would make relay1/relay2 a pair, relay3/relay4 a pair, etc). If a device needed a latch, you'd use the GPIO config to set the pairs correctly as needed. I'd then intercept all the first relay state and commands to be the latch, and suppress the second, including in the ha autoconfig. The tricky bit would be catching everywhere the code checks how many relays are configured. e.g. making sure it doesn't cause mayhem in shutters. The result from the home automation level would be a single relay / switch which works as if we knew nothing of the internals of the device. Which is ideal IMO.

arendst, 2020-07-16:
Latching relays have always been supported by Tasmota. Just select module 16 (EXS_RELAY) as the base module and have a go. Relays 0/1, 2/3, etc form a pair.

antonwantstosleep, 2020-07-16:
@arendst , that is great!!! Thank you so much!!!

closed 2020-07-16 by antonwantstosleep

chamroeunbm, 2020-09-23:
Thanks for the input!!

I see the latching in Tasmota as well, but I don't know if it can send negative pulse or not. I see many latching can't send negative voltage.
This latching valve require positive voltage to Open and Negative Pulse to close it.

grob6000, 2020-09-23:
If the relay requires negative voltage, this will require specific
circuitry to operate. Without knowing your hardware in detail, I can’t
advise much more than that.

With the right driver circuitry, I see no reason this isn’t possible.