arendst/Tasmota #8860: CTPower(ADC) Resets total energy with every restart.

kugelkopf123, 2020-07-05:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
I use the ADC input of my WEMOS as a selfmade current meter for a LED Controller. (ADC Option CT-POWER) Therefore the input is set to CT-POWER. So far it works very well. But the "energy total" - value is zero with each restart. 
This makes no sense at all. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
It would make sense to save the value when switching a relay on and off or before a manual restart. 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Alternatively, you could make the value accessible to the scripter so that you can save it yourself and read it in again. 
Or you can do the calculation completely in the script. But for this you would have to get access to the current value of the analog input.
**Additional context**  
_Add any other context or screenshots about the feature request here._
-
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-07-05:
Sorry, as designed. The Energy total data is stored on flash only at midnight to avoid a fast flash wear.

closed 2020-07-05 by ascillato2

kugelkopf123, 2020-07-05:
@ascillato 
Ok, got it. But is there a possibility to make the value from the ADC, no matter if its the Struct with ADC.Energy nor the normal A0 Value accessable in the scripter?

ascillato, 2020-07-06:
> is there a possibility to make the value from the ADC

Yes, as any other sensor using its JSON string (the same as Tasmota posts in the console)
https://tasmota.github.io/docs/Scripting-Language/#sensor-logging

kugelkopf123, 2020-07-06:
> > is there a possibility to make the value from the ADC
> 
> Yes, as any other sensor using its JSON string (the same as Tasmota posts in the console)
> https://tasmota.github.io/docs/Scripting-Language/#sensor-logging

Ah great! Thats what i was searching for! I already had the idea of fetching the JSON Output of Result, but there was no command to call, to get the value of the ADC in the console. 
Thanks!