arendst/Tasmota #8864: Rotary encoder enhanced features request

ximonline, 2020-07-05:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
If I turned my H801 dimmer into monochromatic mode (setoption68) instead of RGBWW - rotary encoder didn't work.
And there is no changes in mqtt topic, when I adjust brightness by rotary in rgb mode (button 1 change the state "Power 1".
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
It could be better with mqtt, first: all values and states changes could be published in mqtt topoc.
And I request feature for H801 to use rotary encoder in monochromatic mode more useful:
- use 1 or 2 or more rotary (2 and more by use pins from channels).
- choose witch channel or group will be adjust by rotary (for ex. Level1, Level2 = rotary1; Level3 = rotary2). It will be usefull for different zone or rooms with its own rotary encoders.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-06:
Stay tuned, Theo is thinking about to add a second rotary encoder and how to enhance use case
for the rotary encoders.
You can change the behaviour of the encoder with rules. Use `SetOption98 1` to switch

arendst, 2020-07-13:
Latest commit supports seconf rotary encoder to be default configured as CW light control. First rotary encoder would then control RGB.

Not quite what you asked for but currently I won't support more than 2 rotary encoders.

At least it now reports MQTT status when rotary changes happened.

ximonline, 2020-07-14:
> Latest commit supports seconf rotary encoder to be default configured as CW light control. First rotary encoder would then control RGB.
> 
What about monochromatic mode (setoption68), when all channels are white? Last version not properly works...
Maybe 2 encoders could adjust 2 light zones?

closed 2020-07-25 by ascillato2

ximonline, 2021-01-10:
> Stay tuned, Theo is thinking about to add a second rotary encoder and how to enhance use case
> for the rotary encoders.
> You can change the behaviour of the encoder with rules. Use `SetOption98 1` to switch

Hello! Could somebody lep me with rules:

I use H801 for dimming LED stripes in WWWWW mode. For rotary encoder I use this rule:
rule1 on rotary1#pos1 do channel3 + endon
And it's particulary works (channel3 led stripe encrease for 10 by rotating encoder any side)

What rule I need for: turn right encrease led level, turn left decrease led level?

I think I need value of "rotary1#pos1"  and "rotary1#pos2", but can't write it...

How i can see pos1 and pos2 value in console?