arendst/Tasmota #8866: USE_TUYA_TIMER not working

diabl0w, 2020-07-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Setting USE_TUYA_TIMER does not result in the proper time getting set on my thermostat. I have found in general what is causing this, but don't know enough to know the fix. The following set of code is relevant:

>     case TUYA_CMD_HEARTBEAT:
>      AddLog_P(LOG_LEVEL_DEBUG, PSTR("TYA: Heartbeat"));
>      //if (Tuya.buffer[6] == 0) {
>        //AddLog_P(LOG_LEVEL_DEBUG, PSTR("TYA: Detected MCU restart"));
>        //Tuya.wifi_state = -2;
>        #ifdef USE_TUYA_TIME
>        TuyaSetTime();
>        #endif

Normally, the "//" aren't there. I had to comment out those parts in order to make it work. So I am guessing that whatever "If Tuya.buffer[6]==0" is what is actually failing to pass and why the time isn't being updated. So what does that line mean?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Beca Thermostat Model BHP-6000
- [x] Tasmota binary firmware version number used: Latest Development
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode/Platformio
- [x] Flashing tools used: WebUI OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```
  Configuration output here:
00:51:08 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
00:51:08 MQT: %topic/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:51:09 SRC: Backlog
00:51:09 CMD: Group 0, Index 1, Command "MODULE", Data ""
00:51:09 MQT: %topic/stat/RESULT = {"Module":{"54":"Tuya MCU"}}
00:51:09 SRC: Backlog
00:51:09 CMD: Group 0, Index 1, Command "GPIO", Data "255"
00:51:09 MQT: %topic/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:55:40 MQT: %topic/stat/STATUS = {"Status":{"Module":54,"DeviceName":"thermostat","FriendlyName":["thermostat"],"Topic":"family/home/thermostat","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:55:40 MQT: %topic/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:49","StartupUTC":"2020-07-06T04:41:51","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2020-07-06T00:42:00","SaveCount":7,"SaveAddress":"F5000"}}
00:55:40 MQT: %topic/stat/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-06T00:41:02","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"371/699"}}
00:55:40 MQT: %topic/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00010000","00006000"]}}
00:55:40 MQT: %topic/stat/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","05000486","040C8001","00000045","010000C0","00004080","00000000"],"Drivers":"1,2,3,4,9,10,16","Sensors":"1,2,3,5"}}
00:55:40 MQT: %topic/stat/STATUS5 = {"StatusNET":{"Hostname":"family/home/thermostat-4708","IPAddress":"192.168.101.117","Gateway":"192.168.101.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.101.1","Mac":"24:62:AB:04:12:64","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:55:40 MQT: %topic/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.111.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_041264","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:55:40 MQT: %topic/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-07-06T04:55:40","Local":"2020-07-06T00:55:40","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99}}
00:55:40 MQT: %topic/stat/STATUS10 = {"StatusSNS":{"Time":"2020-07-06T00:55:40"}}
00:55:40 MQT: %topic/stat/STATUS11 = {"StatusSTS":{"Time":"2020-07-06T00:55:40","Uptime":"0T00:13:49","UptimeSec":829,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"WIFI24","BSSId":"60:38:E0:B4:0F:81","Channel":6,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set USE_TUYA_TIMER on a BecaThermostat Model BHP-6000 and flash, time won't update

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Time updates

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
As mentioned, commenting out the "if" statement fixes the problem of the time not being set. So the if statement failing is the true problem.

**(Please, remember to close the issue when the problem has been addressed)**

chaosmaster, 2020-07-15:
The option is called `USE_TUYA_TIME` not `USE_TUYA_TIMER`.
The current code only sets the time when an MCU-reset is detected.
It should probably also set the time somewhere during initialization.

diabl0w, 2020-07-15:
> The option is called `USE_TUYA_TIME` not `USE_TUYA_TIMER`.
> The current code only sets the time when an MCU-reset is detected.
> It should probably also set the time somewhere during initialization.

thanks, thats what i gathered from it. not sure why an mcu reset is desirable... my thermostat is always on, i dont want the mcu to reset. commenting out like i did causes the time to be set/synced probably way more often then it needs to be, but at least it works.

should probably just be set after NTP sync

chaosmaster, 2020-07-15:
A reset isn't desired, but in case it happens, time must be set again.
Maybe @Bettman66  can look into also setting time during init.

diabl0w, 2020-07-15:
> A reset isn't desired, but in case it happens, time must be set again.
> Maybe @Bettman66 can look into also setting time during init.

ohhhh okay that makes sense now, thanks for clarifying and for bringing attention to the issue!

Bettman66, 2020-07-16:
I also use the BHT-6000, the time is only set to the Tasmota time once after a restart of the thermostat. I have no problems with time deviations after long use of the thermostat. Do you have another thermostat to test to rule out a hardware defect?

Bettman66, 2020-07-16:
You can also insert and test the function here:
`if (Tuya.heartbeat_timer > 10) {
            Tuya.heartbeat_timer = 0;
            TuyaSendCmd(TUYA_CMD_HEARTBEAT);
            TuyaSetTime();
          }`

closed 2020-07-16 by arendst

diabl0w, 2020-07-16:
> I also use the BHT-6000, the time is only set to the Tasmota time once after a restart of the thermostat. I have no problems with time deviations after long use of the thermostat. Do you have another thermostat to test to rule out a hardware defect?

Thanks for fixing the issue! Mine is actually a BHP-6000, slightly different thermostat. But the good news is that its working, thanks

PR #8926 (2020-07-16): issues #8866
update tuyatime every 60sec

The user had time deviations after long operation

## Description:

**Related issue (if applicable):** fixes #8866

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.2.1
  - [ ] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

 [... cut]