arendst/Tasmota #8869: Different GroupTopics per device (multiple switches)

andker87, 2020-07-08:
**Have you looked for this feature in other issues and in the docs?**  
YES

**Is your feature request related to a problem? Please describe.**  
I have several 4ch sonoff, some channels manage lights, others some sockets
e.g.
1) light-light-light-outlet
2) light-light-light-light
3) outlet-light-light-outlet
I use the grouptopic to turn the lights off massively, but I'd like to exclude the power off of the outlets. for example, turning off the water valve because I am leaving the house could block the dishwasher or washing machine, or turning off the gas valve would force me to reset it manually every time.

**Describe the solution you'd like**  
Assign each channel of the device to a different grouptopic in the case described above:
example above:
cmnd/sonoff-x/grouptopicpower1 lights
cmnd/sonoff-x/grouptopicpower4 outlets
cmnd/sonoff-y/grouptopicpower1 lights
cmnd/sonoff-x/grouptopicpower1 outlets

objective:
1) lights-lights-lights-outlets
2) lights-lights-lights-lights
3) outlets-lights-lights-outlets

**Describe alternatives you've considered**  
I have no other ideas. The only solution would be to use different devices for lights or sockets, but it would not be easy.

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-07-08:
Hi,

You can achieve that behavior using rules. No need to modify Tasmota. You can send an `event` to your grouptopic and in your devices you can set what you need only.

Example:

You can send to your grouptopic `tasmotas` the following commands:
`cmnd/tasmotas/event lights=off`
`cmnd/tasmotas/event lights=on`
`cmnd/tasmotas/event outlets=off`
`cmnd/tasmotas/event outlets=on`

(`lights=off` is the payload of the mqtt message) 

And in your devices you can set the following for example:
```lua
Rule1 1 
Rule1 on event#lights do backlog power1 %value%;power2 %value%;power4 %value% endon on event#outlets do power3 %value% endon
```

Virtual-Nomad, 2020-07-08:
I think this may be a similar requirement situation but I may have my understanding of devgroups a little wrong ..... from what I understand and have found:
- I was very excited to find devgroups that could send actions independent of a broker. which would allow a more robust smart environment for core functions. 
- It appeared to work well by using devgroupname for groups of lights however I was keen to also sync multiple switch relays
- given I am running 4 channel switches I would ideally like to group relays across multiple switches

I thought I could do this with groupname1 - 4 on the switch however I believe the way it plays out is when I action the relay on a switch it broadcasts to the devgroupname for that relay to all switches with the same devgroupname as the primary (based on the bit-wise relay number) .
- Switch Panel 1 (Bathroom) : Relay1 - Bathroom , Realy2 - WIR, Relay3 - Heatlights, Relay4 - ExtractorFan
- Switch Panel 2  (Robe) : Relay1 - WIR , Realy2 - Bathroom, Relay3 - Heatlights, Relay4 - ExtractorFan
- Switch Panel 3 (Bedroom) : Relay1 - Bedroom , Realy2 - WIR, Relay3 - Bathroom , Relay4 - ReadingLights

if I attempt to action Bathroom from Panel 3 Relay 3 001 - it will synch Panel 1 Relay 1 but not Panel 2 Relay 2 
(NB: if I action Bathroom from Panel 3 Relay 3 010 - it will synch Panel 1 Relay 1 and Relay 2 but nothing on Panel 2 
if I attempt to action WIR action Panel 3 Realy 2 001 - it will not synch Panel 1 Relay 2 or Panel 2 Relay as I believe it is sending the broadcast to devices with devgroupname1 of WIR 

Ideally, I would like to be assigning individual groupdevnames to specific relays and have them sync.

This may seem an unusual request ... but it would be ideal if I could simply assign all lights and relays to a given devgroupname and have them sync for multi switching of circuits.

pcdiem, 2020-07-08:
You can enable option 88 (SO88 1). Then each relay will be in a separate device group. See PR https://github.com/arendst/Tasmota/pull/8558

ascillato2, 2020-07-09:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-09 by ascillato2

Virtual-Nomad, 2020-07-18:
> You can enable option 88 (SO88 1). Then each relay will be in a separate device group. See PR #8558

@pcdiem Paul So not sure if I am doing something wrong but for the life of me, I can't get SO88 to work. Here is a rundown of what I am doing in case someone can put me in the right direction.

### Configure Switch
Switch1 4 Gang Switch - Tasmota 8.3.1 Core/SDK Version 2_7_1/2.2.2-dev(38a443e)
Enable DevGroup  `SO85 1`
Enable Separate Relay  `SO88 1`
Set switch/relay for Bath  `DevGroupName1 Bath`
Set switch/relay for WIR  `DevGroupName2 WIR`
devgroupname `/RESULT = {"DevGroupName1":"Bath","DevGroupName2":"WIR","DevGroupName3":"","DevGroupName4":""}`

### Configure Light
Light - (tasmota) 8.3.1 Core/SDK Version 2_7_1/2.2.2-dev(38a443e)
Enable DevGroup - SO85 1
Set Devgroup - DevGroupName1 Bath
DevGroupName `/RESULT = {"DevGroupName1":"Bath","DevGroupName2":"","DevGroupName3":"","DevGroupName4":""}`

### Test 
Check Membership On Switch `DevGroupStatus1` 
Result is `stat/SW02-4Gang-Bath/RESULT = {"DevGroupStatus":{"Index":0,"GroupName":"Bath","MessageSeq":2,"MemberCount":1,"Members":[{"IPAddress":"192.168.1.185","ResendCount":0,"LastRcvdSeq":5,"LastAckedSeq":2}]}`

Switch 1 ON - Light Follows ON, Switch 1 OFF - Light Follows ON 
Switch Light ON - Switch 1 Follows, Switch Light OFF - Switch Follows OFF
So All Working Well 

### Change Light to WIR Group
Set Devgroup Name - `DevGroupName1 WIR`
Check - `DevGroupName` > `/RESULT = {"DevGroupName1":"WIR","DevGroupName2":"","DevGroupName3":"","DevGroupName4":""}`
Check Membership on Switch  `DevGroupStatus2` 
Result `RESULT = {"DevGroupStatus":{"Index":1,"GroupName":"WIR","MessageSeq":4,"MemberCount":1,"Members":[{"IPAddress":"192.168.1.185","ResendCount":0,"LastRcvdSeq":2,"LastAckedSeq":4}]}`

Switch 2 ON - Light Stays OFF, Switch 2 OFF - Light stays OFF
Switch Light ON - Switch stays OFF, Switch Light OFF - Switch stays OFF
So not listening 

NB: have also tried with SO85  incase but then nothing devgroup based works (as expected) 

Hope someone can point out what I am doing wrong