arendst/Tasmota #8876: Is it possible to turn off the red led when device is ON?

matrix200, 2020-07-09:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
The red led is annoying and it would be nice, to be able to turn it off.

**Describe the solution you'd like**  
Updated template that also controls the Red led indicating relay status (on/off) if software controllabe.

**Describe alternatives you've considered**  
Either put an electric tape around the device to hide the red led (could cause oveheating and not very elegant).
Open the device and carefully scrape out light emitting material from the LED (destructive and irreversible).

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-07-09:
The answer depends on the device. If the LED is connected to a GPIO then it may be possible. However, on some devices the LED is connected directly to the relay output state via hardware.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

matrix200, 2020-07-09:
Ok, I will do that there.
Thanks

closed 2020-07-09 by matrix200