arendst/Tasmota #8880: TuyaMCU ability to set a color

tom-23, 2020-07-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have a 700ml diffuser that uses the TuyaMCU. It follows the same set of dplds which are listened on the docs but there is no way to have a custom light color using the method provided even though you can set the light color using TuyaSend3 with a hex value.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like for there to be another fnId with a type something along the lines of "Hex Color" or something like that where you can specify what dpld is to be used to control the light color.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I have tried to by adding 3 PWM items in the module config page and then, using scripting, used TuyaSend for whenever the color has been changed. There are a couple of downsides to this. Firstly, I cannot get grammar correction and I can't fade between the colors. Also, when I set the dimmer value to 0, tasmota will turn off the device including the mister.

**Additional context**  
_Add any other context or screenshots about the feature request here._
TuyaMCU Docs Page: [https://tasmota.github.io/docs/TuyaMCU/#fnid](https://tasmota.github.io/docs/TuyaMCU/#fnid)
**(Please, remember to close the issue when the problem has been addressed)**

Bettman66, 2020-07-17:
[https://templates.blakadder.com/brilex.html](url)

Have you already tried this option?

effelle, 2020-07-23:
@tom-23, 
please paste here the result (tuyareceived) of  TuyaSend command when you set a color manually. Thanks.

cangurellobello, 2020-07-28:
+1.  
I have a tywe3s +mcu lamp (HG02) and cannot set colors without creating a script. 
It has a data4 controller set on dpId 21 that must be set before editing in these 3 modes (RGBW,RGB,RGBW-Pulse)(00,01,02).
The rgb dpId has a data3 stream and the rgbw has a data2. 
Best regards

effelle, 2020-07-29:
@cangurellobello Adding the RGB is not that easy on a tuya MCU, there are device that have real RGB and others that have just a list of colors available and not always respecting an index. If you were able to change the rgb color from console please post here your serialsend or tuyasend3 command and let me see the output.
Also, the White is a CT White or is simply dimmable?

cangurellobello, 2020-07-29:
@effelle 
It's both, I can dim and set the color temperature on RGBW. 
I cannot get the output out of `SerialSend` nor `SSerialSend`.
///////////////////////////////////////////////////////////////////
Here's a short analysis:

**RGBW**

I have found out that the RGBW dpId accepts INTEGER values raging from 0 to 1000 (Warmest to Coldest), is there a second fid 21 I can add to this one?

//Setting first controller on dpId 21 to switch to RGBW mode

```
15:41:45 CMD: TuyaSend4 21,00
15:17:24 CMD: Gruppo 0, Indice 4, Comando "TUYASEND", Dati "21,00"
15:17:24 TYA: Send "55aa00060005150400010024"
15:17:24 {"TuyaReceived":{"Data":"55AA03070005150400010028","Cmnd":7,"CmndData":"1504000100","DpType4Id21":0,"21":{"DpId":21,"DpIdType":4,"DpIdData":"00"}}}
15:17:24 TYA: fnId=0 is set for dpId=21
```

//Sending color to dpId 23 (Ranging from "00000000" (Warmest) to "000003E8" (Coldest))

```
15:41:45 CMD: TuyaSend2 23,0
15:41:45 CMD: Gruppo 0, Indice 2, Comando "TUYASEND", Dati "23,0"
15:41:45 TYA: Send "55aa0006000817020004000000002a"
15:41:45 {"TuyaReceived":{"Data":"55AA0307000817020004000000002E","Cmnd":7,"CmndData":"1702000400000000","DpType2Id23":0,"23":{"DpId":23,"DpIdType":2,"DpIdData":"00000000"}}}
15:41:45 TYA: fnId=0 is set for dpId=23
```

```
15:42:55 CMD: TuyaSend2 23,1000
15:42:55 CMD: Gruppo 0, Indice 2, Comando "TUYASEND", Dati "23,1000"
15:42:55 TYA: Send "55aa0006000817020004000003e815"
15:42:55 {"TuyaReceived":{"Data":"55AA0307000817020004000003E819","Cmnd":7,"CmndData":"17020004000003E8","DpType2Id23":1000,"23":{"DpId":23,"DpIdType":2,"DpIdData":"000003E8"}}}
15:42:55 TYA: fnId=0 is set for dpId=23
```

**RGB**

I'm still trying to find the right data map the chip can accept, in the mean time i can send values ranging from "000003e803e8" (Vibrant Red) to "016503e803e8" (Vibrant Pink).
Apparently the chip cannot set dim the brightness level while handling RGB mode, when I try dimming the light from the slider in the GUI it just reverts back to RGBW and then applies value onto the handled RGBW color.

//Setting first controller on dpId 21 to switch to RGB mode

```
16:03:38 CMD: TuyaSend4 21,01
16:03:38 CMD: Gruppo 0, Indice 4, Comando "TUYASEND", Dati "21,01"
16:03:38 TYA: Send "55aa00060005150400010125"
16:03:38 {"TuyaReceived":{"Data":"55AA03070005150400010129","Cmnd":7,"CmndData":"1504000101","DpType4Id21":1,"21":{"DpId":21,"DpIdType":4,"DpIdData":"01"}}}
16:03:38 TYA: fnId=0 is set for dpId=21
```
//Sending color to dpId 24

```
16:05:19 CMD: TuyaSend3 24,000003e803e8
16:05:19 CMD: Gruppo 0, Indice 3, Comando "TUYASEND", Dati "24,000003e803e8"
16:05:19 TYA: Send "55aa000600101803000c303030303033653830336538fc"
16:05:20 {"TuyaReceived":{"Data":"55AA030700101803000C30303030303365383033653800","Cmnd":7,"CmndData":"1803000C303030303033653830336538","DpType3Id24":"000003e803e8","24":{"DpId":24,"DpIdType":3,"DpIdData":"303030303033653830336538","Type3Data":"000003e803e8"}}}
16:05:20 TYA: fnId=0 is set for dpId=24
```

```
16:07:25 CMD: TuyaSend3 24,016503e803e8
16:07:25 CMD: Gruppo 0, Indice 3, Comando "TUYASEND", Dati "24,016503e803e8"
16:07:25 TYA: Send "55aa000600101803000c30313635303365383033653808"
16:07:25 {"TuyaReceived":{"Data":"55AA030700101803000C3031363530336538303365380C","Cmnd":7,"CmndData":"1803000C303136353033653830336538","DpType3Id24":"016503e803e8","24":{"DpId":24,"DpIdType":3,"DpIdData":"303136353033653830336538","Type3Data":"016503e803e8"}}}
16:07:25 TYA: fnId=0 is set for dpId=24
```

cangurellobello, 2020-07-30:
Update:
Got it, the last "never changing digits" are handling the brightness, going from "0000001" darkest to "3e803e8" brightest.

cangurellobello, 2020-07-30:
I managed to decode the structure of the message:
The zero's inside this packet contain the three divided rgb settings  |0| "165" = (first triplet) color  |0| "3e8" = (second triplet) color saturation |0| 3e8 = (third triplet) brightness.

effelle, 2020-07-30:
Thanks for the feedback!

cangurellobello, 2020-07-30:
> Thanks for the feedback!
> Please post here the result of `SerialSend5 55aa0001000000`, thanks.

```
17:29:15 CMD: SerialSend5 55aa0001000000
17:29:15 CMD: Gruppo 0, Indice 5, Comando "SERIALSEND", Dati "55aa0001000000"
17:29:16 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A2235616462757367616368657766686568222C2276223A22312E312E30222C226D223A307D3E","Cmnd":1,"CmndData":"7B2270223A2235616462757367616368657766686568222C2276223A22312E312E30222C226D223A307D"}}
17:29:16 TYA: MCU Product ID: {"p":"5adbusgachewfheh","v":"1.1.0","m":0}
```

effelle, 2020-07-30:
I see, you didn't add any fnId. Will be necessary add them to enable the new functions (like the sliders on gui). 
I'll contact you if I'll need more info. Grazie.

cangurellobello, 2020-07-30:
I did actually add 2 fids: one (11) for dpid 20 [On/Off Switch] and the other one (21) on dpid 22 [RGBW-only Light Dimmer].
For the other dpids (21,23,24) I didn't assign any fid since the wiki doesen't mention other Dimming, PWM controllers. 
Can one fid be assigned on more than one dpId? 
Buonaserata! :)

effelle, 2020-07-30:
No, you can't. I'll add some new fuctions for that. Ping me on Discord, I'll probably need some help to do testing.

cangurellobello, 2020-07-30:
> No, you can't. I'll add some new fuctions for that. Ping me on Discord, I'll probably need some help to do testing.

Same nick? I can't find you on the official server. Here's mine : MiChiamoPollo#9806

effelle, 2020-07-30:
same nick, write something on channel I'll pick you up.

effelle, 2020-08-29:
Hold on.

antoniomolram, 2020-09-08:
I have some led strip having the same behavior as you describe, for me the question is how to match all of this with a template? there is a simple way to do that? I am quite new with Tasmota (1 day), but I can implement the feature if someone points me to the correct place

cangurellobello, 2020-09-08:
> I have some led strip having the same behavior as you describe, @cangurellobello for me the question is how to match all of this with a template? there is a simple way to do that? I am quite new with Tasmota (1 day), but I am firmware developer, and  I can implement the feature if someone points me to the correct place

I'm in still contact with @effelle and he has made huge progress since we've started chatting on discord.
If you're willing to help on debugging and testing you can contact him with the same nick.

closed 2020-09-17 by arendst

rjmcdougall, 2020-11-05:
I have a decice (XOAMI string bulbs) that is also tuya based.

It works almost the same, but only seems to accept RRGGBB for color.

andronedev, 2022-01-19:
```js
function gen_color(r, g, b) {
    let r_base_16 = r.toString(16);
    let g_base_16 = g.toString(16);
    let b_base_16 = b.toString(16);
    if (r_base_16.length == 1) {
        r_base_16 = '0' + r_base_16;
    }
    if (g_base_16.length == 1) {
        g_base_16 = '0' + g_base_16;
        
    }
    if (b_base_16.length == 1) {
        b_base_16 = '0' + b_base_16;
    }
    let hex = r_base_16 + g_base_16 + b_base_16;
    return hex + "e803e8"
}

```
I had the same problem, I wrote this function in js to transform a rgb color to the right format