arendst/Tasmota #8881: setoption13 keeps resetting to 0

s1nbad, 2020-07-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
at 10:17:54 I enter setoption13 1
at 11:39:46 stat/RESULT = {"SetOption13":"OFF"}

this has been continually happening but previously there had always been a series of button events
this time the console shows no activity just tele messages
still setoption13 sets back to 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:48:33 MQT: hole/CE14/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:48:33 MQT: hole/CE14/stat/RESULT = {"Module":{"18":"Generic"}}
11:48:34 MQT: hole/CE14/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"19":"Button3"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"7":"WS2812"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NO RULES

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

11:49:18 CMD: status 0
11:49:18 MQT: hole/CE14/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"hole/CE14","ButtonTopic":"hole/CE14/vibration/","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"hole/CE14","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:49:18 MQT: hole/CE14/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"hole","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T04:44:00","StartupUTC":"2020-07-11T05:05:18","Sleep":0,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-07-11T10:37:22","SaveCount":53,"SaveAddress":"F7000"}}
11:49:18 MQT: hole/CE14/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(tasmota)","BuildDateTime":"2020-07-10T12:06:54","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"351/699"}}
11:49:18 MQT: hole/CE14/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PKHA",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0400800A","C805C8000000060000005A00000000000000","008080C8","00006000"]}}
11:49:18 MQT: hole/CE14/stat/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:49:18 MQT: hole/CE14/stat/STATUS5 = {"StatusNET":{"Hostname":"hole/CE14","IPAddress":"192.168.1.91","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.250","Mac":"68:C6:3A:F8:CE:14","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:49:18 MQT: hole/CE14/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"DVES_%12X","MqttClient":"DVES_68C63AF8CE14","MqttUser":"mqttUser","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:49:18 MQT: hole/CE14/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-07-11T09:49:18","Local":"2020-07-11T11:49:18","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:59","Sunset":"21:51"}}
11:49:18 MQT: hole/CE14/stat/STATUS10 = {"StatusSNS":{"Time":"2020-07-11T11:49:18"}}
11:49:18 MQT: hole/CE14/stat/STATUS11 = {"StatusSTS":{"Time":"2020-07-11T11:49:18","Uptime":"0T04:44:00","UptimeSec":17040,"Heap":21,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:17:54 CMD: setoption13 on
10:17:54 MQT: hole/CE14/stat/RESULT = {"SetOption13":"ON"}
10:20:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:20:34","Uptime":"0T03:15:16","UptimeSec":11716,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:25:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:25:34","Uptime":"0T03:20:16","UptimeSec":12016,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:30:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:30:34","Uptime":"0T03:25:16","UptimeSec":12316,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:35:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:35:34","Uptime":"0T03:30:16","UptimeSec":12616,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:40:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:40:34","Uptime":"0T03:35:16","UptimeSec":12916,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:45:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:45:34","Uptime":"0T03:40:16","UptimeSec":13216,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"PKHA","BSSId":"D0:17:C2:E8:6B:50","Channel":11,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:50:34 MQT: hole/CE14/tele/STATE = {"Time":"2020-07-11T10:50:34","Uptime":"0T03:45:16","UptimeSec":13516,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Dimmer":49,"Color":"7D0002","HSBColor":"359,100,49","Channel":[49,0,1],"Scheme":0,"Width":1,"Fade":"OFF" [... cut ...]
### TO REPRODUCE
_Steps to reproduce the behavior:_

None just left alone

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

setoption13 1 to remain

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I am looking to use button for single clicks only - no multiple, no Hold, nothing else at all

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-12:
Pls enable more logging with weblog 3 to find out where the ``SetOption13 0`` comes from. I recon it's from here:
```
          if (Settings.flag.button_single) {           // SetOption13 (0) - Allow only single button press for immediate action
            if (Button.hold_timer[button_index] == loops_per_second * hold_time_extent * Settings.param[P_HOLD_TIME] / 10) {  // SetOption32 (40) - Button held for factor times longer
              snprintf_P(scmnd, sizeof(scmnd), PSTR(D_CMND_SETOPTION "13 0"));  // Disable single press only
              ExecuteCommand(scmnd, SRC_BUTTON);
            }
          } else {
``` 
where a very long press on the button will reset it to 0.

ascillato2, 2020-07-14:
Hi, any news on this?

s1nbad, 2020-07-23:
Sorry, been away.
The button was connected to a vibration sensor with multiple activation's in a very short period of time (ms).
The multiple activation's were so short that it was taken as a single press and affected setoption32 (also reduced to improve response time).  Unfortunately setoption32 cannot be disabled and it keeps resetting setoption13 ( consider setoption32 = 0, disabling ability to reset device).  Not the first time I have been affected by features designed to assist in recovering devices.

Anyway, this is completely non standard and not really an issue for most.
I changed to a switch and used rules all works, thx

closed 2020-07-23 by ascillato2