arendst/Tasmota #8882: MI32 Flora send old data from sensor for days while sensor is out of range (xsns_62_MI_ESP32.ino)

d0m1n1qu3, 2020-07-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I found out that if a MI Flora sensor is on the border from in range to out of range some pakets lost and if the battery goes down the sensor data will not arrive the ESP32. but the Tasmote will send the old known data endless until you reboot the ESP32

i could try to fix that but what to discuss this first, may be i have done a misstake ;-)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____ last dev commit 11.07.2020
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: _____VSCode
- [x] Flashing tools used: _____ VSCode
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-07-11:
This is a known behavior for all BLE-sensors.

d0m1n1qu3, 2020-07-11:
but it would be easy to ignore sensors which send nothing in a time window x for example 10minutes. 
or is it already disscused?

Staars, 2020-07-11:
> but it would be easy to ignore sensors which send nothing in a time window x for example 10minutes.
Yes.
> or is it already disscused?
Yes.

d0m1n1qu3, 2020-07-11:
> > or is it already disscused?
> 
> Yes.
> 

where? can i join? 
is this function not wanted or missing time to do it? :-)

Carsten3P, 2020-07-11:
imho a very good idea, tasmota should avoid sending old values over a long time without ensuring they are correct.

d0m1n1qu3, 2020-07-12:
i have made some changes .. https://github.com/d0m1n1qu3/Tasmota/tree/MI32_offline
but you have now insert another approach, maybe you want to check my also :-)

Staars, 2020-07-13:
A direct bridging of the sensor data messages was always a development goal. So I‘d rather will pass all messages in (near) real time to MQTT, send a the usual tele message and then simply delete the sensor data, which will not be sended again with the next tele message (unless new data arrives).
If you really care about the timestamp of the last measurement, you get it automatically with the last MQTT-message. No need to add another property.

d0m1n1qu3, 2020-07-13:
ok understand. but my timestamp was only for looking how old the data is which we received via BLE to decide if we want to send them.
but your approach to send every data from BLE only one per mqtt is also very nice..

Staars, 2020-07-13:
I would try to implement it in the next time and then we'll see, if this is enough to fulfill the needs. My goal is to reach this with minimal code size increase, because more things are already in the pipeline.

mark078, 2020-07-14:
This has nothing directly to do with the topic, but will be read the soil moisture and fertilizer values correctly? 
I have two Mi Flora sensors where the two readings (Moisture & Fertility) are incorrect. On both Mi Floras I have updated the firmware to the latest version.

"Flora-65xxxx":{"Temperature":22.6,"Illuminance":39,"Moisture":1077149696,"Fertility":1080139776}

Tasmota Version | 8.3.1.6 on ESP32

Staars, 2020-07-14:
No, this should be another issue. Please re-check (reflash, erase) and if this will persist, please open a new issue.

closed 2020-10-04 by stale[bot]