arendst/Tasmota #8883: Ewelink GD-DC5 garage door controller - Support for configuration

tiagofreire-pt, 2020-07-11:
### PROBLEM DESCRIPTION

I flashed this controller with Tasmota, successfully. But its relay is clicking all the time, even with the  `Power1 = OFF`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): GD-DC5
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

11:12:40 RSL: stat/tasmota_95B9F1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:12:40 RSL: stat/tasmota_95B9F1/RESULT = {"Module":{"18":"Generic"}}
11:12:40 RSL: stat/tasmota_95B9F1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"29":"Relay1i"},"GPIO13":{"56":"Led1i"},"GPIO14":{"9":"Switch1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
11:13:03 RSL: stat/tasmota_95B9F1/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:13:04 RSL: stat/tasmota_95B9F1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:13:04 RSL: stat/tasmota_95B9F1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

11:13:23 RSL: stat/tasmota_95B9F1/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_95B9F1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:55","StartupUTC":"2020-07-11T10:12:28","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2020-07-11T11:00:19","SaveCount":25,"SaveAddress":"FB000"}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"345/699"}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PSA_WIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS5 = {"StatusNET":{"Hostname":"tasmota_95B9F1-6641","IPAddress":"192.168.11.214","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"C4:4F:33:95:B9:F1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_95B9F1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS7 = {"StatusTIM":{"UTC":"2020-07-11T10:13:23","Local":"2020-07-11T11:13:23","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:59","Sunset":"20:51"}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS10 = {"StatusSNS":{"Time":"2020-07-11T11:13:23","Switch1":"ON"}}
11:13:23 RSL: stat/tasmota_95B9F1/STATUS11 = {"StatusSTS":{"Time":"2020-07-11T11:13:23","Uptime":"0T00:00:55","UptimeSec":55,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"PSA_WIFI","BSSId":"2C:FD:A1:3E:31:58","Channel":12,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:13:48 CMD: weblog 4
11:13:48 RSL: stat/tasmota_95B9F1/RESULT = {"WebLog":4}
11:13:48 CFG: Saved to flash at FA, Count 26, Bytes 4096
11:13:53 HTP: Main Menu
11:13:54 WIF: Checking connection...
11:13:54 SRC: WebGui from 192.168.10.33
11:13:54 RSL: stat/tasmota_95B9F1/RESULT = {"POWER":"ON"}
11:13:54 RSL: stat/tasmota_95B9F1/POWER = ON
11:13:54 CFG: Saved to flash at F9, Count 27, Bytes 4096
11:13:55 HTP: Console
11:13:57 HTP: Main Menu
11:13:59 APP: Button1 multi-press 1
11:14:00 SRC: Button
11:14:00 RSL: stat/tasmota_95B9F1/RESULT = {"POWER":"OFF"}
11:14:00 RSL: stat/tasmota_95B9F1/POWER = OFF
11:14:00 CFG: Saved to flash at F8, Count 28, Bytes 4096
11:14:01 APP: Button1 multi-press 1
11:14:01 SRC: Button
11:14:01 RSL: stat/tasmota_95B9F1/RESULT = {"POWER":"ON"}
11:14:01 RSL: stat/tasmota_95B9F1/POWER = ON
11:14:01 CFG: Saved to flash at F7, Count 29, Bytes 4096
11:14:03 APP: Button1 multi-press 1
11:14:03 SRC: Button
11:14:03 RSL: stat/tasmota_95B9F1/RESULT = {"POWER":"OFF"}
11:14:03 RSL: stat/tasmota_95B9F1/POWER = OFF
11:14:04 CFG: Saved to flash at F6, Count 30, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Just toggle the power button on the board itself or Tasmota Web GUI.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Just toggling the relay once, accordingly to the buttons.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

This is the device: https://www.banggood.com/GD-DC5-WiFi-Smart-Switch-Garage-Door-Opener-Controller-App-for-iPhone-Android-p-1430647.html?cur_warehouse=CN

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-07-12 by ascillato2

Crewski, 2020-07-28:
@tiagofreire-pt did you happen to get this figured out?  I just received mine from banggood (same model as in your picture) and am working to get it setup.  Thanks.

tiagofreire-pt, 2020-07-28:
No, I didn't.

It seems the internal MCU has a role that tasmota can't control properly. I just flashed it back to the original firmware and put it to sale.