arendst/Tasmota #8888: EventGhost binding

kdschlosser, 2020-07-12:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Being able to connect to EventGhost

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

I am the administrator of a project called EventGhost (http://www.eventghost.net  https://github.com/EventGhost/EventGhost). EG (EventGhost) is a an automation program that runs on Microsoft Windows and is written in Python and C. I happened across this firmware for the ESP's by way of IRRemoteESP8266. One of the features that EG has built into it is handling of IR receivers that are able to be connected a variety of ways. EG has been around for quite some time (14 years) and there are sections of the code that needed to sprucing up. The IR decoding needed to be updated in a pretty bad way. 

I set out to see if I was able to locate a decoder that was written in Python or in C code that did not have an attachment to any kind of specialty hardware. I also wanted it to support as many protocols as possible. I came up empty handed in this search. I did locate several decoders that were written in Java and could decode via the command line, unfortunately it would take 2-3 seconds for the decoder to start and to decode the code. This would not work for decoding in real time. I then decided to write a decoder/encoder in pure Python and does not have any kind of an attachment to any device. I have over 150 protocols supported for decoding and encoding and I am now adding in support for air conditioners. There was someone that had posted an issue on IRRemoteESP8266's repo for the Carrier air conditioners and there was a link to this repo for what was being used. That is what lead me here.

I am always on the lookout to expand the functionality of EventGhost as I am sure you are on the lookout to do the same. EventGhost has close to 400 extensions available to it. It is able to get "events" or notifications of any changes made to the computer that it is running on from changing the display resolution to moving the mouse. It is also able to interface with many pieces of software from Excel to Media Player. It is also able to connect to external devices or even devices that are attached directly to the computer EG is running on.

You have an extremely large number of devices that are supported by your software. as does EventGhost. The really cool thing is that there are not that many overlaps. If a binding was made it would be beneficial to both programs if the binding was a 2 way street, that way your software could expand it's functionality through EventGhost and vice versa...

I saw that there is support for Zigbee, This is something I would love to add to EventGhost, I did not notice any support for ZWave in your software.. EG does support ZWave. The ESP's have limited resources and a full featured IR decoder is probably not possible. This is where EG could help out. and your software could help EG as well by being a WiFi IR transceiver.

The IR decoder I wrote is fast... on average it takes just a tad over a half a millisecond to decode an RLC stream, It also has full support for repeat codes so notification of button releases are available.

If there is interest in making a binding we can work together on it. I would need to write some code to make it possible on the EG side of things and we would need to come up with some kind of an API so we can extend each others programs.

I can see this being mutually beneficial for both programs.
Let me know if you are interested.

Thanks for taking the time.
Kevin

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-07-12:
Nice project. We rely on IRremoteESP8266 for IR decoding and encoding. It's a C++ lib optimized for the limited resources of ESP8266. 

The native interface for Tasmota is MQTT. As long as you can send and receive MQTT in your software, the integration should be easy.

kdschlosser, 2020-07-12:
EventGhost does support MQTT, however I try to keep the complexity down to as small as possible. MQTT adds another layer and a complex one at that. Does MQTT offer a control interface?? I thought it acted more like a data repository where multiple applications are able to collect say the data from a sensor reading. MQTT is not really my thing as I always try to make a direct connection to a device. So excuse my ignorance in this area.

What had initially peaked my interest was the possibility of using the ESP with your firmware and and IR receiver and transmitter to make a low cost WiFi IR blaster. To do this some kind of a socket connection would need to be made to pass the RLC stream to EG. and EG could send RLC to it to be blasted. IDK if you have ever had the pleasure of dealing with some of the newish IP based cable companies where they process the actual remote commands not locally but at the head end, This can sometimes be a horrible thing because of button lag. where there is a 2-3 second latency between pressing the button and when something actually happens. It really annoying. This kind of a phenomenon that should not be in someone's home automation. I worked really hard at getting the decode time down to 0.6 milliseconds on average. This is with all decoders enabled. I can still make the decode time 50 times faster by compiling the code as a python extension.

I do understand that you use IRremote for a decoder. While it is a fantastic decoder it is built for resource restricted devices, and as such it is not able to support and run concurrently the large number of available decoders. It is also written in a manner that does not allow the decoders to be enabled and disabled real time. You would have to flash the firmware if you wanted to do this. It isdone this way to reduce the resource footprint and I can understand that. If there is a way to turn off all of the decoders and only deal with the raw RLC data and a persistent connection to EG is made where the data can pass between the ESP and EG this limitation no longer exists. It also provides additional storage and ram so a user can add more to the functionality to the ESP. If there is an open socket connection the ESP would only need to have a small buffer for the RLC. it could shoot the timings across the connections as it creates them. 

I started programming with using Arduinos specifically the Uno. I know all about resource limitations and how valuable those resources are.. the ESP's do not suffer as mush as the Uno does. But the ESP defiantly does not have the resources of a PC behind it and the question then becomes why can't it? 

I also like the ESP because it is not a purpose built device. So it can receive IR and also send IR, But it can also be a temperature and humidity sensor at the same time, also a light sensor or motion sensor. It is a multipurpose device. So instead of having all these boxes that all do a specific task. it can be trimmed down to a single device that does a bunch of tasks.. The best part is it's cheap!

I am all about speed when dealing with anything Home Automation. turning a light on and not getting a status change for several minutes is not good. If you have a script set up so that a light turns on when you turn another light on it wouldn't be useful if it was 2 minutes before the script would run. This is why I like direct connections if at all possible, it removes quite a bit of this type of latency.

meingraham, 2020-07-12:
MQTT is a messaging protocol, not really a repository (even though it can retain a copy of the last message from a device). So rather than a direct connection, information from a device would be transmitted "through" the message broker and then relayed to other subscribers (e.g., EventGhost) and vice versa.

kdschlosser, 2020-07-12:
OK. can it be used to tell your firmware to turn a zigbee light on and off? that kind of thing?

meingraham, 2020-07-12:
Yes, the contents of the messages transmitted via MQTT are completely flexible. Tasmota can receive commands and settings via MQTT as well send status, telemetry, and user customizable messages.

kdschlosser, 2020-07-12:
Does that functionality also work in reverse? Can Tasmota send commands and receive status updates?

meingraham, 2020-07-13:
Tasmota can send messages... and those can be commands EventGhost recognizes.

What should Tasmota do with status updates?

closed 2020-08-15 by stale[bot]