arendst/Tasmota #8889: Random shutoff Exception 3

mobamoba, 2020-07-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I will get random Exception 3 reboots. They won't happen for days at a time then will suddenly occur multiple times in a row.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____ STITCH by Monoprice 27937 Power Monitoring Plug 
- [ x] Tasmota binary firmware version number used: _____8.3.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: _____tuya-convert then used Tasmota to flash most recent
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`: I"m not sending a command - it's just rebooting itself
```
  Configuration output here:
Other than my MQTT broker and Wifi info, everything else is left as default.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:33:38 MQT: stat/Lrac/STATUS = {"Status":{"Module":0,"DeviceName":"Lrac","FriendlyName":["Lrac"],"Topic":"Lrac","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:33:38 MQT: stat/Lrac/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"10T20:13:52","StartupUTC":"2020-07-01T14:19:46","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-06-30T17:35:59","SaveCount":771,"SaveAddress":"F9000"}}
11:33:38 MQT: stat/Lrac/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"378/699"}}
11:33:38 MQT: stat/Lrac/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["n3tEkcAj",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:33:38 MQT: stat/Lrac/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:33:38 MQT: stat/Lrac/STATUS5 = {"StatusNET":{"Hostname":"Lrac-6935","IPAddress":"192.168.1.31","Gateway":"192.168.1.150","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.150","Mac":"BC:DD:C2:E1:5B:17","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
11:33:38 MQT: stat/Lrac/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.46","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E15B17","MqttUser":"DVES_USER","MqttCount":124,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:33:38 MQT: stat/Lrac/STATUS7 = {"StatusTIM":{"UTC":"2020-07-12T10:33:38","Local":"2020-07-12T11:33:38","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:00","Sunset":"20:51"}}
11:33:38 MQT: stat/Lrac/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:33:38 MQT: stat/Lrac/STATUS10 = {"StatusSNS":{"Time":"2020-07-12T11:33:38","ENERGY":{"TotalStartTime":"2020-06-30T17:33:46","Total":1197.627,"Yesterday":84.889,"Today":28.192,"Power":7537,"ApparentPower":7537,"ReactivePower":0,"Factor":1.00,"Voltage":282,"Current":26.690}}}
11:33:38 MQT: stat/Lrac/STATUS11 = {"StatusSTS":{"Time":"2020-07-12T11:33:38","Uptime":"10T20:13:52","UptimeSec":936832,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":124,"POWER":"ON","Wifi":{"AP":1,"SSId":"n3tEkcAj","BSSId":"14:DD:A9:76:09:6C","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}}
11:33:38 MQT: stat/Lrac/STATUS12 = {"StatusSTK":{"Exception":3,"Reason":"Exception","EPC":["401018dd","00000000","40000f68"],"EXCVADDR":"40034ce8","DEPC":"00000000","CallChain":["401019e0","401019e0","40101c7f","40255de0","40255de0","4024e6ed","4024e7e8","4000050c","4024f26f","40101c7f","4024f298","402511b5","40101cb6","40255511","40101c7f","40267200","4024e5a9","4024e3cf","402716aa","402716bb","40266e2b","4026cebf","40000f49","40000f49","40000e19","40105a41","40105a47","4010000d","40265d14","40265cc5","401008f8"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
STATUS12 = {"StatusSTK":{"Exception":3,"Reason":"Exception","EPC":["401018dd","00000000","40000f68"],"EXCVADDR":"40034ce8","DEPC":"00000000","CallChain":["401019e0","401019e0","40101c7f","40255de0","40255de0","4024e6ed","4024e7e8","4000050c","4024f26f","40101c7f","4024f298","402511b5","40101cb6","40255511","40101c7f","40267200","4024e5a9","4024e3cf","402716aa","402716bb","40266e2b","4026cebf","40000f49","40000f49","40000e19","40105a41","40105a47","4010000d","40265d14","40265cc5","401008f8"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Nothing, it's random.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect it not to reboot on its own.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-12:
Try latest development version and report back

mobamoba, 2020-07-12:
Okay will do - since it's random I'll give it a week or so on 8.4.0 and let you know if the error returns or not.

mobamoba, 2020-07-16:
This issue still exists on 8.4.0 only there's nothing in STATUS 12 anymore, by which I mean it's still happening i.e. the switch is rebooting itself, but it's no longer listing any Exception codes.

arendst, 2020-07-17:
So no software error then.

You'll need to enable more logging to find out what triggers a restart. 

It might be a power supply issue.

mobamoba, 2020-07-17:
Okay. Which logs - all of them?

meingraham, 2020-07-17:
Since it's rebooting, you may have to send logs to a remote logger (SysLog) in order to capture the event. Start with `WebLog 4` (default 2) to see if you can capture the event. If not, set up SysLog and set `SysLog 4` (default 0).

digiblur, 2020-07-17:
Might be a different issue but I can say I've been trying to track this down for a few weeks on what causes it.  Random device types, setups, etc. 
[image]

Started with 8.3.1 as my 8.2 stuff is fine.  I do believe to have fixed a few of them buy doing the reset 5 thing and setting things back up manually (not doing a config import).  I was trying to reproduce it easily before I opened an issue on it.

mobamoba, 2020-07-23:
I'm now thinking there were two issues. One was the Exception 3 reboot issue which I think is solved with the dev version - according to the logs, the device has been on for 11 days straight with no exceptions or reboots.

The second issue is not Tasmota-specific though I'll note it here in case others have the same setup. I'm using Tasmota with the Node Red Tasmota node and I noticed it was occasionally sending phantom MQTT power off messages. It was extremely random with a week or so going by without error then a random shutdown followed by an immediate startup - which looks like a reboot but isn't. I've since detached the node and am just sending MQTT commands directly to Tasmota.

In any event, for me Exception 3 is gone and I'm guessing the phantom MQTT power off command is a bug in the Tasmota node red node and not with Tasmota itself.

Jason2866, 2020-07-31:
@mobamoba  any news please close when the Tasmota related issue is solved.

mobamoba, 2020-07-31:
Because it's intermittent, it's very difficult to pin down, but I'm still having the problem. I've been trying to do a process of elimination of figure out whether it's coming from Tasmota or from Node Red which is what I use to control the plug on which I have Tasmota installed. I'm 99% convinced at this point that it's something in Tasmota though I really can't say for sure. For example, the plug shut itself off today then a few seconds later turned itself back on. Here's what was in the debug log:

14:42:17 WIF: Checking connection...
14:42:18 MQT: tele/Lrac/STATE = {"Time":"2020-07-31T14:42:18","Uptime":"19T01:08:15","UptimeSec":1645695,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"ON","Wifi":{"AP":1,"SSId":"n3tEkcAj","BSSId":"14:DD:A9:76:09:6C","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:42:18 MQT: tele/Lrac/SENSOR = {"Time":"2020-07-31T14:42:18","ENERGY":{"TotalStartTime":"2020-06-30T17:33:46","Total":3397.618,"Yesterday":176.944,"Today":66.688,"Period":284,"Power":7239,"ApparentPower":7239,"ReactivePower":0,"Factor":1.00,"Voltage":283,"Current":25.615}}
14:42:37 MQT: Received Topic "cmnd/Lrac/POWER1", Data Size 3, Data "OFF"
14:42:37 SRC: MQTT
14:42:37 CMD: Group 0, Index 1, Command "POWER", Data "OFF"
14:42:37 MQT: stat/Lrac/RESULT = {"POWER":"OFF"}
14:42:37 MQT: stat/Lrac/POWER = OFF
14:42:37 WIF: Checking connection...
14:42:37 CFG: Saved to flash at F6, Count 1324, Bytes 4096
14:42:42 MQT: Received Topic "cmnd/Lrac/POWER1", Data Size 2, Data "ON"
14:42:42 SRC: MQTT
14:42:42 CMD: Group 0, Index 1, Command "POWER", Data "ON"
14:42:42 MQT: stat/Lrac/RESULT = {"POWER":"ON"}
14:42:42 MQT: stat/Lrac/POWER = ON
14:42:42 CFG: Saved to flash at F5, Count 1325, Bytes 4096

So that looks like it received an MQTT power off command followed very quickly by a power on command right? Only I had shut down Mosquitto earlier to rule that out so I know it wasn't some retention issue or random thing Mosquitto was doing because Mosquitto was off. And I only have one Node Red node that sends power commands to that plug and I had put a debug node on it to see if a random command was coming through from Node Red. And, no, it wasn't.

So I don't know. All I know is Mosquitto isn't sending a power off then power on command nor is Node Red nor do I have any other devices sending anything to that plug which is why I think it's something within Tasmota itself but I obviously can't say that for sure because I really don't know.

Jason2866, 2020-07-31:
Tasmota gets via mqtt a command to power off / on. 
This log entry is only created when a mqtt message was received `14:42:37 SRC: MQTT`
You have to search in your mqtt "world". Tasmota just does what it told.

mobamoba, 2020-08-01:
I realize what I'm about to say is going to sound impossible, but I can definitely say the MQTT shutoff command is not coming from my system. I know this because I put a debug monitor on the MQTT on my system and I happened to be watching as the plug shut off with absolutely no MQTT command being sent to it.

Something within Tasmota is telling it to shut off after which my system is telling it to turn back on (the power on command I can see in my MQTT debug. As I mentioned in my initial report, this happens multiple times in a row - just now, as I'm typing this, it happened 4 times, again with nothing being sent by MQTT. At some point it works properly then in a few days, it will do this again, i.e. turn itself off multiple times in a row over roughly a 30 minute period and then will go back to working normally again. 

All I can say is that something within Tasmota is sending a cmnd power off and if there's something further I can do to debug please let me know.

ascillato, 2020-08-01:
Seems that your issue is a retained message in your MQTT broker. Any wifi or mqtt reconnection will make your broker to automatically RESEND this message (and as it is automatic, may be is not displayed in your mqtt debug log - depending on the config).

Try the following command in your tasmota console:

```lua
publish2 cmnd/Lrac/POWER1
```

This should clear any retained message related to that topic.

Please, report back if this solves or not your issue. Thanks.

mobamoba, 2020-08-02:
Okay I'll try that. Also just FYI I have persistence set to false in my mosquitto.conf file so I'm not sure how messages could be retained in any event.

digiblur, 2020-08-02:
Some persistence is a good thing.

mobamoba, 2020-08-02:
Not for my setup - I prefer to have statuses checked from the devices directly rather than kept in a db.

Also, publish2 cmnd/Lrac/POWER1 didn't solve the problem as it just happened again multiple times.

ascillato, 2020-08-02:
Ok, which MQTT broker are you using?

Besides, as this problem occurs quite often in your setup, try to turn off your broker for some minutes to see if this "mqtt ghost switching" happens again.

Please post the console output if happens again.

mobamoba, 2020-08-02:
I use Mosquitto which was the default included with my Ubuntu installation. Because the issue is sporadic and sometimes a week will go by with no problems, I can't realistically shut down my broker as it connects to other things.

However, I detached my broker from the device so Mosquitto is no longer sending messages to the plug though it's receiving them.

And pretend for a minute that I shut down my broker - or put a fake IP into Tasmota's MQTT broker config so Tasmota is connected to no broker at all - and the issue happens again. What might be causing it? I understand Tasmota can't send itself MQTT commands, but is it possible that a bug is making Tasmota think it's received one then acting as if it had? I'm asking because this shutdown was associated with Exception 3 in a prior version and, while that Exception is no longer showing up, the problem is still occurring meaning perhaps the bug wasn't 100% squashed.

ascillato, 2020-08-02:
As we can't reproduce your issue, and we don't have reports of this from other users, may be the firmware is corrupt in your device. So, please, try to erase all flash and flash the precompiled bin ( Tasmota.bin from http://thehackbox.org/tasmota )

With this "start from scratch" we can exclude failures on previous firmware flashings

mobamoba, 2020-08-02:
Actually this is a new install - I only did tuya-convert on this plug a few weeks ago.

One thing I was thinking of trying over the next week or so was turning the plug on and leaving it completely disconnected from MQTT by putting in a non-existent IP address for the broker in the Tasmota config. If it turns itself off in that case, we'll know it has to be Tasmota since it could be nothing else. And as I never had this problem with the original Tuya firmware and it's been plugged into the same device for the past few years, I think we can rule out something hardware-related in the plug itself.

I'll see if there's something else I can think of to try and report back if I find anything.

ascillato, 2020-08-02:
As your device is a plug and serial flashing is not easy on those, please, instead of erasing and flashing, let's try `reset 5` command on the console in order to clean the flash.
After that, all settings and the sdk calibration will be wipe out, except for wifi credentials, so you will be able to continue reaching that device in your wifi network. 
After the command, please, reconfigure your device and let's try again. 
Thanks for all the testings

mobamoba, 2020-08-02:
Okay I just did that. I'll see if the problem shows up again and post here if it does.

mobamoba, 2020-08-03:
Thanks for bearing with me through this. Having gone through my system node by node (!!), I'm beginning to wonder if the problem was a power state reporting lag, i.e. the device was on but because there was a delay in updating the power state, my system toggled the switch thus turning it off at which point the switch reported it was off and my system turned it back on etc.

I'm not sure but if that's the case then I've made changes which should eliminate that as a problem. In any event I think we can close this because the Exception 3 is definitely gone and, unlike a few posts ago :), I'm now thinking the likeliest reason for this problem has to do with how my system is determining whether the switch is on or off and not anything with Tasmota.

closed 2020-08-03 by ascillato2