arendst/Tasmota #8893: Allow access to analog A0 value in script

pwbecker, 2020-07-12:
**Have you looked for this feature in other issues and in the docs?**  Yes

**Is your feature request related to a problem? Please describe.**  
It is not possible to access the raw Analog value of A0 from within a script

**Describe the solution you'd like**  
Add a[x], where x is usually 0 but allows for additional analog pins

**Describe alternatives you've considered**  
Have looked through xdrv_10_scripter.ino to see if its there but not documented

**Additional context**  No
**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-07-12:
Look at #8860

ascillato2, 2020-07-12:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-12 by ascillato2

pwbecker, 2020-07-13:
> Look at #8860

Thanks Kugelkopf. I didn't find that issue when searching because it has a totally different title.  I am still trying to figure out how to set a script variable to the raw adc value - don't see it in any of the examples in the scripting docs, and v=Analog#A0 gives var not found: Analog#A0". I have been looking for a day now - will keep trying!

kugelkopf123, 2020-07-13:
@pwbecker 

```
>D
an=0

>T
an=ANALOG#A0
=>print %0an%
```