arendst/Tasmota #8905: tasmota32 flashing issue

amitubale, 2020-07-14:
getting an error
Detected overlap at address: 0x8000 for file: partitions.bin

C:\Python27\Scripts>esptool.py --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: partitions.bin

closed 2020-07-14 by close-issue-app[bot]

Jason2866, 2020-07-23:
Do not use browser function `save as` in gitbub. Use the download button

amitubale, 2020-07-23:
Hi Jason2866 - is that message for me. DO that mean i should not be downloading  the files from the tasmota32 firmware folder ?
I dont see the option to download the zip file. Any help woudl be appriciated. thanks.

Jason2866, 2020-07-23:
If you click on the file(s) it opens another window. In this webbrowser view you see a small button `download`

amitubale, 2020-07-23:
THats what i did
Tasmota/firmware/tasmota32/ESP32_needed_files
right click on the mouse and downloaded all the files to c:\temp

then downloaded the Tasmota/firmware/tasmota32/tasmota32.bin by right click on the mouse and downloaded all the files to c:\temp

I have used 3 laptopns now and 3 diffrent version so Python 3.8, 3.6, and 3.4 on the last one

i did the "pip install esptool"

and did cut-pasted the command in the readme file. I hope someone can help me figure out what am i doing wrong or the exact right steps to get the firmware on the esp32

I has used platform i/o 3 years a go to make my custom tasmota.bin with my wifi ssid and password. the new steps are to add the addon in VS code..not sure how to complie my own bin file.

Jason2866, 2020-07-23:
NO no right click with the mouse! Download button!!!

amitubale, 2020-07-23:
Oh Man.....thank you sir. that did it..it works...its alive!...
thank you for helping me out with this.
[image]