arendst/Tasmota #8906: Tasmota32.bin Flashing issue . Error

amitubale, 2020-07-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Getting an error when flashing Tasmota32.bin to esp32.
Detected overlap at address: 0x8000 for file: partitions.bin

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
C:\Python27\Scripts>esptool.py --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: partitions.bin

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- install pyton 2.7.x in windows
- downlaod all files https://github.com/arendst/Tasmota/ ( tasmota32.bin, boot_app0.bin, bootloader_dout_40m.bin, partitions.bin)
- run the command from teh readme.txt

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-07-14:
Hi, 

Sorry your flashing command is wrong. You need to flash ONLY Tasmota32.bin file. 

Please, take a look to the docs on how to flash esp32 and if you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2020-07-14:
Hi,

Please, try the following command:

esptool.py --port COM4 -- tasmota32.bin

amitubale, 2020-07-14:
got an error again...are you sure its "-- tasmota32.bin"

C:\Python27\Scripts>esptool.py --port COM4 -- tasmota32.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: '--' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')

C:\Python27\Scripts>esptool.py --port COM4 tasmota32.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: 'tasmota32.bin' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')

C:\Python27\Scripts>

Jason2866, 2020-07-14:
The command in the docs is correct. Flashing does work and Tasmota is running.
````
c:\temp>esptool.py --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 tasmota32.bin
esptool.py v2.8
Serial port COM4
Connecting........_____.
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: fc:f5:c4:2f:8f:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 15872 bytes to 10320...
Wrote 15872 bytes (10320 compressed) at 0x00001000 in 0.1 seconds (effective 971.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 127...
Wrote 3072 bytes (127 compressed) at 0x00008000 in 0.0 seconds (effective 1584.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4809.6 kbit/s)...
Hash of data verified.
Compressed 1071248 bytes to 629097...
Wrote 1071248 bytes (629097 compressed) at 0x00010000 in 10.4 seconds (effective 823.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

ascillato2, 2020-07-15:
> got an error again...are you sure its "-- tasmota32.bin"

Sorry you are right. The command I post was wrong. I was writing from my phone. 

As Jason's, the full command from the docs also works for me.

Try to reinstall esptool.py

Also, you can use platformio for uploading the firmware. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-15 by ascillato2

Jason2866, 2020-07-23:
For users finding this. It is a wrong download from github. Do NOT `save as` You have to use the download button!

christianwis, 2020-08-14:
jason2866 i have same problem... could you solve ?

thanks

Jason2866, 2020-08-14:
Yes, in all cases the needed files where wrong downloaded. Do NOT use brower function for saving
use the button `Download` provided from github.

schrej-zz, 2020-10-13:
Hi,
I still don't get it.
If I go ito the mentioned Github directory
https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota32
I don't see any Download button.
Joerg

Jason2866, 2020-10-13:
Click onto the file and than you have this view 
[image]

vkeiper, 2021-02-26:
> 
> 
> For users finding this. It is a wrong download from github. Do NOT `save as` You have to use the download button!

I want to buy this person a beer or coffee. I have been troubleshooting this for hours and this was the issue.

Jason2866, 2021-02-26:
Glad my small note helped