arendst/Tasmota #8912: MQTT broker connection problems and DNS

iperomero, 2020-07-15:
### PROBLEM DESCRIPTION
I have about 18 devices (Sonoff basic, sonoff mini, touch, th10, etc.) and **all of them** have the same problem:

Eventually, the connection with the MQTT broker fails, when I set the address of the broker using the domain name (broker.***.com) , and it never can't reconnect to the broker  until I reboot the device. The device gets blocked, because it seems that the reconnection hangs it)

If I set the broker address using the IP address, I never had any problem. This is why I suspect it is a problem with DNS resolution.

### REQUESTED INFORMATION
THe problem occurs from Tasmota 6 to Tasmota 8.4 (I tested about 10 versions, both self compiled and pre-built versions), in 18 devices in different locations. Tested with two mosquitto brokers, ant using 1883 and secure 8883 ports. 

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS

### ADDITIONAL CONTEXT

Jason2866, 2020-07-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

myfingerhurt, 2020-07-17:
I have 4 of sonoff_pows, all of them are behaving strangely after I have replaced MQTT IPs with domain names.

Same problems here MQTT with domain name always gets stuck and sometimes, the webUI becomes extremely slow to come up, even appears half loaded webUI with only few letters . MQTT failed and came back repeatedly 

I manully set ipaddress4 to 8.8.8.8, but no obviously better. 

8.3.1(tasmota)
```
22:24:03 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:24:03","Uptime":"0T00:04:10","UptimeSec":250,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"YOUSSID","BSSId":"XX:XX:XX:XX:DB:XX","Channel":1,"RSSI":100,"Signal":-20,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:24:03 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2020-07-17T22:24:03","ENERGY":{"TotalStartTime":"2020-02-10T18:21:26","Total":110.080,"Yesterday":0.145,"Today":0.003,"Period":0,"Power":24,"ApparentPower":34,"ReactivePower":24,"Factor":0.71,"Voltage":102,"Current":0.333}}
22:24:28 MQT: Attempting connection...
22:24:37 MQT: Connected
22:24:37 MQT: tele/sonoff_pow1/LWT = Online (retained)
22:24:37 MQT: cmnd/sonoff_pow1/POWER = 
22:24:42 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:24:42","Uptime":"0T00:04:49","UptimeSec":289,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"YOUSSID","BSSId":"XX:XX:XX:XX:DB:XX","Channel":1,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:24:42 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2020-07-17T22:24:42","ENERGY":{"TotalStartTime":"2020-02-10T18:21:26","Total":110.080,"Yesterday":0.145,"Today":0.003,"Period":0,"Power":24,"ApparentPower":34,"ReactivePower":24,"Factor":0.72,"Voltage":102,"Current":0.333}}
22:25:12 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:25:12","Uptime":"0T00:05:19","UptimeSec":319,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"ON","Wifi":{"AP":1,"SSId":"YOUSSID","BSSId":"XX:XX:XX:XX:DB:XX","Channel":1,"RSSI":100,"Signal":-20,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:25:12 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2020-07-17T22:25:12","ENERGY":{"TotalStartTime":"2020-02-10T18:21:26","Total":110.080,"Yesterday":0.145,"Today":0.004,"Period":0,"Power":24,"ApparentPower":34,"ReactivePower":24,"Factor":0.71,"Voltage":102,"Current":0.334}}
22:25:28 MQT: Attempting connection...
22:25:31 MQT: Connected
22:25:31 MQT: tele/sonoff_pow1/LWT = Online (retained)
22:25:31 MQT: cmnd/sonoff_pow1/POWER = 
22:25:44 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:25:44","Uptime":"0T00:05:51","UptimeSec":351,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"YOUSSID","BSSId":"XX:XX:XX:XX:DB:XX","Channel":1,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:25:44 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2020-07-17T22:25:44","ENERGY":{"TotalStartTime":"2020-02-10T18:21:26","Total":110.081,"Yesterday":0.145,"Today":0.004,"Period":0,"Power":24,"ApparentPower":34,"ReactivePower":24,"Factor":0.72,"Voltage":102,"Current":0.334}}
22:26:14 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:26:14","Uptime":"0T00:06:21","UptimeSec":381,"Heap":8,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"YOUSSID","BSSId":"XX:XX:XX:XX:DB:XX","Channel":1,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:05"}}
22:26:14 MQT: tele/sonoff_pow1/SENSOR = {"Time":"2020-07-17T22:26:14","ENERGY":{"TotalStartTime":"2020-02-10T18:21:26","Total":110.081,"Yesterday":0.145,"Today":0.004,"Period":0,"Power":24,"ApparentPower":34,"ReactivePower":24,"Factor":0.72,"Voltage":101,"Current":0.334}}
22:26:44 MQT: tele/sonoff_pow1/STATE = {"Time":"2020-07-17T22:
```

* MQTT with Domain name
[image]

* MQTT with IP
[image]

IJsbeer-NL, 2020-10-07:
Although this issue is closed, I share the same experiance. Not always, but sometimes de MQTT server is not resolved correctly resulting in a can'not connect message on the console screen. Replacing the domainname by the IP address resolves the issue in all situations.