arendst/Tasmota #8913: Upgrades to 8.4.0 require hard reboot

richard-scott, 2020-07-15:
### PROBLEM DESCRIPTION
When upgrading from 8.3.1 to 8.4.0 the devices require a hard reboot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R2, Wemos D1 Mini, Sonoff RF Bridge
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Upgrade an 8.3.1 device to 8.4.0 OTA.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I'd expect it to reboot and be online without cutting the power.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-07-15:
Why?

arendst, 2020-07-15:
Just verified and noticed this indeed (for the first time as I normally use a stripped down version which works correctly)

First impression is that the bootloader fails to compare the downloaded and optionally unzipped code by reporting this:
```
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
@cp:0
cmp:
 ets Jan  8 2013,rst cause:3, boot mode:(3,7)

ets_main.c 
```
where it hangs waiting for a restart. As can be seen the copy to the final destination finished OK (``@cp:0``) the compare failed to report ``cmp:0``.

For now I suspect a file size issue with regards to the (Arduino core) bootloader and will experiment.

arendst, 2020-07-15:
Narrowing down the issue it seems core 2.7.2 uses a different bootloader than core 2.7.1.

Core 2.7.1 has no issues with the latest Tasmota versions.
Core 2.7.2 fails with files over a certain size (> 560k).

The core 2.7.2 bootloader tries to compare the downloaded binary file with (parts) of the loaded (= copied down the binary to it's final destination) and fails almost immediatly probably due to overwritten parts of flash. This will need to be investigated by the core guys. 

I'm trying to find out what core commit introduced the new bootloader compare feature in core 2.7.2

arendst, 2020-07-15:
Core issue reported as https://github.com/esp8266/Arduino/issues/7458

Jason2866, 2020-07-15:
@arendst  will build a modified Arduino core based on 2.7.2 and revert commit https://github.com/esp8266/Arduino/commit/51daecc236fbff0d1ed30a9c63a3af6e8b5c7392
Should i include the PWM changes too?

arendst, 2020-07-15:
Yes pls do.

arendst, 2020-07-16:
Didn't mean to close it

arendst, 2020-07-20:
As it's related to Arduino Core version 2.7.2 soon to be replaced by 2.7.3.1 this issue will be closed.

Tasmota on platformIO uses a patched core version anyway so this issue is solved.

closed 2020-07-20 by arendst