arendst/Tasmota #8919: Möglichkeit für die Unterstützung des ULN2003 Stepper Motor Driver

Basti149, 2020-07-15:
**Have you looked for this feature in other issues and in the docs?**  
Yes, unfortunately nothing found.

**Is your feature request related to a problem? Please describe.**  

In Tasmota software the motor driver A4988 is supported. Is there a way to support the ULN2003 motor driver?
Unfortunately I have not found any instructions on how I can connect this ULN2003 to the ESP8266 and how I can configure it with Tasmota. There are various instructions or example programs that run with this motor module (ULN2003) and ESP8266. Could you also integrate it with Tasmota.
Thanks in advance.

Und zwar wird in Tasmota Software der Motor Treiber A4988 Unterstützt. Gibt es auch eine Möglichkeit den Motor Treiber ULN2003 zu Unterstützen.
Leider habe ich keine Anleitung gefunden wie ich diesen ULN2003 mit dem ESP8266 verbinden kann und wie ich diesen ihn Tasmota konfigurieren kann. Es gibt diverse Anleitungen bzw Beispiel Programme die mit diesen Motor Modul (ULN2003) und ESP8266 laufen. Könnte man es auch ihn Tasmota integrieren.
Danke im voraus.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-07-15:
Since ULN2003 is just an driver device and has no logic inside there is no need for a support.
There is nothing to configure. If you want to drive stepper motors use the supported A4988.
There will never be low level driver support for stepper motors. This is needed when using only ULN2003 driver IC without any logic inside

Basti149, 2020-07-15:
Also, the ULN2003 is not managed by Tasmota, is it? I guess I have to get another driver and start the engine. I've heard the drivers are all the same. The following board is also possible.

https://tasmota.github.io/docs/_media/peripherals/stepper-motor2.png

I would have to connect it as follows or.

https://user-images.githubusercontent.com/34340210/66860253-d91a8600-ef5a-11e9-9195-c816e329abb8.jpg

Auch wird der ULN2003 von Tasmota nicht geführt oder? Da muss ich mir wohl ein anderer Fahrer zulegen und ein den Motor Anlegen. Ich habe die Treiber sind alle gleich gehört. Geht auch folgendes Board.

https://tasmota.github.io/docs/_media/peripherals/stepper-motor2.png

Da müsste ich es wie folgt anschließen oder.

https://user-images.githubusercontent.com/34340210/66860253-d91a8600-ef5a-11e9-9195-c816e329abb8.jpg

ascillato2, 2020-07-16:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-07-16 by ascillato2

Jason2866, 2020-07-16:
Yes, your last assumptions are correct.
Ja, die letzten Ausführungen / Annahmen sind korrekt