arendst/Tasmota #8921: Shutter support with MCP23017

is4it-lab, 2020-07-16:
**Have you looked for this feature in other issues and in the docs?**  

yes, the feature is mentioned in the documentation, but how-to is missing:

https://tasmota.github.io/docs/Blinds-and-Shutters/
...
With four shutters, eight Relay<x> components are needed. If manual operation switches (Switch<x> or Button<x> pairs) are also used, additional input GPIO are required. The ESP82xx device may not have enough free GPIO to support all the shutter connections required. A GPIO expander such as a PCF8574 or MCP230xx can be used.
...

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

The relays connected to MCP23017 ports con be configured as output with sensor29 commands and can be switched on7of with sensor29 command.
The connection to a shutterrelay cannot be done. What would be the appropriate command ? - shutterrelay1 D1 isn't working.
(in analogy the PCF8574 uses numbers above the configured reley ports on the dvive itself and show them in the webgui) 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

a functionality and description to assign MCP23017 out ports to a shutterrelay and integration in the web giu like the PCF8574

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

because the PCF8574 has only 8 ports and some disadvances, the full integration of MCP23017 would be preferable.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-07-24:
Hi, one question. With the MCP23017 do the relays all showed up with relay 1....8 or do they look different. As long as this condition is met, the shutter should work with any GPIO extender like the PCF8574 . This also just add additional relays,

stefanbode, 2020-07-24:
I just see I have a MCP23017 here on my Desk. I will try to get it run with TASMOTA and see how it will show up in TASMOTA. If the GPIO show up completely different as I assume actually this will get very complicated. This was the reason why the PCF8574 just exposes all his RELAYS as additional RELAYS. to it is transparent to applications where the RELAY is defined.

is4it-lab, 2020-07-24:
Hi, the ports of the MCP23017 are not shown as relays. The only way to access them is with sensor29 command. That ist the problem. I didn't find a method to assign them to a shutterrelay.

stefanbode, 2020-07-26:
In this case i would vote for changing this and magically all the rest will work. This would be an enhancement to the mcp driver.

berhomeauto, 2020-07-26:
so, here is my +1.
First of all, really want to thank all of you guys, who are making this possible. What a project!

I was facing exactly this problem allready too.

Priority 1:
I allready asked for a solution in https://github.com/arendst/Tasmota/issues/8081, but there was sadly no reaction.
I managed it with some very ugly rulesets, waisting GPIOs.
Shuttermode 0:
It is also necessary to switch the Position of the power- and the direction relais, due to timing problems. It seems that the Relais are set one by one and not bankwise. For now it will first switch on power and then the direction. This causes nasty effects, when the shutter gets for some millis the voltage on the "up" wire and then on the finally "down".
Shuttermode 1:
It could also affect immediate inverting directions shutters with one Relais for each direction. I do not really know how my rules would work in combination with the relais interlock.
Also linking the MCP-Inputs to BUTTONs or SWITCHes would be great, so i could control the shutter over other MCP IOs.

Priority 2:
If someone starts to fix it, I would highly appreciate having the possibility to use 8 shutters (perfect for a MCP23017). I know that they are very memory consuming, but other things consume memory too, I don't use.
Because you have to enable Shutters before buidling it should not affect the standard builds. So the memory issues are limited to users which build on their own.
But this would probably also exceed the limit of 8 Relais?

I allready was trying to dig into the code of PCF and MCP driver, but I failed.

But if I can help testing or something like that... here we go!

Jason2866, 2020-07-31:
There will be no support for more https://github.com/arendst/Tasmota/issues/7751#issuecomment-588957427

stefanbode, 2020-07-31:
Hi @Jason2866 , I slightly disagree. One the one hand for sure currently we will support maximum 4 shutters in the default branch. With some changes you can go up to 16, but this will break configuration compatibility and also add quite a lot RAM. Anyhow it might be a smart idea to expose the new GPIO as additional ports during configuration. This would add more flexibility in all areas. My idea is to have additional GPIO in the config page. Anyhow we must respect that the new GPIO may have less functionality. E.g. PWM or I2C is not supported for example for the PCF.

Jason2866, 2020-07-31:
@stefanbode  Theo decided to support max. 4 shutters. I did not say it is not possible to change. 
I dont wont stop you or everyone else doing changes on the MCPxxx driver. 
If this can be done to the existing functions (small flash footprint) as optional (compile time) addon, it could be a nice option and there is a chance to get it included. My comment was "just" that making braking changes in the MCPxxx driver will not have a big chance to get merged.

stefanbode, 2020-07-31:
I took a closer look to the driver and it looks like we have only to make some smaller changes to the OUTPUT definition like I did for the PCF. This will not break current functionality, and add POWERxx as valid inputs to drive the state. I will make a proposal during the weekend.

is4it-lab, 2020-07-31:
Hi, in my opinion four shutters shold be enough for one Device. 
It would be a great enhancement to be able using the MCPxxx for the connectivity of relays and buttons.

stefanbode, 2020-08-03:
I have added a pull request #9023 for the new functionality. The MCP out pins will added after the GPIO PINS. You can see that you can control with POWERxx the out pin. Current known issue: sensor29 0,ON will e.g. change the state of the pin bt will not change the POWER state. It is advised NOT to use sensor29 to switch the pin and just use the RELAY's. 
Further development: Remove the Interlock as it is no needed anymore, expose the senors pins as buttons to e.g. control the shutter. This must currently be done though rules.

stefanbode, 2020-08-03:
the PR was merged. Please can you check, if the functionality works for you @is4it-lab ?

stefanbode, 2020-08-04:
Changes the INTERLOCK functionality with PR #9025. Now also INTERLOCK Groups are normally supported. This is required if you want to operate more than one shutter with one ESP. also the sensor29 command to change the state is now redirected to the POWER command. This avoid any side effect and/or confusion with the POWER state.
Last open point is now the BUTTON assignment. Never done this before. Therefore unsure if and how this can be done

ascillato2, 2020-08-04:
Great job! Thanks 👍

closed 2020-08-04 by ascillato2

is4it-lab, 2020-08-05:
Great ! - Thank you verry much !

tbekavac, 2020-11-13:
@is4it-lab  
Hi! Were you able to set everything up? 
I'm trying to "move" 4 shutters using 8 buttons (in pairs up/down).. I've set up 8 relays (PB0-PB7) & 8 buttons (PA0-PA7) with MCP23017. 

I have used rule to link buttons to relays:

`on event#MCPINT_D0=1 do POWER1 2 endon` 
`on event#MCPINT_D1=1 do POWER2 2 endon `
`on event#MCPINT_D2=1 do POWER3 2 endon `
`on event#MCPINT_D3=1 do POWER4 2 endon ` etc.

But I can't get button actions: 

> To control shutter #1 by two buttons: Backlog ShutterButton1 1 up 1; ShutterButton1 2 down 1 assigns button #1 to act as an "up" button (1x press open, 2x press 50% position, 3x press 74% position) and button #2 to act as an "down" button (1x press close, 2x press 50% position, 3x press 24% position) for shutter #1 including MQTT publish 

using `ShutterButton<x>`  e.g. `Backlog ShutterButton1 1 up 1; ShutterButton1 2 down 1` ....

I have also used Interlock like this: `Interlock 1,2 3,4 5,6 7,8` and prior to everything used `Backlog ShutterRelay1 1; ShutterRelay2 3; ShutterRelay3 5; ShutterRelay4 7; `

Do you maybe know what am I missing here?

@stefanbode 
Hi Stefan, you are doing a great job here... (and other guys also).. Thank you for that!
Were you able to do something about the "BUTTON assignment" you have mentioned?

is4it-lab, 2020-11-20:
> @is4it-lab
> Hi! Were you able to set everything up?
> I'm trying to "move" 4 shutters using 8 buttons (in pairs up/down).. I've set up 8 relays (PB0-PB7) & 8 buttons (PA0-PA7) with MCP23017 (Tasmota 9.1.0.).
> 
> I have used rule to link buttons to relays:
> 
> `on event#MCPINT_D0=1 do POWER1 2 endon`
> `on event#MCPINT_D1=1 do POWER2 2 endon `
> `on event#MCPINT_D2=1 do POWER3 2 endon `
> `on event#MCPINT_D3=1 do POWER4 2 endon ` etc.
> 
> But I can't get button actions:
> 
> > To control shutter #1 by two buttons: Backlog ShutterButton1 1 up 1; ShutterButton1 2 down 1 assigns button #1 to act as an "up" button (1x press open, 2x press 50% position, 3x press 74% position) and button #2 to act as an "down" button (1x press close, 2x press 50% position, 3x press 24% position) for shutter #1 including MQTT publish
> 
> using `ShutterButton<x>` e.g. `Backlog ShutterButton1 1 up 1; ShutterButton1 2 down 1` ....
> 
> I have also used Interlock like this: `Interlock 1,2 3,4 5,6 7,8` and prior to everything used `Backlog ShutterRelay1 1; ShutterRelay2 3; ShutterRelay3 5; ShutterRelay4 7; `
> 
> Do you maybe know what am I missing here?
> 
> @stefanbode
> Hi Stefan, you are doing a great job here... (and other guys also).. Thank you for that!
> Were you able to do something about the "BUTTON assignment" you have mentioned?

is4it-lab, 2020-11-20:
Hi,
I have a setup without shutterbuttons.

SilentBob83, 2021-11-11:
Hello I am using a ESP32 and I would like to control 6 shutters, is there maybe a way to change/override the number of shutters in the user config for compiling my own binary ? 

In the past I wrote a similar general Smart Device Software such as Tasmota but i could not handle a stable wifi / mqtt connection somewhen it stopped working until i pushed the reset button. It was about 1 1/2 year of programming from scratch.
at that time i didn't know about tasmota yet.

In my SW it was possible to manage up to 8 shutters, with the same functionallity as tasmota has. 
But I moved to tasmota due to my instability problems.

ascillato, 2021-11-11:
Hi, your request is already supported. Use latest Tasmota from the Development Branch.

https://github.com/arendst/Tasmota/issues/13566

SilentBob83, 2021-11-11:
Do I need to change anything in the user_config_override.h ?
I am using the Tasmocompiler , development version(not official 10.0) ... and I just made a fresh binary, but I still cannot configure more than 4 shutters

barbudor, 2021-11-11:
As of now, tasmota is limited to 4 shutters (8 relays, 2 per shutter)

There are some discussion to go for higher number but it's not yet done

stefanbode, 2021-11-12:
Theoretically you just need to change the config MAX_SHUTTER to increase the number of shutters. But this is theoretically. The configuration array must have a precisely size. With the change it will grow beyond the limits. @arendst may allow in the future to store additional configs in the filesystem. Then adding more shutters is just a compile option. Currently this is a hard limit like the 32 Relays or 4 Pulsetimers and so on. Many of these have limits originally coming from esp8266 limits and choosing practical values. Now this cannot be changed easily.