arendst/Tasmota #8925: Disable connection between switch and relay if script function is defined.

kugelkopf123, 2020-07-16:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
Yes!
I use a sonoff 4ch with a self compiled Tasmota FW. Need the script function on the sonoff to read a counter. I also use this Sonoff for my garden lamps. There is an "emergency button" connected via GND and IO2. GPIO2 is defined as switch5 in the device settings. I want to use the emergency button to switch all lamps on/off. Very annoying is that switch5 always automatically switches Relay1. With switch1-4 I can still understand this to some extent. But not with Switch5. According to Docs there seems to be no way to disable this function when using the script. 
According to Docs the function would be deactivated at Rules, if you would assign a function to Switch5 in a rule. At least that's how I understand the documentation.
Anyway. It makes no sense for me to have a choice of 8 switches, which all always control relay 1. 

**Describe the solution you'd like**  
A set option to disable the linking of Switch with Relay. Or all switches via Switch4 without a link to the relays.

**Describe alternatives you've considered**  
Already mentioned.

**Additional context**  
_Add any other context or screenshots about the feature request here._

arendst, 2020-07-16:
Or use rules.

Kindly ask Gemu to fix his scripter.

kugelkopf123, 2020-07-16:
@arendst Thanks for your answer.
As already mentioned, rules is not an option. 
Is there an explanation why all switches are connected to relay1? None that I can see at the moment.
Connecting switch 1-4 to relay 1-4 would still make sense to me. But 5-8 would not. 

Mention @gemu2015 to get him on board.

arendst, 2020-07-16:
It's all about automagic. Switch 5 can only be connected to relay1 as it overflows the max amount of obvious relays. Same for all other switches.

Why on earth would one want to connect additional switches to an already overcrowded CH4 except @kugelkopf123

kugelkopf123, 2020-07-16:
> It's all about automagic. Switch 5 can only be connected to relay1 as it overflows the max amount of obvious relays. Same for all other switches.
> 
> Why on earth would one want to connect additional switches to an already overcrowded CH4 except @kugelkopf123

I'm not sure if you got me right. I have a feeling you're taking this a little too personally. 
I do not want switch5 to be connected to relay1 or any other relay. I want to be able to determine for myself what switch5 does. 
The sonoff 4ch is installed in a power sub-distribution unit and is not easily accessible. Well, I can explain to you why @kugelkopf123 ;) has installed another switch on this unit: So that my parents don't break their legs if the RaspberryPi hangs up again and doesn't turn on the light when they open the garden door, there is a standard light switch which, as mentioned above, is connected to the GPIO2 of the 4CH and turns on the garden lights. =)
Obviously there seems to be a kind of "switch" that "disconnects" the connection when using Rules. So it should be possible to use a set option or even a #define to break the connection. 
That's all I kindly ask for.

meingraham, 2020-07-16:
@kugelkopf123 

Is `SetOption73` an option?

kugelkopf123, 2020-07-16:
> @kugelkopf123
> 
> Is `SetOption73` an option?

I'm not shure if I had tried this. Perhaps not, because its mentioned in the docs for Buttons not for Switches. I can try again when I get home.

PR #8927 (2020-07-16): scripter add event result
## Description:

**Related issue (if applicable):** fixes #8925
in >E section return true with eres=1

also fixes error in linked list.h with new gcc

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.2.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-licens [... cut]
closed 2020-07-16 by arendst