arendst/Tasmota #8929: All devices soft bricked after power outage

thucar, 2020-07-18:
### PROBLEM DESCRIPTION
After a power outage where the powerline experienced low and fluctuating voltage for a prolonged period of time, all Tasmota devices connected to that line were reset to a state where I am unable to get them operational again. They all have set up their AP's but the Web WiFi configuration page is messed up. Also, after entering my WiFi credentials, the devices do not even try to connect to my router. Only solution seems to be to flash all devices again over serial.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic, Sonoff Dual, Sonoff S26, Avatto smart plug, HOCH Circuit breaker, ESP8266-12e
- [x] Tasmota binary firmware version number used: 8.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here: Not possible

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: Not possible

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here: Not possible

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here: Not possible

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
After any sort of power failure or fluctuation, the device retains its configuration and resumes work once stable power has been restored

### SCREENSHOTS
[image]

digiblur, 2020-07-18:
Have you tried a different browser/device?

thucar, 2020-07-19:
@digiblur Yes, I have tried from three different devices - android phone, iPad and PC. There is no difference that I can see.

taste66, 2020-07-19:
CAn you acces the 'Reset configuration' menu? Alternatively you could try to reset via teh console using the menu 'reset 2' This will erase all configuration and you start with the firmware defaults again. Lastly you could try press the button for 40s to get the same results.

thucar, 2020-07-20:
I can not access the "Reset configuration" menu. The only part of the UI I can access is the WiFi setup you usually see when you've first flashed Tasmota on a device. And that part of the UI has messed up CSS as you can see from the screenshot above.

I have tried resetting the devices by keeping their button depressed for 40 seconds but that does not make any difference as well. The devices remain in the same state - their AP is up and running but once I connect, I am unable to configure their WiFi.

stefanbode, 2020-07-24:
I have seen a similar very rare event where the CFG got messed up due to whatever event. In this case you're a bit in trouble. Anyhow the reset 1 or reset 2 is also available as a REST message you can send to your device. 

for example: http://192.168.178.127/rt?non=

just replace the IP-Address

yoav1000, 2020-07-31:
> 
> 
> I have seen a similar very rare event where the CFG got messed up due to whatever event. In this case you're a bit in trouble. Anyhow the reset 1 or reset 2 is also available as a REST message you can send to your device.
> 
> for example: http://192.168.178.127/rt?non=
> 
> just replace the IP-Address

Can we use the flash area reserved for update to backup configuration and create some sort of sanity check/ journaling?

thucar, 2020-08-01:
> I have seen a similar very rare event where the CFG got messed up due to whatever event. In this case you're a bit in trouble. Anyhow the reset 1 or reset 2 is also available as a REST message you can send to your device.
> 
> for example: http://192.168.178.127/rt?non=
> 
> just replace the IP-Address

Unfortunately this does not seem to work. I tried connecting to the devices AP's and sending http://192.168.4.1/rt?non=
No response and no change in the device behavior.

ascillato2, 2020-08-01:
> Unfortunately this does not seem to work.

Sorry, seems that your device has a problem in its memory. A full erase may solve your problem but if the root of the issue was a hardware memory failure, it can happen again. It would be good to use that device in something not critical. 

> Only solution seems to be to flash all devices again over serial.

Agree. You should erase all flash and flash by serial in order to try recovering your device.

closed 2020-08-01 by ascillato2

thucar, 2020-08-02:
@ascillato2 I do not believe it very likely that all 6 of those devices experienced a hardware memory failure. I'm not so much worried about a solution on how to fix the devices at this time, but more about if it could happen again.

If all devices in my house can get reset like that, even if it is once in a blue moon kind of thing, then that means Tasmota is not a firmware that is safe to be used in anything that is remotely critical.