arendst/Tasmota #8930: Add new SwitchMode which sends MQTT message when MQTT connection is established and does not turn on/off associated Relay, and turns on/off the relay otherwise

xmas79, 2020-07-18:
**Have you looked for this feature in other issues and in the docs?**  
Yes. 

**Is your feature request related to a problem? Please describe.**  
Similar to #3264, but addresses the case of MQTT connection is temporary not available (eg. enabled but can't connect to the broker). Adding the rules as per #3264 to publish a message (and to stop the relay) doesn't work in this case: the rule fails to publish the message, but the associated relay doesn't flip.

The typical scenario is a tasmota device connected to a physical relay that physically turns on/off a smart bulb: you want the smart bulb always powered on and control it via Home Automation software (Home Assistant etc...), but you also want to be able to control it physically in case HA is not available (by directly controlling the physical relay attached to the tasmota device).

**Describe the solution you'd like**  
A new switchmode that makes button/switches act like the following:
1. Send the standard MQTT message when the MQTT connection is established (and no relay action is performed)
2. The relay action is performed immediately when the MQTT connection is not established

**Describe alternatives you've considered**  
I have an "if" support compiled firmware, and I'm able to implement this with the following rule:
```
rule1 on mqtt#connected do var1 1 endon on mqtt#disconnected do var1 0 endon on button1#state=2 do if (var1==0) power1 1 endif endon
rule1 1
```
The rule solution makes firmware upgrades hard because it needs the "if" support in the firmware.

Having a precompiled binary with the "if" feature is also a viable solution as it allows firmware upgrades without rebuilding custom firmware.

ascillato2, 2020-07-18:
Hi

It is already implemented what you are asking for. 
That is the default behavior of Tasmota when setting `switchtopic`

Please, check the docs for `switchtopic` and test if this behaves as you want. 
Thanks.

closed 2020-07-18 by ascillato2

xmas79, 2020-07-18:
Hi Adrian,
sorry but it seems it doesn't work for me (on 8.3.0). When using `switchtopic 1` I always get the local action (when no rules are intercepting the button press). If I put the rule and I'm not connected to the broker I get nothing at all.

ascillato, 2020-07-18:
Switchtopic is for switches. You need to configure your button as a switch with its correct switchmode.
Then with switchtopic you define a topic for a switch. The topic you want is 1?

xmas79, 2020-07-25:
@ascillato sorry for the late response. Making the story short, I started over with a new flashed device and everything works as expected. Thanks for your support. BR.