arendst/Tasmota #8931: H801 RGBW LED Stripe - Feature request for CT (RGB + Warmwhite)

michaelkandziora, 2020-07-18:
### PROBLEM DESCRIPTION
I'm controlling a RGBW (W: Warmwhite) LED Stripe with a H801 module running Tasmota 8.3.1.

I want to control the CT value with my Alexa. 
Alexa is connected to my "FHEM Smarthome Server" and the H801 module via MQTT too.

Changing the rgb colours is working fine and changing to "warmwhite" too, but on selecting "coldwhite" the module sets the channels to [0,0,0,100,0] or [x,x,x,100,0] (depends which "Option37" mode).

I have tried this with Option37 set to 128 and Option37 set to 0.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): H801 (20)
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled

### TO REPRODUCE
_Steps to reproduce the behavior with SetOption37 0:_
SetOption37 0

CT 500
_Warmwhite LEDs turn on_
_Result: /stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"00000000FF","HSBColor":"255,0,0","White":100,"CT":500,"Channel":[0,0,0,0,100]}_

CT 200
_LED stripe dark_
_Result: /stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"255,0,0","White":100,"CT":200,"Channel":[0,0,0,100,0]}_

HSBColor 255,0,255
_LED stripe still dark_
_Result: /stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF0000","HSBColor":"255,0,100","White":0,"CT":200,"Channel":[100,100,100,0,0]}_

.

_Steps to reproduce the behavior with SetOption37 128:_
SetOption37 128
Power1 off
Power2 off

CT 500
_Warmwhite LEDs turn on_
_Result: /stat/RESULT = {"POWER1":"OFF","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFF00FF","HSBColor":"0,0,100","White":100,"CT":500,"Channel":[100,100,100,0,100]}_

CT 200
_LED stripe dark_
_Result: /stat/RESULT = {"POWER1":"OFF","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFFFF00","HSBColor":"0,0,100","White":100,"CT":200,"Channel":[100,100,100,100,0]}_

HSBColor 0,0,255
_LED stripe coldwhite_
_Result: /stat/RESULT = {"POWER1":"ON","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFFFF00","HSBColor":"0,0,100","White":100,"CT":200,"Channel":[100,100,100,100,0]}_

### EXPECTED BEHAVIOUR
First of all, I will explain what I expected while using SetOption37 to 0:
The behaviour is right, tasmota translates the "coldwhite" value to the cw channel (4).
But it would be nice, if there would be an option to set for a 4 channel LED stripe with warmwhite leds as fourth channel.

The confusing thing is, that on setting ct value to 200 the result shows that all rgb channels are set to 100 percent but none of them are glowing. (I guess this comes from the white value translation to the cw channel)

I have already tried to fix this with the RGBWWTable adjustment option, but that did not solve the problem. 
Maybe misunderstood this feature or this is maybe not the right thing for this issue.

The expected behaviour with SetOption37:
I thought SetOption37 would separate the ct value (for coldwhite) for the rgb channels and the cw channel.
The only thing that changed is that coldwhite is showing up on the led stripe.

### ADDITIONAL CONTEXT
_SetOption82 is set to 1 for Alexa CT value range (200-380)._

Links which could the interesting for this issue:
https://tasmota.github.io/docs/SetOption37/
Combining RGB with W1 or W2 is not possible any more https://github.com/arendst/Tasmota/issues/5741
Allow all 5 PWM channels individually adressable with LEDs. https://github.com/arendst/Tasmota/pull/5748

s-hadinger, 2020-07-18:
Emulating cold/warm white with a 4 channels RGBW is currently not supported. It's in the todo list but don't expect it before a few weeks.

s-hadinger, 2020-08-15:
PR will soon be available

closed 2020-09-20 by stale[bot]